ckb-next/ckb-next #60: v0.2.9 Issue

Fluffinater, 2018-01-22:
I just wanted to let you know that when I tried to install v0.2.9 today it did absolutely nothing after the install process. I appeared to install correctly and the application was sitting where it should be but when I went to open it no window would appear. I uninstalled it and went back to v0.2.8 and am having no issues.

I have a MacBook Pro (Retina, 15-inch, Mid 2014). More detailed system specs are available if you need them.

light2yellow, 2018-01-22:
Not reproducible on 10.13.2.

Try uninstalling:
https://github.com/ckb-next/ckb-next/wiki/macOS-Installation#uninstallation

Then installing 0.2.9.
Then rebooting.

aemc, 2018-01-23:
I too have the same issue. Uninstalled the previous version and installed this one. Does not do anything after trying to open the app. Macbook Pro touchbar 2016

light2yellow, 2018-01-23:
Okay, what happens if you do `open -a ckb`?

aemc, 2018-01-23:
LSOpenURLsWithRole() failed for the application /Applications/ckb.app with error -10810.

light2yellow, 2018-01-23:
Interesting. What is the output of `ls -l /Applications/ckb.app`?

ngthduy90, 2018-01-23:
> Interesting. What is the output of ls -l /Applications/ckb.app?
Ouput: drwxr-xr-x  6 root  wheel  192 Jan 22 03:27 Contents

I have a same problem, cannot run CKB app. And when I run by Terminal. It shows

 dyld: Library not loaded: /usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Applications/ckb.app/Contents/MacOS/ckb
  Reason: image not found
[1]    2687 abort      /Applications/ckb.app/Contents/MacOS/ckb

light2yellow, 2018-01-23:
Thanks, I guess I know what the problem is, at least one of them, because there might be two, actually. Firstly, for some reason the Qt libs were not bundled, secondly, I am 99% sure `root:wheel` is wrong. Interesting, though, what is the problem, since it used to work somehow half a year ago.

Fluffinater, 2018-01-23:
Welp, I am just now able to sit back down and look at this. Sorry I wasn't much help after bringing up the problem but I'm glad it wasn't just me doing something stupid. Was kind of feeling like a failure in my field for a second there.

aemc, 2018-01-23:
Downgraded to 0.2.8. Now my keyboard is not detected :( Lost the rgb functionality

K70 RGB Lux
Macbook Pro Touchbar 2016

Fluffinater, 2018-01-23:
While I do have a K65 RBG I wasnt trying to hook that up. I was just trying to get my Scimitar PRO RGB to work, which by the way it did wonderfully after I went back to v0.2.8. I actually think this application works better in the ease of use department than the actual Corsair CUE does personally. When it works I mean.

light2yellow, 2018-01-23:
Should be fixed now. Redownload the pkg from 'Assets'.

Fluffinater, 2018-01-23:
Hey it worked! The pkg installed and opened fine. I'll have to wait until I get home to check if the mouse will work right with it. Thanks for being on top of it man!

aemc, 2018-01-23:
It works on my end too!

closed 2018-01-23 by light2yellow