ckb-next/ckb-next #64: K95 - crashed after click "change color..."

s3bul, 2018-01-24:
# Bug report

- [x] Yes
- [ ] No

## General information

### Source
I get sources from github (from here) and i run quickinstall

### Branch
master

## Logs & crash reports
[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1661720/ckb-dev-detect-report.gz)

## What is the program's current behavior?
When i click "Change color..." the app crashed.

## What is the expected behavior of the program?
I would like change colors ;]

ZirconiumX, 2018-01-24:
You have [this bug](https://github.com/ckb-next/ckb-next#fedora-26-color-changer-freeze-fix).

closed 2018-01-24 by ZirconiumX

s3bul, 2018-01-24:
Ok, but im not using fedora.

s3bul, 2018-01-24:
I get qt5ct from https://github.com/mati75/qt5ct and seems it is working.

Billy-, 2018-01-25:
I have tried installing qt5ct from the link above, after a lot of troubleshooting it seems to have installed properly, but ckb still crashes. I'm completely out of ideas, can anyone provide any tips, advice or pointers please?

Billy-, 2018-01-26:
I also tried installing qt5ct with apt-get, the [webupd8](https://www.ubuntuupdates.org/package/webupd8/zesty/main/base/qt5ct) ppa says it has it, but I still can't find it from there. I found it [here](https://launchpad.net/~mati75/+archive/ubuntu/qt5ct). However still not luck. It installed, I rebooted, I reinstalled ckb-next, I rebooted, still crashes.

tatokis, 2018-01-26:
Tried `QT_QPA_PLATFORMTHEME=qt5ct ckb` ?

Billy-, 2018-01-26:
I tried with `QT_QPA_PLATFORMTHEME=qt5ct` and `=gtk qt5ct` based on some other thread I saw, I think it was an Arch Linux one (I am on Ubuntu 17.10). I will try the above when I get in.

Can you advise how to check this variable is set correctly and also check if ckb-next is using the correct qt version/theme etc? And/or how to print any other relevant debug information? I ran `echo "$QT_QPA_PLATFORMTHEME"` and it seems to be unset in the terminal - but I'm not sure if it should be set there or not..(?) (sorry if my wording is wrong; I don't have a great understanding of how this all works)

light2yellow, 2018-01-26:
@Billy- this is an upstream issue, I don't know what we can do about it (other than you trying different solutions from the Arch Linux thread from README). Report it to your distro maintainers.

light2yellow, 2018-01-26:
@Billy- also, if you have time, please try building #65 (following https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#build-from-source) and see if you can reproduce the issue, and if you can, copy the output of CMake configuration.