ckb-next/ckb-next #74: Wireless K63 No devices connected (when wired or not)

felbus, 2018-02-05:
Hi,

I am getting no devices connected when connecting this device. I have read all other issues, and it seems support was added, but I am having no luck.

ckb-dev-detect-report is at bottom of this issue.

I have tried building from quickinstall and using the package. Have the latest xcode, and have plugged in to windows and upgraded firmware.

Have removed ckb repo, and qt5, then reinstalled, rebooted, and tried all other suggestions, but still getting No devices connected.  

Current behaviour is that keys work, and are mapped to mac layout (keyboard is US layout), led is working. I can plug in to a windows machine, change the lighting, then put it back in mac and it is saved. 

ckb-dev-detect-report:
```ckb-dev-detect - Mon Feb  5 19:34:11 GMT 2018

Kernel: Darwin X-MacBook-Pro.local 17.4.0 Darwin Kernel Version 17.4.0: Sun Dec 17 09:19:54 PST 2017; root:xnu-4570.41.2~1/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.13.3
BuildVersion:	17D47

USB Devices:
                CORSAIR K63 Wireless Mechanical Gaming Keyboard:

                  Product ID: 0x1b45
                  Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
                  Version: 1.02
                  Serial Number: 1401A00BAF15842759618230F5001BC4
                  Speed: Up to 12 Mb/sec
                  Manufacturer: CORSAIR
                  Location ID: 0x14142000 / 10
                  Current Available (mA): 500
                  Current Required (mA): 500
                  Extra Operating Current (mA): 0
```

ZirconiumX, 2018-02-05:
The daemon doesn't have the USB product ID yet. That in itself isn't a major problem. But they might have tried to change something, so you should join us in IRC.

tatokis, 2018-02-05:
@felbus Can you please try the code from the PR above and check if it works?

felbus, 2018-02-12:
Hi, it was plugged in.

ZirconiumX, 2018-02-12:
Can you take another ckb-dev-detect with it in wireless mode?

felbus, 2018-02-12:
Yep, ill have to do tonight when I get home. Thanks for looking.

ZirconiumX, 2018-02-14:
@felbus Bump.

felbus, 2018-02-15:
Wireless https://pastebin.com/1iBnMPqB

felbus, 2018-02-15:
Plugged https://pastebin.com/2TRJsLrm

felbus, 2018-02-15:
sorry for the delay

bek0s, 2018-02-25:
Hi guys,

I've got the K63 Wireless keyboard:
1) The keyboard works just fine with the USB cable
2) The keyboard does NOT work using the WiFi.
3) The keyboard works with the Bluetooth (BT). I get some random disconnects which forces me to disable and re-enable the BT from my laptop, but I think this is a problem of my BT Linux driver.

My operating system is Debian Buster (aka Testing, aka 10) updated just a few minutes ago.

Here is a dump of the ckb-dev-detect script while the keyboard was connected through BT and the WiFi receiver was also connected on the laptop. : https://pastebin.com/FkbtMRQA

After searching for a few minutes on Google I found a lot of people complaining about Corsair products and Linux support. However, I really don't understand why the K/B works just fine with BT but not with the WiFi receiver. I have never encountered this problem before, and trust me, I have used tens of different WiFi receivers. Can someone explain to me what is so special about the corsair WiFi receiver? Maybe the company uses bleeding edge hardware that takes time to be supported by the kernel? Or there is something else going on? Thank you for your time.

tatokis, 2018-02-25:
@bek0s assuming the dongle doesn't need pairing, and that by WiFi you meant wireless, it's most likely due to a bad HID/USB implementation. 

A few corsair wired mice also don't work with the Linux hid driver, at all. 
I assume adding the k63 to ckb with any changes needed will get it working wirelessly as well. 

I don't remember who it was, but someone complained about it disconnecting over Bluetooth before. So if it wasn't you, it seems to be a device issue.

ZirconiumX, 2018-03-01:
@bek0s I'm guessing it was you who dropped by IRC earlier. Yes, we do plan to support the K63 Wireless, but we need somebody to test it. You arrived at 4am; all the devs were asleep, so maybe try a bit earlier next time.

ZirconiumX, 2018-04-01:
@bek0s @felbus We need somebody to join IRC and help us dump the USB descriptors so we can emulate the device and talk to CUE.

adresner, 2018-05-05:
@ZirconiumX I just got the K63, know how to use IRS and am happy to help/contribute.

ZirconiumX, 2018-05-05:
Mostly I've just not really had the time to code it. It requires moving
some code from led_mouse.c to a new led_wireless.c (they changed the
protocol again) and creating a new vtable for these.

adresner, 2018-05-05:
Thanks for the reply, totally get it, hope you have time soon =)

gilvbp, 2019-01-10:
How can i help?

ZirconiumX, 2019-01-11:
Join us in IRC. We have *most* of the protocol implemented in the wireless branch, but it needs to be rebased, we need to factor out the hacky code I added for testing, we need to work out when the keyboard is connected or not, and we need to get simultaneous firmware update working.

CJ1169, 2019-10-17:
I've been running the wireless branch with a K63 and a Dark Core for a bit over a month. The Dark Core seems stable and I haven't had a single issue with it so far. Granted, I haven't tested all features as I've only used the features I normally use (non-animated lighting, dpi profiles, colors, wired/wireless, etc).
However, the K63 has had a particular issue. Its hard to determine what exactly that issue is for me, but I can provide a decent description and potentially some logs if needed. 

Symptoms: Keyboard does not accept any lighting changes. LED key still cycles through brightness settings in program, but keyboard does not change. all LEDs may be turned off (although IIRC, this is my hardware profile on my keyboard). 
How to correct the problem: Boot up windows and use iCUE to modify keyboard. Reboot to Manjaro.
How to recreate the problem: ??? Not sure, but at least anecdotally it seems to happen on reboot. Its not always consistent.

I've also tried various methods with replugging in the keyboard, using RFUSB, and rebooting without using windows. Using RFUSB and directly plugged in simultaneously or not does not seem to affect the problem in any notable way.

purezen, 2020-08-29:
Hi, I installed the app on a fresh install of pop os 20.04.

It's not detecting my K63 wireless in either modes - wifi / usb. Is it supported?

tatokis, 2020-08-29:
If you have the old k63 wireless, there's some code and some PRs in the wireless branch, but you can't control the lighting from the ui yet.

purezen, 2020-08-29:
Thanks for the response

Don't know which model you are referring to as old. I bought mine around half an year back.

Is it not supported in the usb mode too? If not lightning are the key binding settings working?

tonymurray, 2021-12-06:
Hi, I have a new K63 keyboard wireless. I installed the latest version 0.44, but get 'no devices connected' (also with various bios fixes). 
I then installed the 'wireless' branch and now the connection works  ... both wired and using the wi-fi dongle. However, I cannot get any lighting effects to display on the keyboard.  (I expected lighting effects to show on keyboard on selecting an effect - similar to the Windows ICue software) Is there anything I should try, or something I am missing?

Grawp, 2022-04-08:
May I ask what _wireless_ branch are you all referring to? I see not branch with `wireless` in its name :(

tatokis, 2022-04-10:
@Grawp it's all been merged to master, so you can also use any of the -git packages.
You need to run ckb-next-daemon with `--enable-experimental`.
Everything but lighting should work through the GUI. There is an interface for animations (hwanim command) but I couldn't figure out how it was supposed to work when testing it with someone else.

Grawp, 2022-04-10:
@tatokis I had already tried master before but I was missing the `--enable-experimental`. Thanks!

Now I have another problem. I want to bind the **FN** key to also work as right-winkey. (I use both winkeys in Sway, i3, River windows managers.)
This absolutely works in Windows and the FN+F9,10 still works normally (even though they probably send additional Win+F9,F10 keypress which is not a problem for me). But the bindings tab in ckb-next's GUI does not contain the FN key (the GUI dialog is probably using wired-K63 layout). I guess I'll fire up QtCreator next week and have a look at it.

tatokis, 2022-04-10:
@Grawp It looks like there is also a K63 wireless that has rwin. Can you post the output of `lsusb -d1b1c:`?
To fix this, you need to add a new keymap in the GUI that is based on the existing one and just patches Fn on top of rwin (you can see keymap.cpp/keymap.h for how to do that).

On the daemon side you need to tell it to return the appropriate string for the new keymap in usb.c so that the GUI loads your new one. Make sure to return it only for the usb ids (both wired and wireless) that belong to your specific K63 and not the other one (that presumably has rwin).

Grawp, 2022-04-10:
- when connected via wire:
  ```
  Bus 001 Device 016: ID 1b1c:1b8c Corsair CORSAIR K63 Wireless Mechanical Gaming Keyboard
  ```
- when connected wireless:
  ```
  Bus 001 Device 017: ID 1b1c:1b8f Corsair CORSAIR K63 Wireless USB Receiver
  ```

I've noticed that there are two wireless K63s though I've thought that they were same layout-wise. If I'm not mistaken mine is one which is referred to as the _new_  one?

tatokis, 2022-04-10:
Looks like it, yeah. I guess this is why they made new usb ids for it.
A PR would be appreciated.

tatokis, 2022-04-10:
It appears that the older K63 (0x1b50 dongle) also has Fn instead of rwin, so I guess you should add both K63s to the new layout.

Grawp, 2022-04-22:
If I may ask: What is missing for support of static light control? (At least having the ability to turn off all button LEDs.)
Is that part of the protocol not known or just not implemented or just not tested and turned on on K63?
Thanks.

tatokis, 2022-04-26:
@Grawp 
I actually don't know how to turn off the lights on the keyboard, however hardware animations are supported in the daemon only. There's a brightness command that might work, however it's not user-configurable in ckb-next.

You can find the animation ids here https://github.com/ckb-next/ckb-next/blob/cb37c692243bdb69b7217e82dba2f948b063861d/src/daemon/led_wireless.c#L6-L36
and run them using `echo "hwanim a:X:Y:Z" > /dev/input/ckbA/cmd` while the GUI is running.

for example `echo hwanim a:4:2:1 > /dev/input/ckb1/cmd` will run animation 4 (wave) with speed 2 (medium) and direction 1 (left) on device ckb1.

The command will probably change in the future to accommodate the hw animations of newer devices.

gravityfargo, 2024-05-01:
[OP](https://github.com/ckb-next/ckb-next/issues/74#issue-294522150): 0x1b45 CORSAIR K63 Wireless Mechanical Gaming Keyboard\

[Pastebin](https://github.com/ckb-next/ckb-next/issues/74#issuecomment-365837628)  0x1b50 CORSAIR K63 Wireless USB Receiver

Both ids were added with 3278cc9

Marking this issue as closed, any future problems are bugs related to experimental support beyond the scope of this issue.

gravityfargo, 2024-05-01:
Appologies, I forgot that issues stay open until they leave experimental status!

jermy12321, 2026-05-17:
Does this issue exclude 1b8f and 1b8c? Should a separate issue be opened for that version of the k63?