ckb-next/ckb-next #78: GUI Freezes when I try to change keyboard light colors

adplax18, 2018-02-06:
__Switch on the "Preview" mode for better reading experience.__
Use it to put ticks and then switch back to write in the details where needed.

- [Before you proceed](#before-you-proceed)
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior)
    - [What would you like it to do instead?](#what-would-you-like-it-to-do-instead)
    - [Any other notes & comments?](#any-other-notes--comments)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)
- [Something else](#something-else)

---

# Before you proceed

- [X] __Try__ the latest code by cloning and building the project manually.
- [X] __Read__ the [`README.md`](https://github.com/ckb-next/ckb-next/blob/master/README.md) _in full_.
- [X] __Skim__ through other `.md` documents in the project's root. They contain more specific information.
- [X] __Search__ in open _and_ closed [Issues](https://github.com/ckb-next/ckb-next/issues). [Here's how](https://help.github.com/articles/searching-issues/).
- [X] __Avoid__ opening a new issue in case a similar one already exists (don't worry, we get the notifications for closed issues as well, it will be reopened if needed).
- [X] __Keep__ unrelated issues separate (create a new one if you are sure yours is unique).
- [X] __Use__ [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to write on GitHub. Use triple backticks if copy-pasting more than _one_ line.

Now, __before you proceed even further__, make sure you are following the recommendations above. It will save a lot of time for you and us in the future.

---

# Devices information

Run [`ckb-dev-detect`](https://github.com/ckb-next/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.

---

# New device support request

- [ ] Yes
- [X] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

- [ ] Yes
- [X] No

<sub><sup>leave empty if you have no feature requests</sup></sub>

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

- [X] Yes
- [ ] No

<sub><sup>leave empty if you have no bug reports</sup></sub>

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)
installed from git- reinstalled today, 2-6 to ensure this issue wasn't already fixed
### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)
master
## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)

I ran ckb from the command line and got the following output when the error occurred:
```
QObject::startTimer: Timers cannot have negative intervals
Path  to settings is "/home/alan/.config/ckb/ckb.conf"
Found 3 components in the firmware list.

(ckb:26175): GLib-GObject-WARNING **: cannot register existing type 'GtkWidget'

(ckb:26175): GLib-GObject-WARNING **: cannot add class private field to invalid type '<invalid>'

(ckb:26175): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'

(ckb:26175): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:26175): GLib-GObject-WARNING **: cannot register existing type 'GtkBuildable'

(ckb:26175): GLib-GObject-CRITICAL **: g_type_interface_add_prerequisite: assertion 'G_TYPE_IS_INTERFACE (interface_type)' failed

(ckb:26175): GLib-CRITICAL **: g_once_init_leave: assertion 'result != 0' failed

(ckb:26175): GLib-GObject-CRITICAL **: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(ckb:26175): GLib-GObject-CRITICAL **: g_type_register_static: assertion 'parent_type > 0' failed

(ckb:26175): GLib-GObject-WARNING **: cannot add private field to invalid (non-instantiatable) type '<invalid>'
```
You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?
When I try to change a light (select light on keyboard map and then press the color button), the application freezes and has to be force quit
## What is the expected behavior of the program?
Allow changing of key colors
## Any other notes & comments?
[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1701073/ckb-dev-detect-report.gz)

---

# Something else

<sub><sup>if your issue is about something else, try your best to describe it here, otherwise leave empty</sup></sub>

ZirconiumX, 2018-02-06:
You have [this bug](https://github.com/ckb-next/ckb-next#fedora-26-color-changer-freeze-fix).

ZirconiumX, 2018-02-06:
More specifically, this bug is due to a Qt bug, and there's not much we can do about it. The best workaround we have is to install `qt5ct` - whichever package that is - and then set it [like this](https://github.com/ckb-next/ckb-next/issues/64#issuecomment-360719281).

closed 2018-02-06 by ZirconiumX

ZirconiumX, 2018-02-06:
The [Arch Linux thread](https://bbs.archlinux.org/viewtopic.php?id=214147) just above where I linked in the readme might also be useful.

adplax18, 2018-02-06:
AH, my apologies- I skipped that as I have Ubuntu and not Fedora.  I will try what you mentioned above.

Thanks.