ckb-next/ckb-next #81: Void headset support

ZirconiumX, 2018-02-09:
I'm putting this in a separate issue so that we don't pollute #52.

From @aldistuck:
[CorsairUSBVoid.pcapng.gz](https://github.com/ckb-next/ckb-next/files/1709461/CorsairUSBVoid.pcapng.gz)
I attempted to use Wireshark to capture usb communication from my Corsair usb Void headset (corded). From the start the UI could see usb connections. One of them contained the headset so I just started a capture on that so I'm assuming everything in this is only between the headset and what the capture was calling the host. Nothing began happening until I opened the CUE 2 software (captured this on Windows 10). Once CUE 2 was open I started seeing data in Wireshark. I went to the settings for the headset and cycled through all the settings for lighting effects, EQ, and infomic. I hope this is helpful information.

[CorsairUSBVoid-0.pcapng.gz](https://github.com/ckb-next/ckb-next/files/1709571/CorsairUSBVoid-0.pcapng.gz)
[CorsairUSBVoid-1.pcapng.gz](https://github.com/ckb-next/ckb-next/files/1709572/CorsairUSBVoid-1.pcapng.gz)
Here are a couple more traces I recorded. I opened up a youtube video to get audio going through the headset. The -1 capture seems to have an additional source I did not have prior.

ZirconiumX, 2018-02-09:
Also, an `lsusb` can be found [here](https://github.com/ckb-next/corsair-protocol/blob/master/lsusb/1b1c/1b2a/0007).

ZirconiumX, 2018-02-09:
Right, the protocol for this is completely different to the currently used and extended protocol at the moment; it's built around control and isochronous transfers, where the current HID devices use interrupts. Additionally, the endpoints and packet capture seem to show that the void controls its own LED, and the GUI merely tells it to use a preset mode. It's possible we might decide to hack on this anyway for the hell of it, but don't get your hopes up.

SethBarberee, 2018-02-09:
Another struggle is trying not to claim the audio endpoints of the headset. When you add device ID, the daemon claims all the endpoints that it finds. To this end, it removes the headset from being an audio device.

ZirconiumX, 2018-02-09:
This is true, and a valid point. I think @tatokis mentioned he was planning to do something about that, but still.

ZirconiumX, 2018-02-11:
So, I've been [fishing](https://gist.github.com/ZirconiumX/620140e5068e8300181def7a34d36c07) through the CUE files, and have found a *lot* of IDs for different Void devices. This could be a headache and a half.

SethBarberee, 2018-02-12:
> could be a headache and a half

What especially will be the difficult part? I can contribute what I can. I have a Void headset myself
From Corsair's website, they all support RGB. However, I am curious why they have seperate IDs.....

http://www.corsair.com/en-us/gaming-headsets

ZirconiumX, 2018-02-12:
The packet trace we got shows a completely different protocol to the HID
device protocol we have been using, and it would involve reverse
engineering that from scratch to get usable support; we're really not sure
it's worth it if they break the protocol for each Void.

SethBarberee, 2018-02-12:
> it's worth it if they break the protocol for each Void

Perfectly understandable given the scope and size of this project. Thank you for all of your efforts so far!

ZirconiumX, 2018-02-12:
If you drop by the IRC channel - sometime much later than now, since it's
4am and I have school in five hours - I can see if your particular Void is
any better.

daddydeadpool, 2018-04-19:
any update on how this is going?

tatokis, 2018-04-19:
@daddydeadpool I have created https://github.com/ckb-next/ckb-next/tree/void including some of the work by @SethBarberee.
It should be able to try and talk to the void usb without killing off audio. (Linux only for now.)

We know nothing about the current protocol, so it doesn't do anything useful, but if someone with an actual void could test it and make sure it doesn't break anything, it would be appreciated.

Additionally, I need someone to press any physical buttons on the void and capture the response.

The captures on the first post are a bit confusing, so if someone has a Virtualbox VM with windows running on a linux host, I'd like to have some new captures.

Once again, feel free to drop by the IRC channel to work on this.

TAnthrax, 2018-04-19:
@tatokis Thank you so much! I'll see where I go testing on linux and may be able to help with the captures. :) I'll drop by IRC if I run into any issues.

SethBarberee, 2018-04-20:
@tatokis Thanks for updating that branch! I should be able to pop on IRC this weekend and test some more of the code.

EvangeIion, 2018-05-16:
Hi, I was wondering if this branch also applied to the wireless ver. of this headset?

ZirconiumX, 2018-05-16:
Nope, we're currently focusing (for a given phrase of "focus") on the wired
version alone. If we could capture packets from the wireless headset, that
would help. It's possible they share the same protocol, however.

tatokis, 2018-05-16:
It doesn't, but you can use lsusb to get the device id for your void and replace one of the existing ones with yours in usb.h https://github.com/ckb-next/ckb-next/blob/void/src/daemon/usb.h#L110

You should be able to see if running the daemon breaks anything, no device commands are implemented at the moment, so if you are expecting it to do anything, you'll be disappointed.
Basically what @ZirconiumX said, with my guess being the protocol is pretty similar, if not identical.

Captures would be appreciated.

Lemmmy, 2018-07-24:
If the information is helpful to you at all, here is an [lsusb](https://gist.github.com/Lemmmy/599c73c28ab9cd6b81545326494580fc) for my Void Wireless RGB.

tatokis, 2018-08-22:
@Lemmmy feel free to drop by our IRC room any time you wish

I have hacked together some basic void support.
@SethBarberee or anyone else, can you please test the daemon *only* from the void branch?
```
git checkout void
mkdir build && cd build
cmake ..
make -j4 daemon
sudo ./bin/ckb-next-daemon
```

Once it starts, check that audio on the headset is still working.
Then, 
```
echo active > /dev/input/ckbX/cmd
```
where X is the node id the daemon assigned to the void.

Verify that the headset is now in software mode. I don't know the details, but you should see some behaviour change on the LEDs.
```
echo rgb zone1:ffffff > /dev/input/ckbX/cmd
```
This should set the LEDs white. If so, also try with `ff0000`, `00ff00`, and `0000ff`.

Then run
```
echo idle > /dev/input/ckbX/cmd
```
and check if the headset is back in hardware mode.

SethBarberee, 2018-08-22:
Yep, I’ll get to it this week

Lemmmy, 2018-08-22:
I'm assuming this is just for the wired void? With my void wireless, it does not seem to connect:

```
zedongarch > sudo ./bin/ckb-next-daemon
    ckb-next: Corsair RGB driver 0.3.1-46-gb4e4871
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming SCIMITAR PRO RGB Mouse at /dev/input/ckb1
[I] Connecting Corsair K70R Gaming Keyboard at /dev/input/ckb2
[I] Starting input thread for /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb2
[I] Setup finished for /dev/input/ckb2
[I] Received identification packet: 0e 01 00 00 01 01 00 01 10 01 01 01 1c 1b 3e 1b 01 01 01 30 c1 ff 40 01 46 00 01 01 00 00 00 00 00 00 00 00 01 00 00 01 00 00 02 01 01 01 aa 00 6c dc 02 00 00 00 00 00 00 00 00 00 00 00 00 00
[I] Setup finished for /dev/input/ckb1
```

Audio still works fine after starting the daemon with void support.

SethBarberee, 2018-08-22:
Currently, it only supports the Wired as the USBID was easily accessible for me. I’m away from my pc but ckb-dev-detect might be able to pull the USBID for the wireless and it shouldn’t be too hard adding it

Lemmmy, 2018-08-22:
[Here is the report from ./ckb-next-dev-detect](https://you.wouldnot.download/a-friendly-paste.gz)

Lemmmy, 2018-08-22:
[Plaintext (non-gzipped) gist](https://gist.github.com/Lemmmy/ffc92bd01e39be5338ee92da6c2351cb)

SethBarberee, 2018-08-22:
‘1b1c:1b27’ is the Wireless Void. 
With the lsusb posted earlier, (unless someone beats me to it) I want to cross-examine the Wired/Wireless to make sure it would work by simply adding in the ID.

SethBarberee, 2018-08-22:
@Lemmmy 
To try out Wireless with the code, you could add `1b1c:1b27` underneath [here](https://github.com/ckb-next/ckb-next/blob/void/src/daemon/usb.h#L117). Similar to the format of the others and see if that works.

tatokis, 2018-08-22:
I added that void wireless dongle to the code. Git pull and try again.

Jyrka98, 2018-08-22:
Would it be possible for Void Pro support to be added?

ZirconiumX, 2018-08-22:
Sure, but first we would like to get the wired Void working.

SethBarberee, 2018-08-22:
@tatokis 
Works perfectly for me. When it switches to software mode, the LEDs turned off on mine. The commands to change color worked on all test cases. Switching back to hardware mode worked perfectly. In summary, great job. It works amazingly!

Tested on my Wired Void RGB Headset

Lemmmy, 2018-08-23:
Tested the latest 'void' branch on my Wireless Void RGB. Some strange behaviour happens:

```
   ckb-next: Corsair RGB driver 0.3.1-47-g009ee19
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming SCIMITAR PRO RGB Mouse at /dev/input/ckb1
[I] Connecting Corsair VOID Wireless Gaming Dongle at /dev/input/ckb2
[I] Connecting Corsair K70R Gaming Keyboard at /dev/input/ckb3
[I] Starting input thread for /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb2
[E] os_inputmain (usb_linux.c:346): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7.4/1-7.4:1.0/ep_81
[E] os_inputmain (usb_linux.c:346): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7.4/1-7.4:1.1/ep_82
[E] os_inputmain (usb_linux.c:346): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7.4/1-7.4:1.2/ep_83
[E] os_sendindicators (usb_linux.c:258): Broken pipe
[I] Attempting reset...
[I] Starting input thread for /dev/input/ckb3
[I] Setup finished for /dev/input/ckb3
[I] Received identification packet: 0e 01 00 00 01 01 00 01 10 01 01 01 1c 1b 3e 1b 01 01 01 30 c1 ff 40 01 46 00 01 01 00 00 00 00 00 00 00 00 01 00 00 01 00 00 02 01 01 01 aa 00 6c dc 02 00 00 00 00 00 00 00 00 00 00 00 00 00
[I] Setup finished for /dev/input/ckb1
[E] os_usbsend (via firmware.c:25):  Broken pipe, res = 0xffffffff
[W] _start_dev (device.c:43): Unable to load firmware version/poll rate
[I] Reset success
[E] os_sendindicators (usb_linux.c:258): Broken pipe
[I] Attempting reset...
[I] Reset success
[E] os_sendindicators (usb_linux.c:258): Broken pipe
[I] Attempting reset...
[I] Reset success
[E] os_sendindicators (usb_linux.c:258): Broken pipe
[I] Attempting reset...
[I] Reset success
[I] Setup finished for /dev/input/ckb2
```

This is the output of the ckb-next-daemon before sending the active command. The pulseaudio device for the headset disappeared and sound stopped working.

After sending the `active` command, it began to spam this:

```
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[I] Attempting reset...
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
[E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff
```

Finally, after killing the daemon with `^C`, the LEDs blinked (animations reset) and audio returned, and this was logged:

```

tatokis, 2018-08-23:
@Lemmmy sorry, I didn't add your void properly. git pull and try again please.

Lemmmy, 2018-08-23:
This time, connecting worked, as well as switching to software mode. Audio still worked in both hardware and software mode. All the LEDs turned off, except the infomic turned on bright red.

```
    ckb-next: Corsair RGB driver 0.3.1-48-g6a96b71
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair VOID Wireless Gaming Dongle at /dev/input/ckb2
[I] Connecting Corsair K70R Gaming Keyboard at /dev/input/ckb3
[I] Connecting Corsair Gaming SCIMITAR PRO RGB Mouse at /dev/input/ckb1
[W] _start_dev (device.c:43): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb2
[I] Starting input thread for /dev/input/ckb2
[I] Starting input thread for /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb3
[I] Setup finished for /dev/input/ckb3
[I] Received identification packet: 0e 01 00 00 01 01 00 01 10 01 01 01 1c 1b 3e 1b 01 01 01 30 c1 ff 40 01 46 00 01 01 00 00 00 00 00 00 00 00 01 00 00 01 00 00 02 01 01 01 aa 00 6c dc 02 00 00 00 00 00 00 00 00 00 00 00 00 00
[I] Setup finished for /dev/input/ckb1
```

The only thing that didn't work was changing the colours. I tried several things, such as using different colours (`ffffff`, `00ffff`, `ff00ff`, `ffff00`), and zones 0 to 4. The LEDs on the ears remained off, and the infomic LED remained red.

tatokis, 2018-08-23:
@Lemmmy can you please run `sudo usbhid-dump -d 1b1c:1b27` and paste the output?

@SethBarberee Excellent. Would you mind dropping by on IRC in the upcoming days? Need to try and figure out the EQ stuff.

scotbud123, 2018-11-16:
Any updates or news on this?

Really sucks to waste so much battery life with these stupid LEDs...and they were off before but now randomly turned on.

Using Mint 19 by the way.

ZirconiumX, 2018-11-17:
Not really; despite it being ckb-next's 2nd birthday today (happy birthday), both main developers have kinda burned out on it. Corsair's policy of not acknowledging our existence means that a lot of things are more or less opaque to us.

For example, support for K95P/Glaive/Scimitar animations has more or less stalled because of a big binary blob that we can't figure out, and unless we do, we're not touching it. EU law prevents us from being sued for reverse engineering; it does not prevent us from being sued for copyright infringement.

scotbud123, 2018-11-17:
This really sucks...have to lose so much battery life to those damn LEDs being on when I'm booted into Linux really puts a damper on everything, it's quite a downside.

freakingjohnson, 2018-12-06:
on fedora 29 this works to turn off RGB lights for wireless. However mute button and volume scroll don't work and battery indicator and mic mute light stay off as well. not a super big deal though. If anyone knows how to regain those functions/lights without enabling the RGB leds again I would be much obliged.

n01anh, 2019-03-24:
HI I have a Void Pro headset and it doesn't look like the daemon supports my model does the pro not work with this software?

Hellzbellz123, 2019-04-21:
Copyright law in the EU keeps you save from copyright stuff as long as you Reverse Engineered it

discoli, 2019-06-02:
Any update on this, would love 7.1 virtual surround sound </3

Mirppc, 2019-06-02:
i'd rather have battery info and control over the RGB.

SethBarberee, 2019-08-06:
@tatokis Sorry about the silence. just came back to my dorm for university for the year and have desktop/void again. Is it worth it to just merge the RGB now and worry about other stuff when we both can sync up time? I can easily run a test with the code with a git rebase to verify with my wired VOID and submit PR if needed.

SabbathHex, 2019-08-15:
I am using Void pro but with slightly different device ID:

    # lsusb
    ...
    Bus 002 Device 008: ID 1b1c:0a14 Corsair

I tried that 009ee19 commit with updated `usb.h`:

```
--- a/src/daemon/usb.h
+++ b/src/daemon/usb.h
@@ -115,7 +115,7 @@

 #define P_VOID_USB_1         0x1b29
 #define P_VOID_USB_2         0x1b2a
-#define P_VOID_WIRELESS_1    0x1b27
+#define P_VOID_WIRELESS_1    0x0a14
```

But this results in the headset losing audio:

```
[I] Connecting Corsair VOID PRO Wireless Gaming Headset at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[E] os_inputmain (usb_linux.c:346): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/ep_81
[E] os_inputmain (usb_linux.c:346): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.1/ep_82
[E] os_inputmain (usb_linux.c:346): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.2/ep_83
[E] os_sendindicators (usb_linux.c:258): Broken pipe
[I] Attempting reset...
```

If `echo 'active' > /dev/input/ckb1/cmd`, the daemon gets stuck in a cycle

    [E] os_usbsend (via device_keyboard.c:40):  Broken pipe, res = 0xffffffff

The messages related to `wMaxPacketSize` are weird. manually I can get it:

    # realpath ./2-3:1.3/ep_83/wMaxPacketSize
    /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.3/ep_83/wMaxPacketSize
    # cat ./2-3:1.3/ep_83/wMaxPacketSize
    0020

Should I change the ID anywhere else?

SabbathHex, 2019-08-15:
Looks like an extra line needs to be fixed, this works:
```
diff --git a/src/daemon/usb.h b/src/daemon/usb.h
index a54af72..fc3bc0b 100644
--- a/src/daemon/usb.h
+++ b/src/daemon/usb.h
@@ -115,7 +115,7 @@

 #define P_VOID_USB_1         0x1b29
 #define P_VOID_USB_2         0x1b2a
-#define P_VOID_WIRELESS_1    0x1b27
+#define P_VOID_WIRELESS_1    0x0a14
 #define IS_VOID(kb)          ((kb)->vendor == V_CORSAIR && ((kb)->product == P_VOID_USB_1 || (kb)->product == P_VOID_USB_2 || (kb)->product == P_VOID_WIRELESS_1))

 #define N_MODELS 40
@@ -215,11 +215,11 @@ const char* product_str(short product);
 #define IS_MOUSEPAD_DEV(kb)             IS_MOUSEPAD((kb)->vendor, (kb)->product)

 // Headset test
-#define IS_HEADSET(vendor, product)     ((vendor) == (V_CORSAIR) && (product == P_VOID_USB_1 || product == P_VOID_USB_2))
+#define IS_HEADSET(vendor, product)     ((vendor) == (V_CORSAIR) && (product == P_VOID_USB_1 || product == P_VOID_USB_2 || product == P_VOID_WIRELESS_1))

```
`echo 'active'` and `'idle'` works fine, but `rgb` does not work, which is totally fine

tatokis, 2019-09-01:
@SethBarberee It's been so long, I completely forgot what needs to be done. I do not have access to my desktop at home at the moment, so I can't really do much.
If you're willing, you can set up wireshark with usbpcap on windows or wireshark with usbmon and a VM, and get me packet captures of when you press/release the mute button, and URB captures of URB interrupts the void sends to the computer in response to URB controls sent to it.

That said, @discoli that's not done in hardware. It's all software, and we do not have control over the audio stream. You'll need to find something for pulseaudio if you want to get such functionality.

I am also strongly suspicious there's no DSP in the void, which means the EQ is also done in software, so we most likely can't control that either.

tatokis, 2019-09-01:
@SabbathHex did the wMaxPacketSize errors go away after adding it as a headset? Or at least, some of them?

SabbathHex, 2019-09-02:
> @SabbathHex did the wMaxPacketSize errors go away after adding it as a headset? Or at least, some of them?

All errors were gone

kevhawk2914, 2019-11-05:
@tatokis you were asking for a hid dump from the void from Lemmmy but you never received it.

I'm having the same issue with once the headset is set active all lights are off except a bright led on the mic while sound continues to work.

Heres the dump for you https://gist.github.com/kevhawk2914/49b147e27bd4169c1dc627e8e643d38a

FlashOnFire, 2019-11-15:
I would like to have the support for the void too but i dont know c++. If i can do anything that can help you implementing it, let me know.

SabbathHex, 2019-11-15:
@FlashOnFire , have you tried compiling that commit 009ee19 above? Depending on your version of Void (check `lsusb`) you might need to change a couple of values but it should work in general.

FlashOnFire, 2019-11-15:
I haven't tested, i'll try tomorrow.

FlashOnFire, 2019-11-16:
I have tested :
I had to modifiy a value with the one i found in lsusb (0a17). The void is detected in ckb but colors doesn't work....

MAPReiff, 2020-01-13:
Any update on this? I have a void pro rgb wireless and would love to disable the rgb so I could get good battery.

SethBarberee, 2020-01-13:
I've been busy with school and graduation. However, I can see what I can do to get to code to work again with my wired void (no guarantees on how fast though).

SethBarberee, 2020-01-14:
Please check out #528 and give feedback on whether daemon picks up your headset. RGB is still gonna be a while as I'm dissecting it with my wired Void.

bG9sIHVyIGhvbW8, 2021-05-06:
I know literally nothing about C but I have a Void Elite wireless, what can I do to help?

polluxau, 2025-01-24:
i see that corsair void headsets have added more information about it in kernel 6.13, would this be added as a update to supporting these headsets in this software more?

https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/commit/?h=for-6.13%2Fcorsair