ckb-next/ckb-next #82: Invisible mouse wall when gaming with CKB

Thalagyrt, 2018-02-09:
Hiya! I apologize for not having more diagnostic information as I have uninstalled the CKB driver entirely - if you need it, I will reinstall it temporarily to get it. I was using v0.2.9 from the releases pages with both a Corsair Scimitar and Corsair M65 and running into issues while gaming.

When playing a game that requires click + drag or right click + drag as in many shooters, the OSX cursor moves, hidden, while running the CKB driver. Eventually it hits the edge of the screen and cannot move any further, and the visual result is that in game you hit an invisible wall where you can't turn anymore. For example, while playing Skyrim via Steam Streaming, I would try shooting a bow only to find that I can't track my target because the cursor is hitting an invisible wall past which it cannot move. This only happens with a button pressed down, and happens in Mac native games, happens in Geforce Now, and also in Steam Streaming.

With CKB uninstalled and using the native OSX mouse drivers, the problem disappears entirely and everything behaves as expected. The problem occurs whenever ckb-daemon is running, whether or not CKB the app is running or not, which initially led me to believe it was an OSX problem since I thought I had completely exited CKB. When I found nobody else on any forums with the same issue, I suspected the mouse driver and looked into how to completely remove it and the problem went away completely.

If there's any way I can reinstall temporarily and get you some logs/extra info I'd be happy to! If you want an easy target to test this with on your own machine, Golf With Your Friends is $6 on Steam and is Mac native and exhibits the problem with CKB as you have to click and drag forward to make a shot. The invisible wall effect leads to you being unable to shoot at full power in many situations.

Thanks so much for all the awesome work on this otherwise phenomenal driver, and I look forward to being able to use it again!

ZirconiumX, 2018-02-09:
I distinctly remember somebody else posting about this bug, I'll see if I can find it.

Thalagyrt, 2018-02-09:
Wow, that was a fast reply! I'm happy to do as much testing as you want. Hopefully we can find a fix. :)

Thalagyrt, 2018-02-09:
Also if you don't have any games to test with, I'll happily gift you one of the ones that had the problem on Steam so you can test as I don't feel you should have to pay out of pocket to fix this. You do enough as is maintaining it.

ZirconiumX, 2018-02-09:
Actually, the problem is a bit more serious than a game that costs a couple of bucks; only @fleischie has a Mac - and a job - and @light2yellow has a Hackintosh, so we're a bit limited in terms of hardware and reproducibility.

Anyway, let us continue this in #23.

closed 2018-02-09 by ZirconiumX