ckb-next/ckb-next #92: K95 platinum profile switch key

delooper, 2018-02-14:
Greetings, 

I acquired a K95 platinum this morning and was happy to discover ckb-next.  I'm running Ubuntu 16.04, and installed ckb-next via the git repo.  The only feature of the K95 I know of that I have not been able to use is the "profile switch key".  Is this presently not supported in ckb-next?  Is there anything I could do to help?  

I don't have much experience writing drivers but I do have quite a bit of coding and debugging experience. 

Thanks for all the effort! It's much appreciated.

-ryan

ZirconiumX, 2018-02-14:
Would you mind clarifying what you mean by "profile switch key"? Do you
mean the MR button?

tatokis, 2018-02-14:
@delooper We should perhaps try and get it renamed to something more obvious then, however have you tried binding it?
[image]

ZirconiumX, 2018-02-14:
> In the manual it's called the "Profile Switch Key" and on the keyboard it
> looks like a cartoon figure of a head on a torso, just above the F1 and F2
> keys. In the manual it's not called an "MR key". . . the K95 manual does
> not refer to a "MR key".
>
> The K95 platinum has memory for three stored-on-the-keyboard profiles and
> pressing the profile switch key is supposed to cycle through the three
> profiles. But with the current ckb-next it does not appear to do anything.
> It's unclear if the profile switch key is deactivated or if there's only
> the one profile in memory (and it's cycling through a list of one).
>
> I imagine ckb-next has a way of logging all keyboard activity and I could
> perhaps capture a keypress. I'll see if I can do that.

Right, hardware animations, like on the K95 Platinum, are currently
unsupported because we don't have the infrastructure in place for it. To do
so would require us to rewrite the animation API, which assumes real-time
software control, and right now that's fairly low priority.

ckb-next doesn't really have any packet capture facilities; we tend to use
wireshark. I believe the daemon will print a warning when it receives an
unknown keypress though.

delooper, 2018-02-14:
Ah, thank you!  Yes, in that menu it would be the MR key.  

Apologies for being new to this. 

I tried binding the MR key as "Switch to mode (next) wrap-around".  After applying and "save to hardware" the MR key continues to do nothing.  I have 4 rather different modes in the  main (left side of window) list.

tatokis, 2018-02-14:
@delooper First of all, Save to hardware isn't supported for your model, so please do not use it.
Second, as @ZirconiumX said, the daemon prints a warning when it receives an invalid keypress.
Please run ckb-dev-detect, which will collect any output from it, and upload it here.

ZirconiumX, 2018-02-14:
"Save To Hardware" on the K95 Platinum, and on v3 firmwares, is currently
broken. It requires a fix for switching message types, but I think there's
currently a pull request for that.

delooper, 2018-02-14:
Thanks for all the comments.  Here is the output of ckb-dev-detect in case it's helpful -- perhaps it's redundant at this point. 

[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1723310/ckb-dev-detect-report.gz)

tatokis, 2018-02-14:
@delooper Can you control the LEDs on that button using ckb?
Run `cat /dev/input/ckb1/notify1` and press the button. It should tell you what it thinks it is.

delooper, 2018-02-14:
No, it looks like I can't.   

When I press the MR key the notify1 file produces:

key +#70
key -#70

ZirconiumX, 2018-02-14:
Hmmm, looks like we need to add that key to the daemon.

tatokis, 2018-02-16:
Closed by https://github.com/ckb-next/ckb-next/commit/d79bf1880a5b4f6af60395a33d69e9435374b682

closed 2018-02-16 by tatokis

delooper, 2018-02-16:
Thanks for all the help.  Glad this is resolved now.