ckb-next/ckb-next #95: K95 Platinum animation support

ZirconiumX, 2018-02-14:
Soon save to hardware for the K95 Platinum [will be disabled](https://github.com/ckb-next/ckb-next/pull/93) due to Corsair removing the static single frame save we had previously.

We should probably investigate how to do this.

delooper, 2018-02-15:
Let me know if there might be something I could do to help.

ZirconiumX, 2018-02-15:
We actually have two people with K95 Platinums; yourself and @jerryaldrichiii, who helped us add support for it in the first place. You're welcome to join us all in IRC; we'd be happy to show you the ropes.

ZirconiumX, 2018-03-07:
So, progress update for those curious. I've been working through some packet captures @jerryaldrichiii sent me of saving hardware animations.

The K95 Platinum seems to have almost a filesystem on its flash, and the hardware animations consist of writing and reading from this filesystem.

The files I've managed to find and capture are the following:
`lightcnt.cnt`: This was a blank file in the static animations, making me wonder if this is for playback or something.
`PROFILE.I`: This seems to be the closest analogue of the previous single frame I could find. It contains the profile GUID followed by its UTF-16LE name.
`PROFILE.ZIP`: This seems to be CUE metadata for a profile. It's a Deflate compressed XML file, and I'm going to hazard a guess that they didn't put decoders for those formats on the firmware.
`PROFILE.DAT`: This was a four-byte word that makes little sense at the moment.
`PROFILE.MAP`: This was also a four-byte word that makes no sense.

I'm going to need packet captures of saving dynamic animations to decode the latter two, I suspect.

As for the protocol, Corsair have introduced field 0x17 for hardware animation access, and I've been compiling a list of what these do.

`07 17 05` - This is followed by a filename and appears to mark the start of a file write.
`07 17 07` - This is also followed by a filename and appears to mark the start of a file read.
`07 17 08` - This seems to mark the end of a file read/write
`07 17 09` - This is for writing a file to the flash, and follows up to 5 `7f` payload packets.
`07 17 0c 0X` - This switches the active animation to X (0-2).
`0e 17 0d` - Instead of [waiting arbitrary amounts](https://github.com/ckb-next/ckb-next/blob/master/src/ckb-daemon/firmware.c#L122), CUE uses a synchronisation packet to signal when to move onto the next series of packets.

I'm still working on some other packets though, like `0e 17 03` and `0e 17 0b`, which I think might be different sync packets; `07 17 0a`, which might be a "prepare to read" command; and `07 17 0e`, which I have no clue about.

delooper, 2018-03-07:
I ?might? be able to help with the packet capture in the future.  VirtualBox has been a little buggy since the meltdown patches have arrived in Ubuntu, but it's mostly working now.  For most tasks my Windows 10 client appears to be stable.  Haven't tried wireshark in it yet.

ZirconiumX, 2018-03-07:
I tend to recommend QEMU/KVM, but as long as you can get USB passthrough running, I'll be able to show you what to do. Wireshark can be run outside of the VM, through use of the Linux `usbmon` module.

ZirconiumX, 2018-03-08:
Update two: Jerry sent me another capture of saving all-green to the K95P.

There are now some new files to look at:
`light_00.d`: Another helpfully named (sarcasm) file, which is almost entirely blank. No idea.
`light_00.k`: This is a reasonably short file filled with binary. I wonder if it's another compressed XML file.
`light_00.r`: This is just repeating `00 ff` sequences. It's strange.

And `lightcnt.cnt` is now 1; the other files weren't touched at all.

delooper, 2018-03-09:
Thanks, my understanding is QEMU/KVM is virtualization software.  I believe I already have USB passthrough working using the guest extnensions of VirtualBox (took some fussy business playing with options -- the graphics accelleration was freezing my system).  

My host system is Linux and my guest (I'm assuming) should be Windows to run the Corsair software.  It sounds like you're saying I can use usbmon in Linux rather than Wireshark in Windows to capture the USB packets?

ZirconiumX, 2018-03-09:
>Thanks, my understanding is QEMU/KVM is virtualization software.
So is VirtualBox.

>I believe I already have USB passthrough working using the guest extnensions of VirtualBox (took some fussy business playing with options -- the graphics accelleration was freezing my system).
Yeah, you need USB passthrough for CUE to even get your device.

>My host system is Linux and my guest (I'm assuming) should be Windows to run the Corsair software.
Yeah, CUE runs on Windows.

>It sounds like you're saying I can use usbmon in Linux rather than Wireshark in Windows to capture the USB packets?
usbmon on Linux is equivalent to USBPCap on Windows; both are used as backends by Wireshark, and any USB traffic between the keyboard and the VM would be picked up by usbmon.

ZirconiumX, 2018-03-09:
Feel free to drop by IRC by the way. It'd be much faster.

delooper, 2018-03-09:
Will stop by on IRC as soon as I have some real time to put into this.  Right now I'm behind on too many projects. . .

ZirconiumX, 2018-03-14:
Update three: I have been working on the k95p-profile branch, trying to get at least feature parity with other keyboards. Jerry has sent me captures of interactive animations.

`light_00.d` I think might be a bitmap of what the animation is using.
`light_00.k` could be a binary blob of how to animate. @ScoreUnder pointed out that it was a very regular file, so it could be almost anything.

raparkhurst, 2018-04-16:
Do you guys need help with this still?  I've been looking at getting the K95 platinum and wouldn't mind helping.

ZirconiumX, 2018-04-16:
I have all the pieces of the puzzle. But putting them together in something that works is very difficult, because it's so completely unlike anything else Corsair have done before (or after).

It requires synthesizing and uploading files to the keyboard, and I don't know enough about the file formats to feel safe uploading one to somebody's keyboard in case we break things.

NoorulainKhaliq, 2018-08-31:
Any updates on k95P yet?

ZirconiumX, 2018-09-01:
A few, yeah. I'll eat dinner and then go over what I've found.

NoorulainKhaliq, 2018-09-01:
Okay, thanks for the update. 
So that means that importing cue profiles is likely not an option either and k95P owners are likely stuck with default profiles for mac.

ZirconiumX, 2018-09-01:
I'm sure I could give it another go if I bought a Scimitar or something.

NoorulainKhaliq, 2018-09-01:
That would be great! Thank you very much!

ZirconiumX, 2018-09-01:
First I need the money for a Scimitar.

tatokis, 2018-09-01:
@NoorulainKhaliq I should probably mention, this doesn't include importing CUE profiles. This is strictly writing mappings and led data to the keyboard itself for when there's no software running on the computer.

NoorulainKhaliq, 2018-09-04:
@tatokis , what do you mean when you say there's is no software running on the computer?

I'm hoping that at some point we can eventually import CUE profiles. If that seems like a possibility, i don't mind giving the money/Scimitar to @ZirconiumX to try out.

ZirconiumX, 2018-09-04:
CUE and CKB use largely incompatible formats, and converting between them
is nontrivial. But possible.

If anything, the person you need to pay for profile conversion is @tatokis,
since he wrote the CKB profile code.

tatokis, 2018-09-04:
@NoorulainKhaliq if ckb-next isn't running on your computer, your keyboard switches to the profile stored on it. This PR is about saving to the memory on your keyboard.

NoorulainKhaliq, 2018-09-04:
@ZirconiumX , I see. On the K95, i think you can only save 3 profiles at a time, and not profiles that are CUE profiles that have been imported. 

Being able to import CUE profiles for awesome. Is there a branch for importing cue profiles that I can be directed to? That's something that interests me. thank you.

NoorulainKhaliq, 2018-09-04:
@tatokis , I sent you an email. 
Let's talk about what needs to be done to import cue-profiles to ckb-next.