ckb-next/ckb-next #97: ckb consumes up to 100% of the CPU

sashko, 2018-02-16:
# Devices information

[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1731342/ckb-dev-detect-report.gz)

---

# Bug report

- [x] Yes
- [ ] No

## General information

### Source
e6f3ff446a6899be09d2391cd46573285a1f3360

### Branch
master

## What is the program's current behavior?
ckb always consumes 20-100% of CPU.

[image]

[image]

---

# Something else

I would be more than happy profiling it using Valgrind or whatever you want if needed.

ZirconiumX, 2018-02-16:
... Huh. What're your hardware specs?

Our usual suggestions for this kind of thing are to lower the animation
frame rate and to disable "show animated" in the GUI.

sashko, 2018-02-16:
@ZirconiumXi7-6560U and 16Gb RAM

sashko, 2018-02-16:
i7-6560U and 16Gb RAM.

I have just played with different animation profiles and do not seem to make any difference: always get 20-40% CPU usage, which is crazy!

ZirconiumX, 2018-02-16:
Did you try disabling "show animated"? With that turned off, I can use CKB even on a raspberry pi.

sashko, 2018-02-16:
Disabling "show animated" seems to drop average CPU consumption by about 25%, which is not nearly enough.

ZirconiumX, 2018-02-16:
Next, go into "more settings" and lower the frame rate, if it's above about
30fps.

tatokis, 2018-02-16:
@sashko If you try a profile without any animations, do you still get high CPU usage? This could be a bug.

sashko, 2018-02-16:
It was set to 30. Lowering it to 10 dropped CPU usage to ~8%.

sashko, 2018-02-16:
@tatokis profile without animation and FPS = 30 has ~15% CPU usage. Lowering FPS to 10 drops it to 5-7%

ZirconiumX, 2018-02-16:
I think you should drop by IRC; I have a hunch here, but we'll be able to
test it faster there, rather than in GitHub.

ZirconiumX, 2018-02-17:
I'm tagging it as this, because even though @sashko can reproduce it, I think this might be meltdown related, and so we would need a Skylake system to verify.

light2yellow, 2018-02-17:
30 FPS, spatial dithering disabled, "show animated" disabled - rainbow takes ~2% on an old i3. So, can't reproduce either. The daemon takes ~5% for me (combined), it looks same for you.
[image]
[image]

sashko, 2018-02-17:
@light2yellow you're checking the wrong process: it's not the ckb-deamon that is consuming CPU but the ckb app

tatokis, 2018-02-17:
@sashko top one is the GUI.
I can't reproduce on an i7-4600U either, getting around the same usage as @light2yellow, fresh from master.

ZirconiumX, 2018-02-18:
So, @tatokis noticed that the GUI sends even blank frames to the daemon, which would result in a lot of syscalls, which are noticeably more expensive on Skylake upward. As a long-term guide, we should detect and avoid sending blank frames to the daemon, and likewise avoid sending duplicate frames, which could perhaps kill two birds with one stone.

sashko, 2018-02-18:
Glad it has helped you find a bottleneck (hopefully the one I've bumped into).

Just curious: why process name is 'ckb' in my case and 'ckb-next' in @light2yellow's case?

tatokis, 2018-02-18:
@sashko Sadly I don't think much can be done in your case.
Also, see #65

closed 2022-06-22 by sashko