ckb-next/ckb-next #99: K70 shift key is "stuck" when using it to select multiple items (MacOS High Sierra)

arclightcreative, 2018-02-17:
Not sure if this is related to other modifier key issues, but this problem has become wildly infuriating.

I've been having this intermittent issue where in a Finder window, I hit shift + click to select multiple items. For some reason when I go to click away from the selected items, the shift key acts as if it's still pressed down, and won't allow a normal click until I hit Shift again. It's almost as if the modifier is "stuck" and has to be toggled off like a Caps Lock key. Not a mechanical issue, as this has happened on two separate Corsair keyboards.

I've tried every troubleshooting step I can think of, and nothing has worked. At this point i'd much rather disable the modifier keys in CKB and use Karabiner to manage them (and just use CKB for light management). Any help thrown my way would be greatly appreciated, as I REALLY want to avoid having to return a second Corsair keyboard in one week, just to go with a Logitech or SteelSeries. :)

ZirconiumX, 2018-02-17:
Right, I suggest you join our IRC channel, since we need to confirm if this
still happens with master. To do that, you need to build from source.

light2yellow, 2018-02-17:
> At this point i'd much rather disable the modifier keys in CKB and use Karabiner to manage them (and just use CKB for light management).

Note that it's impossible to use both at the same time. You first have to stop ckb-daemon to use karabiner's one, and vice versa.

ZirconiumX, 2018-03-08:
Is this bug still occurring? If not, please close the issue.

tatokis, 2018-07-19:
Closing due to inactivity. Chances are it's fixed by karabiner.

closed 2018-07-19 by tatokis