ckb-next/ckb-next #101: Dark Core Support

mrv1k, 2018-02-18:
Hey, thank you very much for maintaining this repo. You're doing a great job.

Could be coupled with their new MM1000 mousepad but I think that's overkill.

# New device support request

- [x] Yes
- [ ] No

[DarkCoreWireless-report.gz](https://github.com/ckb-next/ckb-next/files/1734371/DarkCoreWireless-report.gz)

Thank you.

ZirconiumX, 2018-02-18:
Right, you'll need to do a few things. First off, we also need a ckb-dev-detect while the Dark Core is plugged in. Secondly, you'll need to join us in IRC. In the meantime, I'll try to merge my dark-core-product-id branch.

mrv1k, 2018-02-19:
Didn't realize it would produce different results based on connection.

Receiver only: [Receiver only.gz](https://github.com/ckb-next/ckb-next/files/1735289/Receiver.only.gz)

Wired only: [Wired only.gz](https://github.com/ckb-next/ckb-next/files/1735287/Wired.only.gz)

Both receiver and cable: [Wired+Plugged.gz](https://github.com/ckb-next/ckb-next/files/1735285/Wired.Plugged.gz)

Also, I own Void Pro RGB wireless and on Tuesday will become a proud owner of K70 LUX so if you need results of those devices let me know.

I'll be able to join IRC on Tuesday and after that, do you need my email to send an invite?

ZirconiumX, 2018-02-19:
```diff
--- Wired.only          2018-02-19 00:18:19.000000000 +0000
+++ Receiver.only       2018-02-19 00:19:18.000000000 +0000
@@ -11,7 +11,7 @@
 USB Devices:
 Couldn't open device, some information will be missing

-Bus 001 Device 006: ID 1b1c:1b35 Corsair
+Bus 001 Device 007: ID 1b1c:1b64 Corsair
 Device Descriptor:
   bLength                18
   bDescriptorType         1
@@ -21,7 +21,7 @@
   bDeviceProtocol         0
   bMaxPacketSize0        64
   idVendor           0x1b1c Corsair
-  idProduct          0x1b35
+  idProduct          0x1b64
   bcdDevice            0.33
   iManufacturer           1
   iProduct                2
```

These are the IDs I was looking for.

The Void we're looking into, but you could put them in #81, and I can see if it helps any.

ZirconiumX, 2018-02-24:
We're mostly waiting on you to join IRC. I can set up a PR that I *think* will work, but the only way to know for sure is for you to test it.

tatokis, 2018-02-24:
@ZirconiumX May as well make a PR and mark it as awaiting testing. Will be easier for people dropping by to have ready to test code, as it's clear we aren't getting a response.

mrv1k, 2018-02-24:
I forgot sorry about that. I'll be home in about 2 hours and joint IRC then.

mrv1k, 2018-02-24:
Plans changed and I'm not gonna be home today. I'll be on tomorrow. Not
sure about time though. I created an issue I'll help you resolve it alright.

ZirconiumX, 2018-03-01:
So, thank you for the packets you've given us so far.

For tomorrow, it'd be nice to get the following done:

On Windows, capture CUE's colour change by changing static colours from red to green to blue, which should give us a start as to how to get colour changes working.

On Linux, `git clone https://github.com/ZirconiumX/corsair-re; cd corsair-re/src; systemctl stop ckb-daemon; ./run_me.bash` and upload output.tgz.

mrv1k, 2018-03-16:
Hey guys, sorry I went rogue on you. I'm still busy but I'll make sure to record the data this month.
 I'll hop on IRC and post results there.
✌️

ZirconiumX, 2018-03-16:
Sure thing. I've been working on K95 Platinum support in the meantime;
though I have heavy suspicions that they use different animation formats.

We also think they changed the firmware update protocol, so we'll need a
packet capture of that, too. On past wired devices they've had fallback
systems that have stopped the device from being completely bricked, but
there *is* the possibility all the same.

mrv1k, 2018-03-27:
All of the data is in IRC. Good luck guys, lemme know if you need anything else.

tatokis, 2018-03-27:
@mrv1k Thank you very much for that. Much appreciated

DewaldDeJager, 2018-10-05:
Has this feature been completed? I would also like to see support for the Dark Core. I do not have the SE version.

ZirconiumX, 2018-10-05:
We have "support" in the wireless branches.

However, we've both burned out, and until we recover you'll have to wait.

DewaldDeJager, 2018-10-05:
@ZirconiumX So is that support not in master? I am installing the `ckb-next-git` package from the AUR.

Is there any way I can contribute?

afreepenguin, 2018-11-12:
I have this mouse and some coding skills. If you can point me in the right place I may be able to assist.

droidmahn, 2018-12-24:
Just subscribing out of interest and to stay updated. I own the mouse as of today if you require anything.

DarkLogicX, 2018-12-30:
I'm interested in this too.
Though I'm only interested in it as a wired mouse.
The MM2000 support would be nice though, but not sure if I'd use that.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/2716827/ckb-next-dev-detect-report.gz)

SleepyBoyy, 2019-02-17:
Obviously I am respecting burnouts.
But I just switched from a Glaive to a Dark Core (no SE) and I hate it to see no driver for linux distros.
I'd really like to use Ubuntu with my setup and, after some time, I fixed every error I got from no 7.1 audio to chipset errors.
I really hate that there are no official drivers from corsair for the secondly most used OS in the whole desktop area.
I love you for trying to do what Corsair doesn't want to do.
So are there any plans on supporting the Dark Core in the future?
I'd really appreciate it.
Keep up your nice work and get through your burnout!

ZirconiumX, 2019-02-17:
So, to get *some* support for the Dark Core, you can use the wireless
branch. At the moment it seems to work okay in wired mode and (if you edit
usb.c) wireless mode. However, it gets confused when you have the Dark Core
plugged in along with the dongle.

Equally, the K63 WL - last time it was tested - needs some more work.

mdPlusPlus, 2019-03-22:
So, I also own an Corsair Dark Core RGB (not SE) and have access to Windows and Ubuntu systems.

If there's anything to help you guys get this going, I'd be glad to help.

P.S.: Is the K66 support complete (wireless, donge, encryption)? Because I own that too and could provide logs if needed.

ximiii, 2019-04-07:
What is it going to take to get the dark core mouse to work with ckb-next. Id like to help

rbh00723, 2019-04-16:
I too have win 10 access and Pop OS (ubuntu) and I'd really like to use the forward and back buttons on the side of my mouse. Please send help!!!

tatokis, 2019-04-16:
It will be a bit until we get proper dark core support unfortunately. I am busy with a bunch of other things.

That said, you should be able to bind your side buttons using something like xbindkeys

ivan0013, 2019-07-31:
> So, to get *some* support for the Dark Core, you can use the wireless branch. At the moment it seems to work okay in wired mode and (if you edit usb.c) wireless mode. However, it gets confused when [...]

Can you refer to some documentation about editing usb.c?
thanks for your work

tatokis, 2019-09-01:
If you are still interested, I have rebased the wireless branch and made the changes.
Just don't have both wired and wireless plugged in at the same time

CJ1169, 2019-09-15:
After about 24hrs of use, the wireless branch seems good. While using the wired and wireless at the same time, I don't actually have unexpected issues. Granted, both profiles I've set to be the same.
Using K63 wired/wireless and Dark Core Wired/Wireless.
Things I've tested: Colours, Performance, Wired/Wireless (w/ same profile)
Things I haven't tested: Bindings, individual lighting effects
Things that don't seem to work, but I haven't tested: poll rate changes, lighting effects (i.e. rainbow or breathing)
For Arch/AUR users, install the `ckb-next-git` package and edit the `PKGBUILD` source to `source=('ckb-next-git::git+https://github.com/ckb-next/ckb-next.git#branch=wireless')`
Let me know if you'd like any log files or if I could help further.

olielvewen, 2019-11-07:
@tatokis 

Could you rebase the wireless branch ? By Advance Thanks a lot.

tatokis, 2019-11-12:
I just rebased the branch

olielvewen, 2019-11-12:
@tatokis 

Thanks a lot. I have been able to update the wireless branch on Manjaro.

When do you think to merge this branch in the main/master ?
What are the factors blocking this operation? What to do to help you ?

tigermick, 2020-02-20:
Hi,
I have a dark core rgb mouse too et ckb-next doesn't find it.
How to proceed to have ckb-next works with my mouse please ?

theslimshaney, 2020-02-25:
@tatokis any plans?

anolting, 2021-06-15:
Hi, I would like to share my report as my dark core has a different id compared to the example above. The report was created while the mice was wired.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/6659003/ckb-next-dev-detect-report.gz)

olokos, 2021-06-29:
I can confirm the issue.
Dark Core RGB Pro Se is not detected at all by the ckb-next under ubuntu.

I have tested it with:
version 4.2 I had earlier,
4.4 from ppa for focal
4.4 git from ppa for focal
4.4 git clone ckb-next, built from source

ultimately ckb-next git ppa for focal seem to report same version `ckb-next 0.4.4-65-g2b824fd`
as the one built from source, including the latest commit 2b824fde876833b2285cb339e404f34a7226c0b4
that also reports the `ckb-next 0.4.4-65-g2b824fd` version in the UI.

I think it proves the point that the CKB Next doesn't support it yet.

[ckb-next-report-1624998268.gz](https://github.com/ckb-next/ckb-next/files/6736270/ckb-next-report-1624998268.gz)
[ckb-next-report-1624997060.gz](https://github.com/ckb-next/ckb-next/files/6736271/ckb-next-report-1624997060.gz)
[ckb-next-report-1624996631.gz](https://github.com/ckb-next/ckb-next/files/6736272/ckb-next-report-1624996631.gz)

The 3 reports are in order:
1. built from source commit 2b824fde876833b2285cb339e404f34a7226c0b4
2. downloaded using ckb-next git ppa
3. probably 4.2 version of ckb-next installed from source

so 1 is the latest, 2 is similar but ppa, 3 is the oldest report
all of them I did today.

olokos, 2021-06-29:
Opened #728 
in order to help solving this issue.

Useful info:
Dark Core RGB Pro SE which I have is similar in protocol to the new Ironclaw Wireless, instead of Dark Core