ckb-next/ckb-next #102: Feral Interactive games wont start with ckb-next [Bug report]

unelsson, 2018-02-19:
"Segfault 11" results when running games like DiRT Rally or Total War Shogun 2. I'm Running Linux Manjaro 4.9.81-1-MANJARO x86_64 GNU/Linux. Manjaro is based on Arch linux. My keyboard is Corsair Strafe RGB. Problem is solved by disabling ckb-next daemon.

I don't know why this happens, and whether or not this is more of Feral's problem or a bug in ckb-next. I'll report this to feral interactive too.

tatokis, 2018-02-19:
@unelsson Unfortunately, this one is probably on their end.
Assuming you have the udev rule that marks ckb-daemon devices as non-joysticks, there is probably not much we can do.

Would you mind linking to the feral bug report so that we can follow it?

I might install Tomb Raider and try it, but again, I doubt much can be done.

ZirconiumX, 2018-03-08:
Should we close this? We have no upstream bug to track, and we cant do anything on our end.

closed 2018-04-09 by tatokis