ckb-next/ckb-next #104: Corsair M65 Pro mouse no longer detected as a mouse

ThoreauHenry, 2018-02-22:
Hello,

I have a K95 Platinum RGB keyboard and a M65 Pro mouse.  I recently(today) grabbed the git clone and installed the driver to update the code.  I have been using the driver for half a year now without seeing this issue before. But, enough of the pleasantries.

The M65 mouse is now being listed as a configurable keyboard, and the mouse itself works but there are no more lights.  It's dead stick, as it should be, since the driver thinks it's a keyboard.

I have completely removed and reinstalled the driver as per the README.  It is repeatable. 

Here are my system specs.

https://pastebin.com/AFDQ8HHh

Here are the usb specs.

https://pastebin.com/F3EbYtD2

This is the bug report, before the uninstall.

[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1746387/ckb-dev-detect-report.gz)

Here is the bug report after the install(again).

[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1746413/ckb-dev-detect-report.gz)

Here is what it looks like via gui:

https://imgur.com/a/QBSpC

Very exciting.  I would take a picture of the mouse, but it's dark.  You can't see it.  Thanks for your time.

tatokis, 2018-02-22:
I don't know how this happened, but it's hilarious. 

With the driver running, can you please run the following commands in a terminal and post their output? 

cat /dev/input/ckb1/features 
cat /dev/input/ckb2/featured 

Thanks for reporting, and apologies for this.

ThoreauHenry, 2018-02-22:
$ cat /dev/input/ckb1/features

corsair k95p rgb bind notify fwversion fwupdate

$ cat /dev/input/ckb2/features

corsair k68 rgb pollrate adjrate bind notify fwversion fwupdate hwload

oddalot, 2018-02-22:
I'm having the same problem. My M65 Pro no longer works. I need to get my LED game on with the quickness - it's dark and lonely on the right side of my desk.

tatokis, 2018-02-22:
This was embarrassing. I pushed a fix, please get the latest code from master and report if it has been fixed.

closed 2018-02-22 by tatokis

ThoreauHenry, 2018-02-22:
This has been fixed.  Thank you.

https://i.imgur.com/rXPNmVM.jpg

commit 5edc7af2 (2018-02-22): Correct regression introduced by #72

Devices would be recognised as K68 due to a wrong comparison.
Closes #104