ckb-next/ckb-next #106: Random key freezes and missing out keypresses

joveice, 2018-02-23:
Keyboard K63
Mouse HARPOON

Can't upload dev-detect-report due to strict network policy.

# Bug report

- [x] Yes
- [ ] No

## General information

### Source
built manually using quickinstall script

### Branch
master

## What is the program's current behavior?
Keyboard randomly freezes on keys and starts spamming them.
Keyboard misses presses and leaves out keypresses.

## What is the expected behavior of the program?
Not whats listed above

## Any other notes & comments?
Works without driver, but needed it for the mouse buttons

ZirconiumX, 2018-02-23:
You really do need to give us the ckb-dev-detect. Try sending it when you're at home or something.

corporategoth, 2018-03-16:
I have the same issue, if I run CKB (or have it running in the background), I get the keyboard repeating keys or dropping keypresses (a lot).  CKB needs to be running (even in the background) for the keyboard to perform any animation.

`ckb-dev-detect - Fri Mar 16 00:59:10 EDT 2018

Kernel: Linux temple 4.15.9-1-ARCH #1 SMP PREEMPT Sun Mar 11 17:54:33 UTC 2018 x86_64 GNU/Linux

OS Version:
LSB Version:	1.4
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

USB Devices:

Bus 006 Device 003: ID 1b1c:1b2e Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b2e 
  bcdDevice            2.06
  iManufacturer           1 Corsair
  iProduct                2 Corsair Gaming M65 Pro RGB Mouse
  iSerial                 3 16013012AEBA044655DE9929F5001946
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      58
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
[... 275 lines cut ...]
Corsair Gaming K70 RGB RAPIDFIRE Keyboar
/dev/input/ckb1/notify0
/dev/input/ckb1/cmd
/dev/input/ckb2
/dev/input/ckb2/notify4
/dev/input/ckb2/notify3
/dev/input/ckb2/notify2
/dev/input/ckb2/notify1
/dev/input/ckb2/pollrate
1 ms
/dev/input/ckb2/fwversion
0206
/dev/input/ckb2/features
corsair m65 rgb pollrate adjrate bind notify fwversion fwupdate
/dev/input/ckb2/productid
1b2e
/dev/input/ckb2/serial
16013012AEBA044655DE9929F5001946
/dev/input/ckb2/model
Corsair Gaming M65 Pro RGB Mouse
/dev/input/ckb2/notify0
/dev/input/ckb2/cmd

Environment:
`

joveice, 2018-07-19:
Here @ZirconiumX

```
ckb-next-dev-detect - Thu Jul 19 10:39:15 CEST 2018

Kernel: Linux <removed> 4.13.9-300.fc27.x86_64 #1 SMP Mon Oct 23 13:41:58 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 27 (Twenty Seven)
Release:	27
Codename:	TwentySeven

USB Devices:
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing

Bus 001 Device 005: ID 1b1c:1b3c Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b3c 
  bcdDevice            1.01
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      58
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
[... 271 lines cut ...]
/dev/input/ckb2/pollrate
1 ms
/dev/input/ckb2/fwversion
0101
/dev/input/ckb2/features
corsair harpoon rgb pollrate adjrate bind notify fwversion fwupdate hwload
/dev/input/ckb2/layout

/dev/input/ckb2/productid
1b3c
/dev/input/ckb2/serial
0900F012AF090C0358B475E6F5001941
/dev/input/ckb2/model
Corsair Gaming HARPOON RGB Mouse
/dev/input/ckb2/notify0
/dev/input/ckb2/cmd

Environment:
XDG_SESSION_DESKTOP=gnome
QT_IM_MODULE=ibus
XDG_CURRENT_DESKTOP=GNOME

Kernel command line:
BOOT_IMAGE=/vmlinuz-4.13.9-300.fc27.x86_64 root=/dev/mapper/fedora-root ro rd.lvm.lv=fedora/root rd.luks.uuid=luks-442ba304-522c-4258-a1af-3c85857ebe84 rd.lvm.lv=fedora/swap rhgb quiet LANG=en_GB.UTF-8
```

ZirconiumX, 2018-07-19:
Can you try some of the older releases like 0.2.9 and 0.2.8 and see if you can find a version which worked?

joveice, 2018-07-19:
Now running 0.2.9. Shame I can't change the lights as it just crashes. But I managed to turn them off which is fine by me, all I want is my mouse back and fourth buttons to work. (The only reason I'm using this)

joveice, 2018-07-19:
Still got the same issue, now trying 0.2.8. Is there a way I can get only the stuff for the mouse? or have it ignore the keyboard in any ways?

ZirconiumX, 2018-07-19:
Hmmm, this sounds almost a hardware issue, honestly.

To test, you'll need to enable USB debugging mode, which requires  `cmake -DDEBUG_USB_INPUT=1` when you're building it, and then you'll need to send us the log (keep in mind that it's effectively keylogging, so don't type anything sensitive,  but just try to reproduce the issue). You might need to join us in IRC for this.

joveice, 2018-07-19:
Works 100% fine when I'm not using the driver (none / uninstalled this). And it works 100% fine on my Windows PC. (I'm working at 2 PC's with a KVM switch).

I used the quickinstall, how should I do this? 

I can't join a IRC, still on a really strict network.

gravityfargo, 2023-09-18:
Stale.

closed 2023-09-18 by gravityfargo