ckb-next/ckb-next #109: N_KEYS_HW + 7 is not divisible by 8

tatokis, 2018-02-25:
Just came across this while working on the mousepad code.
MSC's comments say that N_KEYS_HW in keymap.h should be divisible by 8, which it is.
I assume that the comment was added there due to N_KEYBYTES_HW dividing it.

However, the existing macro adds 7 to it, and the result is not divisible by 8.

One of their last commits to that file has N_KEYS_HW set to what it is now, so it is not a regression from when we took over.
https://github.com/ckb-next/ckb-next/blob/7eaf1ae387b2d77d9282ee9061f495535515ddd8/src/ckb-daemon/keymap.h#L23-L25

This should be investigated at some point, as N_KEYS_HW by itself is divisible, without having to add anything.

anall, 2018-03-16:
I'd imagine that is intentional and is a trick to round `N_KEYBYTES_HW` up to have enough bytes to fit `N_KEYS_HW` bits, just in case `N_KEYS_HW` **isn't** divisible by 8.

tatokis, 2018-03-16:
Yes, I know why it's done. 
The issue is that the result, after adding 7 isn't divisible by 8, and whether that affects anything or not.

nonchip, 2018-03-17:
> I think it should either be +8 or +0 in this case.

`(X+Y-1)/Y` is the correct (positive) integer arithmetic to "divide X byY while rounding up" though, so the code is right.

also the comment specifies the `N_KEYS_HW` constant should be divisible by 8, not that everything in (or even the evaluated value of) the `N_KEYBYTES_HW` macro is.

tatokis, 2018-03-17:
@nonchip That makes sense.

Initially I misread the comment, without realising that it's rounding, which is what led me to opening this.
At least it's cleared up. Thanks!

tatokis, 2018-03-17:
@anall Yeah, sorry. Thanks for the response too.

closed 2018-03-17 by tatokis