ckb-next/ckb-next #111: Unable to type a special character after ckb-next install

Kallestofeles, 2018-02-28:
Greetings, a strange issue has occurred. Namely, after installing ckb-next and starting the daemon, I am unable to type a special character which is frequently used in Estonian - Ä. I can produce an uppercase Ä with shift+Ä (Swedish layout, the second button to the right from "L"), however a lowercase Ä can no longer be produced in a browser. I can get the lowercase Ä in a terminal though with pressing Ctrl+Ä which also makes no sense. Also, number SIX (6) is gone - I can enter it via numpad however not from the straight line on top of qwerty.
Keyboard layout in XFCE4 environment and vconsole is et for estonian. No-deadkeys variant produces no difference and both were working just fine prior to installing ckb-next.
The worst part is that I cannot even revert this process - completely uninstalling ckb-next with the daemon does not revert the issue, so what exactly is the daemon/program doing that it has re-mapped or overwritten system-wide keyboards and how can one revert its changes?

Keyboard:
Corsair K95 RGB Platinum
Arch linux
ckb v0.2.9

# Bug report

- [X ] Yes
- [ ] No

## General information

### Source
ckb-next package from AUR

## Logs & crash reports
[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1767984/ckb-dev-detect-report.gz)

## What is the program's current behavior?
No 6 nor Ä on the keyboard
## What would you like it to do instead?
Work like a normal keyboard and be able to type 6 and Ä - even if it means reverting from ckb
## Any other notes & comments?
See above

Kallestofeles, 2018-02-28:
Found a solution. It seems that XFCE does some weird magic with keyboard mappings. Namely, the issue resolves itself once I unmap a keyboard shortcut from TILDE button, even though they are completely different buttons with different values:
^0x29 0xa9 
ä0x28 0xa8 

Also, the special character worked just fine in lightdm login screen and in a direct TTY session, hence the suspicion fell onto XFCE itself and ckb-next.
The case closed, thank you and sorry for the trouble. =)

closed 2018-02-28 by Kallestofeles

ZirconiumX, 2018-02-28:
Actually, you mentioned you had a Swedish layout. Would you mind joining IRC - there's a button for it on the top of the read me. We've been looking for people with rare keyboard layouts, and you might just fit the bill.