ckb-next/ckb-next #113: CKB unresponsive with STRAFE RGB

SG420, 2018-03-03:
- [Before you proceed](#before-you-proceed)
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior)
    - [What would you like it to do instead?](#what-would-you-like-it-to-do-instead)
    - [Any other notes & comments?](#any-other-notes--comments)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)
- [Something else](#something-else)

---

# Before you proceed

- [x] __Try__ the latest code by cloning and building the project manually.
- [x] __Read__ the [`README.md`](https://github.com/ckb-next/ckb-next/blob/master/README.md) _in full_.
- [x] __Skim__ through other `.md` documents in the project's root. They contain more specific information.
- [x] __Search__ in open _and_ closed [Issues](https://github.com/ckb-next/ckb-next/issues). [Here's how](https://help.github.com/articles/searching-issues/).
- [x] __Avoid__ opening a new issue in case a similar one already exists (don't worry, we get the notifications for closed issues as well, it will be reopened if needed).
- [x] __Keep__ unrelated issues separate (create a new one if you are sure yours is unique).
- [ ] __Use__ [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to write on GitHub. Use triple backticks if copy-pasting more than _one_ line.

Now, __before you proceed even further__, make sure you are following the recommendations above. It will save a lot of time for you and us in the future.

---

# Devices information

Run [`ckb-dev-detect`](https://github.com/ckb-next/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.

---

# New device support request

- [ ] Yes
- [x] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

- [ ] Yes
- [x ] No

<sub><sup>leave empty if you have no feature requests</sup></sub>

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

- [x] Yes
- [ ] No

Whenever I open ckb, it will open the GUI, but the keyboard buttons on the GUI will al be white, and the program will be completely unresponsive. It does not set the lighting on the keyboard. I have tried installing it through both the `ckb-next-git` AUR package, and installing it directly from source. I am running a fresh install of Artix lxqt on a Thinkpad T420.

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)
Currently using the version built manually with the quickinstall script.

### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)
Master branch
## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)
running it through a terminal outputs the following:
```
QObject::startTimer: Timers cannot have negative intervals
Path to settings is "/home/$USER/.config/ckb/ckb.conf"
Scanning  "/usr/lib/ckb-animations/ckb-gradient"
Scanning  "/usr/lib/ckb-animations/ckb-heat"
Scanning  "/usr/lib/ckb-animations/ckb-mviz"
Scanning  "/usr/lib/ckb-animations/ckb-pinwheel"
Scanning  "/usr/lib/ckb-animations/ckb-rain"
Scanning  "/usr/lib/ckb-animations/ckb-random"
Scanning  "/usr/lib/ckb-animations/ckb-ripple"
Scanning  "/usr/lib/ckb-animations/ckb-wave"
Starting  "/usr/lib/ckb-animations/ckb-wave"
Starting  "/usr/lib/ckb-animations/ckb-gradient"
```

You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

<sub><sup>if your issue is about something else, try your best to describe it here, otherwise leave empty</sup></sub>

SG420, 2018-03-03:
[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1777029/ckb-dev-detect-report.gz)

[image]

ZirconiumX, 2018-03-03:
Hmmm, this is interesting.

Does running `echo "rgb ff0000" > /dev/input/ckb1/cmd` **as root** make your keyboard turn red? If it does, the bug is in the GUI (which is what I'm expecting at the moment), otherwise it's in the daemon.

SG420, 2018-03-03:
@ZirconiumX Yes, it does turn the keyboard red.

tatokis, 2018-03-03:
@SG420 Did this happen after a ckb update?
I guess I'll try installing Artic Linux in a VM again, as it appears I can't use pacman properly from a live environment.

SG420, 2018-03-03:
@tatokis No, this was a fresh install for me, so it's possible there was an update before I installed it (I only installed it yesterday). As for pacman working properly in artix, you need to disable the testing sources in `/etc/pacman.d/mirrorlist-arch`

tatokis, 2018-03-03:
@SG420 That helped with pacman, however `basestrap /mnt base base-devel` is failing with missing dependencies.
My patience has pretty much ran out at this point. Maybe I will try another OS with LXQT to test.

tatokis, 2018-03-04:
I tried a live lubuntu next image, however I can't reproduce the bug there.
Maybe try running the GUI with strace and see what the last few lines are before it locks up.

[image]

SG420, 2018-03-05:
@tatokis the issue doesn't only occur on lxqt, but also on i3 (which I am using), it doesn't happen on Manjaro on my desktop, which is also running i3. With Artix, it has the calamares installer, which is what I used to install it, at least on the lxqt version. 

I ran ckb with ctrace, and this is the last few lines:
```
statx(18, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_ALL, {stx_mask=STATX_BASIC_STATS, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=1, ...}) = 0
read(18, "\n", 16384)                   = 1
read(18, "", 16383)                     = 0
close(18)                               = 0
semop(229376, [{0, -1, SEM_UNDO}], 1)   = 0
getpid()                                = 5348
semop(229376, [{0, 1, SEM_UNDO}], 1)    = 0
poll([{fd=4, events=POLLIN}], 1, 18)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 22)    = 0 (Timeout)
openat(AT_FDCWD, "/dev/input/ckb0/connected", O_RDONLY|O_CLOEXEC) = 18
statx(18, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_ALL, {stx_mask=STATX_BASIC_STATS, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=1, ...}) = 0
openat(AT_FDCWD, "/dev/input/ckb0/version", O_RDONLY|O_CLOEXEC) = 19
statx(19, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_ALL, {stx_mask=STATX_BASIC_STATS, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=7, ...}) = 0
statx(19, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_ALL, {stx_mask=STATX_BASIC_STATS, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=7, ...}) = 0
read(19, "v0.2.9\n", 16384)             = 7
read(19, "", 16377)                     = 0
close(19)                               = 0
statx(18, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_ALL, {stx_mask=STATX_BASIC_STATS, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=1, ...}) = 0
read(18, "\n", 16384)                   = 1
read(18, "", 16383)                     = 0
close(18)                               = 0
semop(229376, [{0, -1, SEM_UNDO}], 1)   = 0
getpid()                                = 5348
semop(229376, [{0, 1, SEM_UNDO}], 1)    = 0
poll([{fd=4, events=POLLIN}], 1, 8)     = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 1 ([{fd=4, revents=POLLIN}])
read(4, "\1\0\0\0\0\0\0\0", 16)         = 8
poll([{fd=4, events=POLLIN}], 1, 0)     = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 32)    = 0 (Timeout)
poll([{fd=4, events=POLLIN}], 1, 33^Cstrace: Process 5348 detached
 <detached ...>

```

ZirconiumX, 2018-03-06:
Strace tends not to be very helpful in locating the problem. Maybe if you
drop by IRC, we can try to interactively debug.

SG420, 2018-03-06:
Sure, if you give me the server details I can join.

ZirconiumX, 2018-03-06:
#ckb-next on Freenode. I'll be free in about six hours from now.

ZirconiumX, 2018-03-10:
A gdb backtrace showed that this seems to be because libpulse is locking up when the PulseAudio daemon isn't running. We still can't reproduce on our end, but the backtrace gave us a reasonable guide. It can be mitigated by not building with PulseAudio support in the GUI.

The fix might be to check if PulseAudio is even running before using libpulse.

tatokis, 2018-03-11:
It should be fixed by https://github.com/ckb-next/ckb-next/commit/ee1a4cc06a946dfc5b0f9d46cf6974462c692c45

Please confirm that the issue is indeed solved.

tatokis, 2018-03-11:
Confirmed fixed on IRC.

closed 2018-03-11 by tatokis