ckb-next/ckb-next #114: Japanese K65 RGB - Caps Lock key is seen as Left Alt

manvari, 2018-03-08:
__Switch on the "Preview" mode for better reading experience.__
Use it to put ticks and then switch back to write in the details where needed.

- [Before you proceed](#before-you-proceed)
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior)
    - [What would you like it to do instead?](#what-would-you-like-it-to-do-instead)
    - [Any other notes & comments?](#any-other-notes--comments)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)
- [Something else](#something-else)

---

# Before you proceed

- [x] __Try__ the latest code by cloning and building the project manually.
- [x] __Read__ the [`README.md`](https://github.com/ckb-next/ckb-next/blob/master/README.md) _in full_.
- [x] __Skim__ through other `.md` documents in the project's root. They contain more specific information.
- [x] __Search__ in open _and_ closed [Issues](https://github.com/ckb-next/ckb-next/issues). [Here's how](https://help.github.com/articles/searching-issues/).
- [x] __Avoid__ opening a new issue in case a similar one already exists (don't worry, we get the notifications for closed issues as well, it will be reopened if needed).
- [x] __Keep__ unrelated issues separate (create a new one if you are sure yours is unique).
- [x] __Use__ [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to write on GitHub. Use triple backticks if copy-pasting more than _one_ line.

Now, __before you proceed even further__, make sure you are following the recommendations above. It will save a lot of time for you and us in the future.

---

# Devices information

Run [`ckb-dev-detect`](https://github.com/ckb-next/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.

[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1792541/ckb-dev-detect-report.gz)

---

# New device support request

- [ ] Yes
- [x] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

- [ ] Yes
- [x] No

<sub><sup>leave empty if you have no feature requests</sup></sub>

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

- [x] Yes
- [ ] No

<sub><sup>leave empty if you have no bug reports</sup></sub>

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)

Built manually from the latest git commit (as of 08 March 2018) using quickinstall.

### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)

master

## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)

You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?

For the Japanese K65 RGB keyboard, the Caps Lock key is displayed as and acts like the Left Alt key in ckb-gui. As a result, the Caps Lock key doesn't really exist for ckb. Rebinding the Left Alt key to something else also changes the binding for the Caps Lock key.

## What is the expected behavior of the program?

The Caps Lock key should appear in ckb-gui, and it should be independent from the Left Alt key.

## Any other notes & comments?

---

# Something else

<sub><sup>if your issue is about something else, try your best to describe it here, otherwise leave empty</sup></sub>

ZirconiumX, 2018-03-08:
This could be a keymap issue.

Can you reproduce this with 5edc7af2 (the commit before @tatokis fixed the
Fn key issue)?

 You know where to find us on IRC.

tatokis, 2018-04-09:
[image]

tatokis, 2018-04-09:
Funny how the screenshot above has corruption (left alt is super). I think I have fixed it though, so please delete your gui profile and reply if it happens again.

closed 2018-04-09 by tatokis

manvari, 2019-11-09:
Apologies for not replying back when I opened the issue. The keymap corruption has indeed been fixed since then.

Also, with the xkbmap `jp 106` layout, I figured out why pressing on Caps Lock wouldn't activate caps: it's mapped to a Japanese-specific modifier.

```
❯ xev | awk -F'[ )]+' '/^KeyPress/ { a[NR+2] } NR in a { printf "%-3s %s\n", $5, $8 }'
66  Eisu_toggle
```

The correct way to activate caps with the `jp 106` layout is to press on Shift + Caps Lock simultaneously.

Related: https://bugs.freedesktop.org/show_bug.cgi?id=3193