ckb-next/ckb-next #116: K68 is not detected after K68 RGB merge

mauriciofauth, 2018-03-13:
# Bug report

K68 is not detected after K68 RGB merge (#72).

Last working commit is d79bf1880a5b4f6af60395a33d69e9435374b682.

## General information

### Source
built manually using quickinstall script

### Branch
master

tatokis, 2018-03-13:
They did it again.
https://github.com/ckb-next/ckb-next/blob/master/src/ckb-daemon/usb.h#L182
Replace IS_K68(kb) with 0 so that it's `#define IS_V3_OVERRIDE(kb) (0)` and see if that fixes anything

mauriciofauth, 2018-03-13:
No. That alone does not fix the issue.

tatokis, 2018-03-13:
Then please explain what you mean by not detected. 
Run the dev detect script and upload the output.

mauriciofauth, 2018-03-13:
When I run the `./quickinstall` script, I'm asked if I want to run the driver, and then the keyboard stops working.
```
Run the driver as a startup service [Y/n]? y
```

And on ckb-next the keyboard is never detected.

    No devices connected

mauriciofauth, 2018-03-13:
[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1808308/ckb-dev-detect-report.gz)

ZirconiumX, 2018-03-13:
What does running ckb-daemon say? I'm pretty sure that @tatokis' fix should
have solved the problem. The keyboard "not working" is a symptom that
suggests the daemon recognises and connects to the keyboard, but can't
communicate with it.

mauriciofauth, 2018-03-13:
```
$ sudo ./bin/ckb-daemon
    ckb: Corsair RGB driver v0.2.9
[I] Root controller ready at /dev/input/ckb0
[I] Connecting Corsair Gaming K68 Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
Segmentation fault
```

tatokis, 2018-03-13:
@mauriciofauth I have pushed a needed fix.
Please get the latest master code, and if it doesn't work, from the root of the source, run:
`./qmake-auto && make debug`
and then run the daemon with gdb. `sudo gdb ./bin/ckb-daemon`
Once it crashes, in the gdb prompt enter bt full and press enter `(gdb) bt full`
and paste the output here.

mauriciofauth, 2018-03-13:
```
(gdb) bt full
#0  rawmemchr () at ../sysdeps/x86_64/rawmemchr.S:37
No locals.
#1  0x00007ffff786cff2 in _IO_str_init_static_internal (sf=sf@entry=0x7ffff6de5b20, ptr=ptr@entry=0x0, size=size@entry=0, pstart=pstart@entry=0x0) at strops.c:41
        fp = 0x7ffff6de5b20
        end = <optimized out>
#2  0x00007ffff7860267 in _IO_vsscanf (string=0x0, format=0x4196be "%hx", args=args@entry=0x7ffff6de5c48) at iovsscanf.c:40
        ret = -138725472
        sf = {_sbf = {_f = {_flags = -72515584, _IO_read_ptr = 0x0, _IO_read_end = 0x0, _IO_read_base = 0x0, _IO_write_base = 0x0, _IO_write_ptr = 0x0, _IO_write_end = 0x0, _IO_buf_base = 0x0, 
              _IO_buf_end = 0x0, _IO_save_base = 0x0, _IO_backup_base = 0x0, _IO_save_end = 0x0, _markers = 0x0, _chain = 0x0, _fileno = 0, _flags2 = 0, _old_offset = 0, _cur_column = 0, 
              _vtable_offset = 0 '\000', _shortbuf = "", _lock = 0x0, _offset = 0, _codecvt = 0x7fffe800c480, _wide_data = 0xffffffffffffffff, _freeres_list = 0x0, _freeres_buf = 0xf6de5d20, 
              __pad5 = 140737335155840, _mode = -1, _unused2 = "\000\000\000\000<E\207\367\377\177\000\000d\000\000\000\000\000\000"}, vtable = 0x7ffff7bb37a0 <_IO_str_jumps>}, _s = {
            _allocate_buffer = 0xf6de5d20, _free_buffer = 0x7fff00000001}}
#3  0x00007ffff785a917 in __sscanf (s=<optimized out>, format=<optimized out>) at sscanf.c:32
        arg = <error reading variable arg (Attempt to dereference a generic pointer.)>
        done = -138725472
#4  0x000000000040ae8a in os_inputmain (context=0x629128 <keyboard+328>) at usb_linux.c:330
        nextentry = 0x7fffe800b060
        finalpath = "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/ep_83"
        ep = 131
        path = 0x7fffe800b0a0 "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2"
        epstr = "/ep_83"
        child = 0x0
        sizehex = 0x0
        size = 64
        i = 2
        kb = 0x629128 <keyboard+328>
        fd = 3
        vendor = 6940
        product = 6975
        index = 1
        urbcount = 3
        __func__ = "os_inputmain"
        urbs = {{type = 1 '\001', endpoint = 129 '\201', status = 0, flags = 0, buffer = 0x7fffe800d490, buffer_length = 8, actual_length = 0, start_frame = 0, {number_of_packets = 0, 
              stream_id = 0}, error_count = 0, signr = 0, usercontext = 0x0, iso_frame_desc = 0x7ffff6de5d98}, {type = 1 '\001', endpoint = 130 '\202', status = 0, flags = 0, 
            buffer = 0x7fffe8001e80, buffer_length = 64, actual_length = 0, start_frame = 0, {number_of_packets = 0, stream_id = 0}, error_count = 0, signr = 0, usercontext = 0x0, 
            iso_frame_desc = 0x7ffff6de5dd0}, {type = 0 '\000', endpoint = 0 '\000', status = 0, flags = 0, buffer = 0x0, buffer_length = 0, actual_length = 0, start_frame = 0, {
              number_of_packets = 0, stream_id = 0}, error_count = 0, signr = 0, usercontext = 0x0, iso_frame_desc = 0x7ffff6de5e08}}
        dev_udev = 0x62a420
        enumerate = 0x7fffe80008c0
        udeventry = 0x7fffe800c440
        retval = 0
#5  0x00007ffff7bc16ba in start_thread (arg=0x7ffff6de6700) at pthread_create.c:333
        __res = <optimized out>
        pd = 0x7ffff6de6700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737335158528, 4558189627788347558, 0, 140737343548975, 140737335159232, 0, -4558173971548092250, -4558172612419925850}, mask_was_saved = 0}}, 
          priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
---Type <return> to continue, or q <return> to quit---
```

tatokis, 2018-03-13:
@mauriciofauth I have pushed two fixes in master for this, as there were two separate issues. Please try now.

mauriciofauth, 2018-03-13:
@tatokis Nicely done! Now it's working.

commit 4250bb2d (2018-03-13): Remove K68 NRGB from IS_V3_OVERRIDE
Closes #116
closed 2018-03-13 by mauriciofauth