ckb-next/ckb-next #117: Left click doesn't always register on macOS

kebian, 2018-03-14:
# Devices information

[ckb-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1813206/ckb-dev-detect-report.gz)
---

# New device support request

- [ ] Yes
- [x] No

---

# Feature request

- [ ] Yes
- [x] No
---

# Bug report

- [x] Yes
- [ ] No

## General information

### Source
Cloned from this git repository and built using the `quickinstall` script on macOS 10.13

### Branch
master

## Logs & crash reports
Output of `ckb-daemon`
```    ckb: Corsair RGB driver v0.2.9
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair Gaming SCIMITAR RGB Mouse at /var/run/ckb1
[E] iterate_devices_hid (usb_mac.c:877): Failed to create device plugin: e00002be
[I] Starting input thread for /var/run/ckb1
[I] Setup finished for /var/run/ckb1
[I] Registering EventTap for modifier keys.
```
Output of `ckb`
```
QObject::startTimer: Timers cannot have negative intervals
Path to settings is "/Users/robs/Library/Preferences/com.ckb.ckb.plist"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-gradient"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-heat"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-pinwheel"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-rain"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-random"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-ripple"
Scanning  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-wave"
Starting  "/Applications/ckb.app/Contents/Resources/ckb-animations/ckb-wave"
qt.network.ssl: Error receiving trust for a CA certificate
qt.network.ssl: Error receiving trust for a CA certificate
Downloaded new firmware list. 31 entries found.
```
## What is the program's current behavior?

Left clicks are not always registered.  This is particularly noticeable if you open a few tabs in Chrome and quickly try clicking between them.  Sometimes you can even get it to drag a different tab than the one the mouse pointer is positioned over.

## What is the expected behavior of the program?

Clicks should always happen at the current coordinates of the mouse pointer.

## Any other notes & comments?

I'm using a Corsair Scimitar RGB.  I don't know if other mice have the same issue but I've tried it on 2 different iMacs.

There was a small error in the source that prevented compilation on the master branch.  I fixed it using the following.

```From 40e39f6486d50aa56c3989529f8b5941d83c1f36 Mon Sep 17 00:00:00 2001
From: Rob Stiles <robs@codexsoftware.co.uk>
Date: Wed, 14 Mar 2018 21:12:25 +0000
Subject: [PATCH] Fix fakedb

---
 src/ckb-daemon/usb_mac.c | 5 +++--
 1 file changed, 3 insertions(+), 2 deletions(-)

diff --git a/src/ckb-daemon/usb_mac.c b/src/ckb-daemon/usb_mac.c
index c3dce12..ae22acc 100644
--- a/src/ckb-daemon/usb_mac.c
+++ b/src/ckb-daemon/usb_mac.c
@@ -789,7 +789,8 @@ static usbdevice* add_hid(hid_dev_t handle, io_object_t** rm_notify){
     int handle_idx;
 
     // The usbdevice struct is created later on, however in order to use IS_V3_OVERRIDE we need one with vendor and product, so a temporary one is created
-    struct {uint16_t vendor; uint16_t product;} fakekb;
+    struct fakekb_t {uint16_t vendor; uint16_t product;} fakekb;
+    struct fakekb_t *pfakekb = &fakekb;
     fakekb.vendor = idvendor;
     fakekb.product = idproduct;
 
@@ -808,7 +809,7 @@ static usbdevice* add_hid(hid_dev_t handle, io_object_t** rm_notify){
                 (input == 8 ||                  // Keyboards
                  input == 7)) ||                // Mice
             ((fwversion >= 0x300 ||             // FWv3 hack
-                IS_V3_OVERRIDE(&fakekb)) &&
+                IS_V3_OVERRIDE(pfakekb)) &&
                 input == 64 &&
                 output <= 2 &&
                 feature == 1))
-- 
2.14.3```
---

kebian, 2018-03-15:
Not sure why I assigned a pointer in that patch.  I probably could have just cast it in the IS_V3_OVERRIDE macro.

```
IS_V3_OVERRIDE((fakekb_t *) &fakekb))
```

ZirconiumX, 2018-03-15:
The patch doesn't make sense to me, since `&fakekb` makes a pointer of the
`fakekb_t` type. You shouldn't have to cast a pointer type to itself to get
it to compile.

But anyway: we can try our best, but the dev team is currently two people
down, and they're the two people with Macs.

kebian, 2018-03-15:
The compilation error is...
```
usb_mac.c:811:17: error: member reference type 'struct (anonymous struct at usb_mac.c:792:5)' is not a pointer; did you mean to use '.'?
                IS_V3_OVERRIDE(&fakekb)) &&
                ^               ~~~~~~
./usb.h:183:44: note: expanded from macro 'IS_V3_OVERRIDE'
#define IS_V3_OVERRIDE(kb)              (kb->product == P_K68)
                                         ~~^
usb_mac.c:811:32: error: cannot take the address of an rvalue of type 'uint16_t' (aka 'unsigned short')
                IS_V3_OVERRIDE(&fakekb)) &&
                ~~~~~~~~~~~~~~~^~~~~~~~
./usb.h:183:42: note: expanded from macro 'IS_V3_OVERRIDE'
#define IS_V3_OVERRIDE(kb)              (kb->product == P_K68)
                                         ^~~~~~~~~~~
```

I think I understand what you're saying.  `&fakekb` is a reference.  References are technically addresses and therefore equivilent to pointers but the compiler won't let us dereference it as such here.

A better patch would be the following...

```
From befcc301157a5d392ce050cbfa30c2e4a7546fc6 Mon Sep 17 00:00:00 2001
From: Rob Stiles <robs@codexsoftware.co.uk>
Date: Thu, 15 Mar 2018 17:00:59 +0000
Subject: [PATCH] Fix fakekb

---
 src/ckb-daemon/usb_mac.c | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/src/ckb-daemon/usb_mac.c b/src/ckb-daemon/usb_mac.c
index c3dce12..e3ab391 100644
--- a/src/ckb-daemon/usb_mac.c
+++ b/src/ckb-daemon/usb_mac.c
@@ -789,7 +789,7 @@ static usbdevice* add_hid(hid_dev_t handle, io_object_t** rm_notify){
     int handle_idx;
 
     // The usbdevice struct is created later on, however in order to use IS_V3_OVERRIDE we need one with vendor and product, so a temporary one is created
-    struct {uint16_t vendor; uint16_t product;} fakekb;
+    struct fakekb_t {uint16_t vendor; uint16_t product;} fakekb;
     fakekb.vendor = idvendor;
     fakekb.product = idproduct;
 
@@ -808,7 +808,7 @@ static usbdevice* add_hid(hid_dev_t handle, io_object_t** rm_notify){
                 (input == 8 ||                  // Keyboards
                  input == 7)) ||                // Mice
             ((fwversion >= 0x300 ||             // FWv3 hack
-                IS_V3_OVERRIDE(&fakekb)) &&
+                IS_V3_OVERRIDE(((struct fakekb_t *)&fakekb))) &&
                 input == 64 &&
                 output <= 2 &&
                 feature == 1))
-- 
2.14.3
```

Note that I even had to add extra brackets on the cast to make sure the macro sees it as a pointer. 

Anyway, thanks for your offer to help.  I do have Macs here and I'm keen to help where I can.  If you've any suspicions about where abouts in the code might be a good place to start looking then I can have a poke around too!

It's like the click is being seen at an older coordinate.  Perhaps a cached one?

ZirconiumX, 2018-03-16:
Actually the fix for the error is much easier than that:

```diff
- #define IS_V3_OVERRIDE(kb)              (kb->product == P_K68)
+ #define IS_V3_OVERRIDE(kb)              ((kb)->product == P_K68)
```

because `&kb->product` tries to make a pointer of `product`, not `kb`. @tatokis could you push this fix please?

As for the Mac code, I honestly have *no clue*. The system backend is `usb_mac.c`, but as-is, that file is technically ***illegal***, because it's APSL2 in a GPL2+ project. Additionally, the Mac driver model requires us to implement mouse acceleration ourselves because the OS wants "cooked" input instead of "raw" input. Maybe the error is in there? You'll have to wade through tons of essentially copy-pasted Apple code, though.

I wish you luck.

tatokis, 2018-03-16:
@ZirconiumX usb_mac.c is perfectly legal. It's input_mac_mouse.c that's problematic, as that's responsible for acceleration.

I'll push a fix for the override asap.

tatokis, 2018-03-16:
@kebian with that out of the way, make sure the clicks work as expected without ckb-daemon running.

I am not really sure what could be causing it, but I am currently the only person with access to macOS Sierra, and I do not have a corsair mouse. Unfortunately, you will have to look into this yourself.

kebian, 2018-03-16:
@ZirconiumX Oh yeah that's much nicer  👍 

@tatokis The clicks seem fine without the daemon.

I've been messing around and found a possible hint to the problem.  If I set the double-click speed to the slowest possible in the general mouse preferences in OSX then it registers every single left click as a double click.  If I set the double-click speed to the fastest possible then the mouse behaves much better.

ZirconiumX, 2018-03-16:
Maybe it's a timing issue in the code, then? That'll be fun to debug.

ianhorse, 2018-06-28:
Can confirm that I'm seeing this same issue on MacOS High Sierra 10.13.5 with an M65 Pro. I'm getting REALLY good at doubleclicking fast!

tatokis, 2018-06-28:
Apologies for not looking into this, but this issue will most likely be fixed with the new Karabiner backend code. 
More details and a build can be found here https://github.com/ckb-next/ckb-next/issues/186

closed 2018-07-19 by tatokis