ckb-next/ckb-next #120: how to bind compose key?

nonchip, 2018-03-17:
using the K70 RGB RAPIDFIRE on linux+Xorg, I want to bind a compose key. couldn't find it anywhere in the GUI, so I grepped the source code, and found https://github.com/ckb-next/ckb-next/blob/master/src/ckb-daemon/keymap.c#L77. so I tried binding the macro "+rmenu,-rmenu" in case this actually composes (instead of being the menu button as suggested by the name), but it doesn't.
is there any way to bind a compose key in ckb?

there's also no "compose:rmenu" (just "compose:menu") setting in the default xkb rules...
I think I'm going with rwin and "compose:rwin" for now, but if there's a way that allows me to do this without going through a "dummy keybind" it'd be nice to know

tatokis, 2018-03-17:
KEY_COMPOSE Is the kernel's compose keycode. https://elixir.bootlin.com/linux/v3.4/source/include/linux/input.h#L339

There is no "dummy keybind".
You need to bind to rmenu (which is the ckb name), so that the kernel receives the right keycode, and then tell X to accept compose sequences with that.

That said, you can bind to F13~F20 keys, and use those for compose sequences without taking up any of the existing keys.

nonchip, 2018-03-17:
> There is no "dummy keybind".

oooh i see, there isn't actually a "compose key" (as in keycode or -map), just a "hey xkb, use that key as compose"?

> That said, you can bind to F13~F20 keys, and use those for compose sequences without taking up any of the existing keys.

actually i already use F13-F15 for a 3-switched usb foot switch for exactly that reason :D
but since I use up a key (right windows in my case) for the compose function anyway i just keep using `compose:rwin` i guess :)

with that being the kernel's compose key anyway i guess that makes most sense (so i can have it as compose on TTYs too).

thanks for the clarification.

tatokis, 2018-03-17:
>oooh i see, there isn't actually a "compose key" (as in keycode or -map), just a "hey xkb, use that key as compose"?
I think that is the case, yes. Don't think it would be named KEY_COMPOSE otherwise.

I noticed that the kernel has up to F24, and we only do up to F20. So I should probably look into adding the extra ones.

nonchip, 2018-03-17:
> I noticed that the kernel has up to F24, and we only do up to F20. So I should probably look into adding the extra ones.

afaik think Xorg actually only has up to F20 though.
I only need up to F15, but not in ckb anyway, but yeah, more F-keys might be nice for e.g. the models with additional macro keys to bind to those (for if you don't need actual macros but just keybinds).

aaanyway i guess since my question is answered i'll just close this issue :)

closed 2018-03-17 by nonchip

tatokis, 2018-03-17:
Interestingly enough, they are mapped to XF86 for me.
For example, F23 is XF86TouchpadOff.

I'll add them anyway.

nonchip, 2018-03-18:
yeah exactly, not to F-keys though. rather to what you'd find on e.g. laptop keyboards as "Fn+F*" combinations. and you should call them e.g. "TouchpadOff", instead of "F23" then, because people should be able to expect their desktop environment to handle them in those ways.

patvdleer, 2019-07-04:
I'm looking for a way to set the `Fn+F#` to F13~F24.

I know assumption is the mother of all fuckups but I thought I could do this by selecting the buttons while holding cntrl in de binding menu but that set the F13 to F1 and Fn, not the combination. 

I haven't played with the macro's yet (got the K68 yesterday) but is it me or could this be dealt with much easier?

nonchip, 2019-07-05:
@patvdleer i don't think ckb-next can bind things to key combinations (yet?), just single keys. also this isn't really related to this issue, you would probably be better off adding a new one as a feature request for handling key combinations. then again handling combos is usually the job of things like `xbindkeys` or your window manager.

JoshDreamland, 2020-11-22:
Sorry to be a pest, but could we reopen this as a feature request to handle the binding for us? At the point where ckb-next can run custom macros in response to a key, it seems supporting compose key as a binding isn't a huge deal.

The reason I care is, the "proper" way for me to do this (as was mentioned) is to bind the key to menu and run `setxkbmap -option compose:menu`. The problem is, I have to rerun this command every time the keyboard is initialized. So adding it to my bashrc or profile is insufficient; it will not be rerun when the keyboard disconnects and reconnects (e.g. on system suspend).

So what I've done instead is tell ckb-next to run `xdotool key Multi_key` upon key-up. This is an incredibly ugly solution that works perfectly.

My feature request would be to allow binding virtual keys to a hardware key, but then run the `setxkbmap` command (or call some API equivalent?) immediately after keyboard initialization. Heck, giving me an editable script that runs after initialize would be fine, but automatically handling compose key would be super nice.

Thanks!