ckb-next/ckb-next #122: [arch] Strafe RGB setting mode programmatically not working

t03i, 2018-03-18:
As indicated by the documentation, I am trying to set the mode of my Strafe RGB programmatically using: 
```
echo 'active' > /dev/input/ckb1/cmd
echo 'mode 2 switch' > /dev/input/ckb1/cmd
```
The keyboard is clearly activated indicated by the numlock light, which somehow turns off.
However, the keyboard just flickers shortly without actually changing anything.
This is especially strange, since changing the mode using the GUI works flawlessly.

I am using the prebuild and up-to-date ckb-next-git package on my Arch-Linux installation.

ckb-next-dev-detect-report:
[data.txt](https://github.com/ckb-next/ckb-next/files/1823251/data.txt)

tatokis, 2018-03-19:
I don't think the strafe supports multiple modes.
That said, the GUI needs to not be running to switch between modes that way.

ZirconiumX, 2018-03-19:
Don't be confused between "modes", which are hardware colour settings that
work when ckb-next is off, and "profiles", which are software colour
settings controlled by the GUI.

Your keyboard only supports a single mode; while profiles are controlled by
the GUI alone.

I'm not happy with that arrangement, but for now, that's how it is.

t03i, 2018-03-19:
That was my error. Thank you for pointing it out! 
I'll have to find a manual solution then, thanks.

closed 2018-03-19 by t03i

ZirconiumX, 2018-03-19:
When I get home, I'll clarify the wording of that document for you.