ckb-next/ckb-next #123: Can't use it as evdev inpu

Alkeryn, 2018-03-20:
So i use qemu to do gpu passthrough
and the keyboard is nowhere in /dev/input/by-path/
i can find it by /dev/input/event3
but there is another issue

/dev/input/ckb1/notify1 is readable as plain text
for other users, which is really unsecure
i wouldn't want my apps to be able to read what i'm typing everywhere
including password and all

tatokis, 2018-03-20:
Regarding the first one, there is nothing that can be done.

The second one is answered here. https://github.com/ckb-next/ckb-next/blob/master/DAEMON.md#security

closed 2018-03-20 by tatokis

Alkeryn, 2018-03-20:
Wow quick
i mean i would like to do evdev passthrough ^^"
but the onlyway is to kill ckb
plus i can't kill it because it hang

also my computer hang while shutdown because it can't kill ckb

and in the logs i see the it get sigkill in the end any idea how to fix that ? :/

tatokis, 2018-03-20:
You're either going to have to make a symlink with `ln -s` from the eventX file manually, or point qemu to the event file.

The daemon hanging is a known issue, however since no one in the dev team can reproduce that issue, it is quite difficult to debug it.
Only solution at the moment is to sigkill.
I thought there was an issue open in the bug tracker, however I can't seem to find it.

Do you have one or multiple devices managed by ckb? (For example, keyboard and mouse.)
Can you please run ckb-next-dev-detect and upload the output?

Alkeryn, 2018-03-20:
Well i just have a keyboard
Corsair k70 rapidfire rgb

Yeah but what is the event file ?
there is one that send plain text but that's not what it should look like ^^

tatokis, 2018-03-20:
Check in `/sys/devices/virtual/input/input*`.
The software creates a mouse and a keyboard for every physical device.
So for example, your rapidfire shows up as two devices, a mouse and a keyboard.

Those are created inside the input directory. Once you find both, you can look inside the directories.
Inside them, there will be an "eventX" file, where X is a number.

If you `sudo cat /dev/input/eventX` you will get binary evdev data.
So you can pass both of those event files through to the VM.

Alkeryn, 2018-06-24:
I made this script that fix the issue with an autodetect symlink
[](https://github.com/Alkeryn/dotfiles/blob/master/bin/corsairid)
just load it with your init after corsair daemon load and it should be good

tatokis, 2018-06-27:
Thanks for this. I will add a link to your script in the wiki