ckb-next/ckb-next #124: High pitched noise coming from the keyboard

gianmarcotoso, 2018-03-21:
# Something else

The keyboard emits an high pitched noise when led illumination is enabled through `ckb-next`. The problem does **not** happen with the official Corsair Utility in Windows and is consistent across multiple machines. It happens regardless of whether the keyboard is connected through a USB switch or directly to the machine.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1834250/ckb-next-dev-detect-report.gz)

ZirconiumX, 2018-03-21:
This is...strange. I have no idea what might cause it other than PWM issues, which begins to sound more like a hardware issue than anything.

tatokis, 2018-03-21:
@gianmarcotoso Try changing the refresh/frame rate to 60 if it is set to 30, or vice versa.
It's under Settings -> More settings

gianmarcotoso, 2018-03-21:
@tatokis : I did try that, but unfortunately nothing changes.
@ZirconiumX: I don't exclude the possibility, what strikes me as weird is that this only happens under Linux and only with ckb-next running. If I don't start ckb and leave the keyboard "rainbowing", no sound comes from it, and neither does any sound come from it under Windows using the official client and any kind of color configuration. 

Is there a way to load a profile from hardware through ckb (either the client or the daemon)? I tried sifting through the daemon doc and I tried echoing the commands to change "mode" but I wasn't able to load an hardware profile I did save in Windows. A colleague of mine which is more into embedded programming and did some work with LEDs suggested it might have something to do with the frequency to which the RGB LEDs are set to their color - the wrong frequency might do something like this. Could this be the case? In any case, if it does not do it when using an hardware profile (the rainbow thingy), it shouldn't do it when loading that hardware profile - even through ckb. It might be worth a try.

tatokis, 2018-03-21:
Hardware profiles are not yet implemented for the platinum, so no, that won't work.

It is indeed very interesting. We haven't had anyone else report such a problem, so I can't really say much.
Have you tried plugging in the keyboard to a powered usb hub, or maybe another usb controller.
It could be some kind of power management done by linux that causes this, although a very remote possibility.

ZirconiumX, 2018-03-21:
> @tatokis <https://github.com/tatokis> : I did try that, but unfortunately
> nothing changes.
> @ZirconiumX <https://github.com/zirconiumx>: I don't exclude the
> possibility, what strikes me as weird is [...]
Which makes sense, because the keyboard is animating it, not ckb-daemon.

> Is there a way to load a profile from hardware through ckb (either the
> client or the daemon)?
I've been trying to work on that in the k95p-profile branch.

> I tried sifting through the daemon doc and I tried echoing the commands to
> change "mode" but I wasn't able to load an hardware profile I did save in
> Windows.
I have doubts this will ever be possible.

> A colleague of mine which is more into embedded programming and did some
> work with LEDs suggested it might have something to do with the frequency
> to which the RGB LEDs are set to their color - the wrong frequency might do
> something like this.
Which was why @tatokis suggested changing the framerate.

Could this be the case? In any case, if it does not do it when using an
> hardware profile (the rainbow thingy), it shouldn't do it when loading that
> hardware profile - even through ckb. It might be worth a try.
Except ckb - **and CUE** - are purely software controlling it.

tatokis, 2018-03-21:
If anything, the noise would be coming from the PWM controller as it's scanning the LEDs, so it doesn't matter what is animating it. That's the troubling part...

gianmarcotoso, 2018-03-21:
I did try all of the following

- Connect the keyboard directly to my Arch Linux box (noise)
- Connect the keyboard directly to my Windows box (no noise)
- Connect the keyboard directly to my Windows box and run Ubuntu from a USB drive (noise)
- All of the above through a powered USB switch (same results)
- All of the above through an unpowered USB switch (same results)

Tomorrow I'll try to see if a K70 I have lying around somewhere does the same thing, but I remember using it with Linux and ckb and never noticing anything strange...

gianmarcotoso, 2018-03-21:
@ZirconiumX: I'm also going to make a couple more tries with the framerate. Maybe there is a sweet spot somewhere (hopefully it's an integer number).

tatokis, 2018-03-21:
If the packets sent by the daemon were causing the noise, then the frequency would change with the update rate, which you said isn't the case. I don't really have anything else in mind.

tatokis, 2018-03-21:
If you start the daemon with the gui not running, do you still get the noise?

ZirconiumX, 2018-03-21:
@tatokis Actually, there is *one* thing I don't think you considered. Remember how the K70 RGB (the original ones) have 9-bit native colour, but support the 24-bit colour update packet? Perhaps something similar is happening here, but more likely that it's representing the software update as a single-frame hardware animation.

gianmarcotoso, 2018-03-21:
@tatokis: No, but the deamon itself doesn't automatically load any profile until I load the gui...
@ZirconiumX: I am not familiar with this codebase but if there is some tweak I can make to test some ideas I'm more than happy to do that, recompile and see what happens!

tatokis, 2018-03-21:
@ZirconiumX That was caused because the LED drivers didn't support 24 bit colour, so it scanned multiple times to give the illusion, thus flickering.
I don't see how something similar could happen in this case. I'll try and get a capture from CUE.

@gianmarcotoso in that state, try echo active > /dev/input/ckb1/cmd and see if that makes the noise start. It tells the keyboard to switch to software mode.

ZirconiumX, 2018-03-21:
You've picked an awkward time to file this bug because we're all europeans and it's fairly late night here. 

Perhaps it would be a little more productive if you [joined us in IRC](http://webchat.freenode.net/?channels=%23ckb-next&uio=d4).

gianmarcotoso, 2018-03-21:
Hey, I'm in Italy and I was in fact about to call it for the day :)

I'll join IRC tomorrow and update you there, thank you!

gianmarcotoso, 2018-03-22:
It's a hardware issue. I was sure there was no sound coming from the keyboard when in hardware mode (maybe there wasn't), but now I can hear it even when I'm the BIOS and the keyboard is rainbowing (on either machine). I've opened a ticket with Corsair.

Now for the weird part: the official Corsair Utility does, in fact, make the noise go away. I have no idea what kind of sorcery happens when that tool takes control, but the sound completely vanishes as soon as it takes control. Do you want me to capture some packets?

ZirconiumX, 2018-03-22:
It can't hurt to capture packets. Do you know how to?

gianmarcotoso, 2018-03-22:
Not really, if you can point me in the right direction I'll try to figure it out though :)

ZirconiumX, 2018-03-22:
Essentially, download [Wireshark](https://www.wireshark.org/) making sure to enable winpcap and usbpcap, which are part of the installer. **Then reboot**, find out which interface shows `URB_INTERRUPT` messages on the left (set it to a rainbow or something interactive and mash keys), then set CUE to a static colour profile, restart the capture, and unplug/replug your keyboard back in, waiting about 10 seconds, That should get us everything we need, and if you're interested we can go through and show you what it's actually doing.

gianmarcotoso, 2018-03-22:
Here you go. There are some trailing packet by my mouse, but everything else should be the keyboard. Let me know if you need more captures, and if this sheds some light of what might have been going on! Thank you! :)

[k95-platinum-noise.zip](https://github.com/ckb-next/ckb-next/files/1838787/k95-platinum-noise.zip)

ZirconiumX, 2018-03-22:
I wrote [a Wireshark script](https://github.com/ZirconiumX/wireshark-cue) to filter out the noise, actually. It's pretty useful.

I think for comparison - though I have a hunch - can you try to use Wireshark on Linux (you'll need to [setup usbmon](https://wiki.wireshark.org/CaptureSetup/USB#Linux)) with ckb, doing a similar thing?

gianmarcotoso, 2018-03-22:
I will... in the weekend!

gianmarcotoso, 2018-03-28:
Here you go. I'm sorry for the noise (there is a lot of it), you should be able to filter it out. The linux machine is less accessible than the other one and I had to use a hub. Let me know if it helps!
[ckb-linux.pcapng.zip](https://github.com/ckb-next/ckb-next/files/1855755/ckb-linux.pcapng.zip)

gianmarcotoso, 2018-03-28:
I did the capture right before packing up the keyboard and leaving it to the courier, sorry! I'll do that when the new one arrives though!

ZirconiumX, 2018-03-28:
>The daemon is trying to talk to your keyboard using URB Controls.
Actually it's using a mixture of controls and interrupts. The `0e 01` [device identification] packet is a control (packet 405), but the subsequent packets are interrupts (packets 461-524). That shouldn't happen.

If it happens again, we have a bug on our hands.

tatokis, 2018-03-28:
@ZirconiumX there is no bug. That's how the old code worked.
It used Controls for everything that needed responses, and interrupts for writes on devices with firmware that supports them.
https://github.com/ckb-next/ckb-next/blob/c46614ae876651508583f8e71375b7541abb292a/src/ckb-daemon/usb_linux.c#L70

ZirconiumX, 2018-03-28:
Ah, yeah, that was because of #98. My bad.

delooper, 2018-04-01:
How high pitch is this noise you're talking about?  

I've got two K95 platinum's being driven by ckb-next on Ubuntu and I can't detect noise from either one. 

Is it close to the limit of normal human hearing?   I don't have extraordinary abilities on that front.

gianmarcotoso, 2018-04-05:
It was I guess close to the limit, if you don't normally hear those sounds I guess you wouldn't be able to hear this one. I am unfortunately able to hear each and every one of this maddening hums and whines. Anyway, Corsair accepted the RMA and the replacement is on its way, I just hope it's not that common of an issue because I have a K70 that does the same thing (regardless of OS or running software) when on the lowest brightness setting.

I'll update the ticket with results from the new keyboard and dumps of the new one with the `master` version of ckb.

OvermindDL1, 2018-04-05:
Yeah I'm really curious about that sound issue.  I hear high-pitched sounds *very* well (better than I hear speech volume according to tests irritatingly) and I've not experienced that, but I don't know what would be causing that unless a capacitor is leaking or a transformer is in the keyboard or so (I've not taken it apart so unsure of it's contents)...

tatokis, 2018-04-05:
As an owner of an original K70 RGB, I can tell you that it whines only in 16.8m colour mode (which ckb doesn't use on these old devices ). 

The reason that model whines is because the led drivers don't support the full advertised colour range, so they update them as fast as they can to emulate the extra colour resolution. That's also the reason they flicker badly in 16.8m colour mode with CUE. 

However, that is not the case with the platinum or any of the newer devices, as they have led drivers capable of the full colour resolution.

gianmarcotoso, 2018-04-05:
Mine is a K70 Vengeance, not RBG so I don't really use any software to control it. It whines even when in the BIOS, it's a defect of my keyboard I guess...

tatokis, 2018-04-05:
Ah. Those legacy devices are completely different.

delooper, 2018-04-05:
If replacement doesn't cure your high-frequency noise problem, could you perhaps train a mic on your keyboard and identify the frequency?  I have a good mic that can detect well above the human-audible range.  Could try to verify if this is a feature, or a bug.

gianmarcotoso, 2018-04-05:
I can try, I do have a RØDE NT3, it should be able to pick it up. I hope I don't have to though! :)

tatokis, 2018-04-08:
@gianmarcotoso Completely unrelated, but since you said you have a legacy K70, would you mind plugging it in and capturing a few key input packets with usbmon+wireshark, as well as running `ckb-next-dev-detect` and `usbhid-dump`?

You can pop in on IRC and we can guide you through it if you want.
I am working on a rewrite of the input handling code and I know very little about these legacy devices.

gianmarcotoso, 2018-04-08:
@tatokis I will, in the following days as soon as I find the time! I'll keep you posted!

tatokis, 2018-04-21:
Bump @gianmarcotoso, both for the K70 and a status update on the platinum.

gianmarcotoso, 2018-04-22:
Sorry, got sidetracked.

The new K95 does not hum or buzz in any way, so it was indeed an hardware problem. I'll now dump the K70!

ZirconiumX, 2018-04-22:
> Here are both the packet capture and the dev-detect. Both were made with
> the latest ckb version from master running, and the latest
> ckb-next-dev-detect was used. The capture data is a little bit noise, as
> the keyboard is connected through a usb switch (but there are no other usb
> devices connected to the computer). Still, reading through them, they seem
> mostly inintelligible (at least to me).
>
> I'm having issues with usbhid-dump, it states "Failed to claim: Resource
> busy". Google is not of much help in this case :(

Don't run `usbhid-dump` with ckb-next-daemon running. That'll be why it's
not working.

gianmarcotoso, 2018-04-22:
Whops :)

Here's the dump. There are three items, one is the usb switch, the other two are the K70.

[dump.zip](https://github.com/ckb-next/ckb-next/files/1935404/dump.zip)

tatokis, 2018-04-22:
@gianmarcotoso Good to hear, and thank you.
The capture doesn't seem to be working at all. Just has a single json line according to wireshark.

However, if you do redo it, please make sure you press the a few alphabet keys, as well as a few media ones. If there's a volume wheel, move that too.

Thank you again, and I'll close this issue as it no longer has anything to do with the original problem.
If you do have another capture, then please post it here and I'll move them to the appropriate place.

closed 2018-04-22 by tatokis

gianmarcotoso, 2018-04-23:
@tatokis I did mash some keys, it should be towards the end of the capture (which I stopped by means of CTRL+E, by the way). I'll try again, maybe I'll attempt to connect the keyboard directly to the computer and see if something changes.

tatokis, 2018-04-23:
@gianmarcotoso Try opening the file you uploaded in wireshark, and you'll see what I mean.

gianmarcotoso, 2018-04-23:
Oh. I must have f'd something up when saving it then, I'll try again as soon as I'm near that computer again!