ckb-next/ckb-next #125: K90 Support

mthome, 2018-03-21:
Linux support for K90 
Mainly this would allow the use of macro programming and events, I presume.

Product page [https://www.corsair.com/us/en/Categories/Products/Keyboards/VENGEANCE%C2%AE-K90-Performance-MMO-Mechanical-Gaming-Keyboard/p/CH-9000003-NA](url) 

# Devices information
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1834576/ckb-next-dev-detect-report.gz)

# New device support request
Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above

ZirconiumX, 2018-03-21:
Yeah, the K90 likely isn't going to be supported.

I *believe* you can get "better" support from the Linux hid-corsair module,
but there's nothing we can offer that either that module or some X macro
programs can't.

OvermindDL1, 2018-03-21:
Just from curiosity, what about it is broken that prevents being able to support it?

_/me doesn't have that keyboard_

mthome, 2018-03-21:
Yah, but I figured someone might both have one and the time to beat it into submission, assuming the wire protocol is close enough.  I was looking mostly for consistency (with supported drivers) and the ability to actually set the in-keyboard memory for macros.  In any case, it IS still a perfectly functional keyboard.

tatokis, 2018-03-21:
@mthome I don't think the daemon supports writing bindings to the keyboard.
That said, there is some support for these old devices. We aim to mostly implement CUE functionality, and since these devices are not supported in CUE, it is a bit of a pain.

@OvermindDL1 the protocol is completely different.
There is no corsair input (0x03 packet type), the endpoint configuration and lengths are completely different, and they send more bytes than their descriptor indicates under certain circumstances.

@mthome as an experiment, replace 1b08 with 1b02 in both those lines https://github.com/ckb-next/ckb-next/blob/master/src/daemon/usb.h#L86-L87 and see what happens.

mthome, 2018-03-27:
@tatokis Interestingly, the 1b08 to 1b02 seems to result in a functional application that allows for definition and saving (!) of keybindings and macros.  In particular, the G key bindings seem to work properly.

I'm not quite sure what the right path forward is here - on the one hand, it would be nice to have this working, but if nobody else cares, I'm happy with what I now have... and on the other other hand, I may just upgrade to a modern corsair, since my intended replacement may never arrive (see "DAS Kickstarter fiasco")

tatokis, 2018-03-27:
@mthome I'll add it.
Did you notice anything missing or something that shouldn't be there due to adding it as a K95?
If not, I'll add it as is, as there's no point in creating a separate k90 entry

tatokis, 2018-03-28:
Added it to master anyway.
If there are any problems, open another issue with the specific problem you are encountering.

closed 2018-03-28 by tatokis