ckb-next/ckb-next #126: Cant install on ubuntu 17.10

rafilkmp3, 2018-03-23:
I cant install on ubuntu 17.10 

i tried 
sudo apt-get install build-essential cmake libudev-dev qt5-default zlib1g-dev libappindicator-dev libpulse-dev libquazip5-dev

but 

Lendo listas de pacotes... Pronto
Construindo árvore de dependências       
Lendo informação de estado... Pronto
build-essential is already the newest version (12.4ubuntu1).
build-essential configurado para instalar manualmente.
cmake is already the newest version (3.9.1-1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-0ubuntu2).
zlib1g-dev configurado para instalar manualmente.
Alguns pacotes não puderam ser instalados. Isto pode significar que
você solicitou uma situação impossível ou, se você está usando a
distribuição instável, que alguns pacotes requeridos não foram
criados ainda ou foram retirados da "Incoming".
A informação a seguir pode ajudar a resolver a situação:

Os pacotes a seguir têm dependências desencontradas:
 libappindicator-dev : Depende: libdbusmenu-glib-dev (>= 0.1.8) mas não será instalado
 qt5-default : Depende: qtbase5-dev mas não será instalado
E: Impossível corrigir problemas, você manteve (hold) pacotes quebrados.

can someone help me install ?

ZirconiumX, 2018-03-23:
Try `sudo apt-get update` first.

rafilkmp3, 2018-03-23:
@ZirconiumX  not works

tatokis, 2018-03-23:
What about just `sudo apt-get -f install` ?

tatokis, 2018-03-23:
Also run `apt-mark showhold` and paste the output here

closed 2018-04-08 by tatokis