ckb-next/ckb-next #127: [new feature] Music Visualizer

DanielChan27, 2018-03-24:
is it possible for playing music visualizer on mac？ my keyboard is K70 RGB

tatokis, 2018-03-24:
I've been thinking of adding mac support for it, however I know nothing about CoreAudio and how macOS handles sound.

The visualiser isn't in a great state, so I've been putting it off until it works properly.
However since there is now a request for it, I may look into it these upcoming days.

Seems a bit complicated, but I am sure it can be managed. https://developer.apple.com/library/content/documentation/MusicAudio/Conceptual/AudioQueueProgrammingGuide/AQRecord/RecordingAudio.html

DanielChan27, 2018-03-25:
awesome! so happy to hear the news that you are already looking into it.

webdev23, 2018-03-26:
Hi  DanielChan27 , i use to made a workaround for k70, it should works on mac. If you are willing to test, or play around:  https://github.com/webdev23/phpCorsair

tatokis, 2018-04-08:
I looked into it a bit more today, and from what I can see there is no easy way to capture desktop audio without using third party software, so I am not sure how to proceed from here.

OvermindDL1, 2018-04-09:
Doesn't CoreAudio have a similar source/sink setup as PulseAudio?  If so then like PulseAudio perhaps there is a way to create a new sink that outputs to a named pipe that the visualizer reads from?

tatokis, 2018-04-10:
@OvermindDL1 It does not, unfortunately.
Best solution I could find was to do a normal capture but tell people to use https://github.com/mattingalls/Soundflower

webdev23, 2018-04-10:
About audio capture to real time data array:
With CAVA https://github.com/karlstav/cava and a special config, here named cava.conf:

```
[general]
bars = 8

[output]
bit_format = 8bit
method = raw
style = mono
```
Then to run: 

    unbuffer ./cava -p cava.conf

[image]

My first works was with the web audio api, in browser, with `Uint8Array()` wich provide similar realtime data.

tatokis, 2018-04-10:
@webdev23 Thanks for the suggestion, but the issue isn't converting audio to data we can process at the moment.
The issue is that macOS doesn't let people capture the audio that's playing on the computer without third party software (from what I can see, anyway).

webdev23, 2018-04-10:
Okay, this can be a security concern, as i guess. Browsers ask users the first time audio is connected, but i never tried to use it headless 

Keep it up amigos!

ItsEcholot, 2018-09-06:
It seems like other software always require 3rd party software like Sound Siphon so I think it should be possible to work with the default sound inputs and let the user select the correct input whatever that may be (mic, sound siphon, ...)

tatokis, 2018-09-06:
Both me and @fleischie tried at some point to just record any audio device, but it seems ridiculously overcomplicated to do even that.
I've been focusing on other things recently, so I didn't really look into this again, but this is more difficult than it should be.

ItsEcholot, 2018-09-06:
Ah I see... Didn't think it would be that complicated if you're using input audio devices. It for sure is not a high priority but would be an awesome thing to have. Thanks for the response.

JBA474, 2019-02-09:
> Both me and @fleischie tried at some point to just record any audio device, but it seems ridiculously overcomplicated to do even that.
> I've been focusing on other things recently, so I didn't really look into this again, but this is more difficult than it should be.

This is a non-free, non-open source option so I'm not sure the comfort level.  I make nothing from this suggestion- What about a $12 (at this time) license for AirParrot?  [https://www.airsquirrels.com/airparrot/features/mac](url)
It has the option to capture the audio (I realize I'm repeating @tatokis comment from above).
I respect that Apple has taken steps to prevent this sort of audio capture and rerouting.  It would be nice if there was an option to toggle it off. I don't know that there isn't parity with some other open/free alternative.  For those people willing to work around the problem using this software, would you consider supporting?  Thanks for considering.

ZirconiumX, 2019-02-09:
Any CKB animation relies on the GUI, which is GPLv2. Anything which is
incompatible with that license is not redistributable, which limits the
purpose of having that animation in the first place.

OvermindDL1, 2019-02-11:
> Any CKB animation relies on the GUI, which is GPLv2.

Eh, only if they use CKB's headers.  I ended up not using them in my 'animation' for system notifications just because I found them harder to use than writing my own wrappers around the stdin/stdout API.

Still, I'd agree that if not GPLv2 then it shouldn't be distributed with ckb regardless.

ZirconiumX, 2019-02-11:
Anything which doesn't use the GUI headers will be effectively incompatible
with the GUI.

tatokis, 2019-02-11:
One can manually implement the API on their own and not use the headers. It's just text and two command line arguments.

That said, I have been working on porting mviz to portaudio, which works on mac and windows.

This way, mviz will just capture from any audio device, and it'd be up to the user on mac to provide a loopback device, whether it's proprietary, or soundflower, without having any license issues.
It's also useful to be able to visualise something like line in as well.

That aside, unfortunately portaudio on linux only supports alsa, and pulseaudio's alsa emulation seems absolutely terrible. Every time I run mviz on linux it manages to lock up pulseaudio and all applications using it on every frame, and that's excluding the fact that mviz ends up having terribly high latency.

The plan is to keep using the pulse simple api on linux by default, and fall back to portaudio on other platforms, or if there is no pulseaudio package at all installed on linux.

The code is very rough, and I need to use another OS in a VM to test the portaudio code, but if anyone wants to have a go it's in the `rework-mviz` branch.
Hopefully I can get it to compile on macOS and test it there.

OvermindDL1, 2019-02-11:
> One can manually implement the API on their own and not use the headers. It's just text and two command line arguments.

Yep, it was *very* simple actually one I reverse engineered it.  ^.^

> This way, mviz will just capture from any audio device, and it'd be up to the user on mac to provide a loopback device, whether it's proprietary, or soundflower, without having any license issues.
> It's also useful to be able to visualise something like line in as well.

Sounds like the way to go.

> That aside, unfortunately portaudio on linux only supports alsa, and pulseaudio's alsa emulation seems absolutely terrible. Every time I run mviz on linux it manages to lock up pulseaudio and all [...]

Yeah the alsa interface is not good, really need setup the vizualizer as a pulseaudio sink and let the user hook things up to it as they wish.  Building a sink is pretty simple and it is designed for very low latency.

webdev23, 2019-02-13:
Hey, guys, let's face it. 
One year++ later, nothing evolved.
This is just an empty shell.
I suspect that you are Corsair insider trolls, and this project a marketing action. 

The original situation where the «original dev disappeared» was already extremly suspicious.

That is both sad, for you and the community, and will start to warn my numerous Linux friends about Corsair, from now. 

My fact. Bye.

ZirconiumX, 2019-02-13:
Your comments are harsh and unjustified, @webdev23, and are not welcome here.

We're not Corsair insiders, we're students, trying our best to work on this despite being at university and being severely burnt out on this.

If it were marketing, they would happily advertise support for Mac and Linux, but they don't.

ccMSC has a life outside of this too, and they're well within their rights to do as they please.

You're under no obligation to use ckb-next, but the original project is dead, and we are still here.

So instead of being bitter, how about you contribute something useful instead?

webdev23, 2019-02-13:
I made my own program, and i make my own lightning just scroll up.
Got many scripts unpublished, and it's good like so.
There is nothing complicated to blink a led.
Keep watching this project for fun, it match the feeling I had about the empty shell.
This program is a bunch of spaghetti, nobody can really make it better.*

Proof is the one year for nothing.

ZirconiumX, 2019-02-13:
Okay, good, you can blink a led. Do you also handle input while doing that?
You need exclusive control of the device for this, and unless you handle
input, nobody can type while you're blinking your led.

webdev23, 2019-02-13:
You didn't even run it, nor opened the page. I can nothing for that.
You are trying to dissolve my comment for less importance, regarding your marketing motivation.
Actually, yes «ckb-next» is randomly blocking typing, beside numerous bugs. It's just unreliable, cpu rush, nothing works.
While can type fast while gradient using simpler scripts, [like this one](https://github.com/webdev23/phpCorsair/blob/master/rainbow)
Using 2% CPU not 50%.
Maybe, you should provide develloper docs for your project.
This should had been done since a long time. 
Empty shell.

```

Hey, guys, let's face it.
One year++ later, nothing evolved.
This is just an empty shell.
I suspect that you are Corsair insider trolls, and this project a marketing action.

The original situation where the «original dev disappeared» was already extremly suspicious.

That is both sad, for you and the community, and will start to warn my numerous Linux friends about Corsair, from now.

My fact. Bye.
```

ZirconiumX, 2019-02-13:
Oh, so your script does nothing more than piggyback on us. C'mon, even I
thought you were trying harder than that.

webdev23, 2019-02-13:
It use ckb daemon, not ckb-next. 
Again, you are showing your lack of understanding of this project, but you are fast to answer.
Maybe you should code more and talk less.
```

Hey, guys, let's face it.
One year++ later, nothing evolved.
This is just an empty shell.
I suspect that you are Corsair insider trolls, and this project a marketing action.

The original situation where the «original dev disappeared» was already extremly suspicious.

That is both sad, for you and the community, and will start to warn my numerous Linux friends about Corsair, from now.

My fact. Bye.
```

tatokis, 2019-02-13:
@webdev23 
> I suspect that you are Corsair insider trolls, and this project a marketing action.

I can't tell if you are trolling or not.
You wish, but sadly no. I'm just a random guy on the internet spending hours of his time trying to reverse engineer corsair's protocols so that you can complain about it.
```
DISCLAIMER: ckb-next is not an official Corsair product.
It is licensed under the GNU General Public License (version 2) in the hope that it will be useful, but with NO WARRANTY of any kind.
```
Don't like it? Don't use it.

> This is just an empty shell.

If the changelog and the git log are anything to go by, then yeah, absolutely nothing has changed.

> The original situation where the «original dev disappeared» was already extremely suspicious.

Welcome to open source hobby projects. Where people work their ass off to write code without getting paid, only to get shit back.

>That is both sad, for you and the community, and will start to warn my numerous Linux friends about Corsair, from now.

What's sad is that you decided to attack us, on our issue tracker, on an open source project that we don't get paid for and Corsair pretends we do not exist.

> I made my own program, and i make my own lightning just scroll up.

Great. Good for you.

> Got many scripts unpublished, and it's good like so.

Your loss. It doesn't affect us.

> There is nothing complicated to blink a led.

You clearly have no idea how any of this works if you think that.

>  This program is a bunch of spaghetti, nobody can really make it better.*

You never contributed any code, you never fixed any bugs, you never suggested any way to make it less spaghetti. All you are doing is complaining with self entitlement.

> It use ckb daemon, not ckb-next.
> Again, you are showing your lack of understanding of this project, but you are fast to answer.

lol

> My fact. Bye.

Farewell, your insulting comments and toxic attitude will not be missed.
You are no longer welcome here. Any further comments from you, and any further comments relating to this will be deleted.

webdev23, 2019-02-13:
> Hey, guys, let's face it.
> One year++ later, nothing evolved.
> This is just an empty shell.
> I suspect that you are Corsair insider trolls, and this project a marketing action.
>
> The original situation [...]

ItsEcholot, 2019-02-13:
Dude stop being childish... I reported you for spamming and toxic behavior

OvermindDL1, 2019-02-13:
Whoa, haven't seen an actual troll in the wild in quite a while actually!!

Just to note, a *lot* of us absolutely love the work you all do here!  And there are a **lot** of silent users as well, like my wife who has done magic with the software from here (she likes a flashy computer ^.^; ).  Don't let trolls get to you.  :-)

Lockjututz, 2019-02-13:
I'm one of those silent users and I'd just like to give you my thanks for the great work you do!

tatokis, 2019-03-05:
Next release will feature macOS support https://github.com/ckb-next/ckb-next/commit/db1eb4ea442ec4b1d0291a1b8b2abc382ee99423

closed 2019-03-05 by tatokis