ckb-next/ckb-next #129: Ubuntu 17.10 Installation Error using ./quickinstall

samdf96, 2018-03-27:
# Bug report

## General information

Installation on Ubuntu 17.10 - not installing.

### Source

  * built manually using quickinstall script

## Logs & crash reports

When running the `.quickinstall` script, the code runs for awhile, and gets to this point where an error pops up:

```
...
[ 48%] Automatic MOC, UIC and RCC for target ckb-next
Generating MOC predefs moc_predefs.h
[ 48%] Built target ckb-next_autogen
[100%] Built target ckb-next
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/ckb-next-daemon
CMake Error at src/daemon/cmake_install.cmake:60 (file):
  file INSTALL destination: /usr/lib/systemd/system is not a directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:37 (include)
  cmake_install.cmake:37 (include)

Makefile:117: recipe for target 'install' failed
make: *** [install] Error 1
An error occurred,
press enter to exit.

```

#### Wondering if there was any quick fix to this issue. I think 17.10 is not officially supported, but I thought I'd give it a try anyway and see. Seems like one of the directories it's trying to write to doesn't actually exist, this might be a new 17.10 issue. Thanks team!

---

tatokis, 2018-03-27:
@samdf96 Sorry for not responding on IRC. I was asleep.

Additionally, I just tried it on my Ubuntu 17.10 VM and it installed just fine.
Can you please run `ls -la /usr/lib/systemd` ?

samdf96, 2018-03-27:
Here is what it outputted when I run ls -la under that directory:
```
/u/l/systemd ❯❯❯ ls -la
  rwxr-xr-x    root    4 KiB  Thu Mar  8 00:43:58 2018    ./
  rwxr-xr-x    root   12 KiB  Mon Mar 26 21:24:48 2018    ../
  rwxr-xr-x    root    4 KiB  Fri Jan  5 13:35:20 2018    boot/
  rwxr-xr-x    root    4 KiB  Fri Mar 23 21:39:18 2018    catalog/
  rwxr-xr-x    root    4 KiB  Fri Jan  5 13:35:20 2018    tests/
  rwxr-xr-x    root    4 KiB  Fri Mar 23 21:39:18 2018    user/
  rwxr-xr-x    root    4 KiB  Fri Mar 23 21:39:18 2018    user-environment-generators/
  rwxr-xr-x    root    4 KiB  Wed Oct  4 06:28:34 2017    user-generators/
  rw-r--r--    root  181 B    Thu Mar  8 00:57:49 2018    system
```

tatokis, 2018-03-27:
Turns out, "system" was a file instead of a directory. Deleting it fixed the issue.

I tried making cmake check if it is a directory or not, but I couldn't figure it out.
I'll leave it as-is for now, but if we get another report about this, I will look more towards a workaround.

closed 2018-03-27 by tatokis