ckb-next/ckb-next #134: CMake Script Unable To Find libudev on Solus

crredsox13, 2018-03-30:
Hello, 

I am trying to install ckb-next on solus, however I am running into an issue with the CMake script while trying to find libudev. I am able to confirm that it does exist on my machine but I receive the below error when trying to run the quick install script as well as cmake -H. -Bbuild -DCMAKE_BUILD_TYPE=Release -DSAFE_INSTALL=ON -DSAFE_UNINSTALL=ON. 

-- Checking for module 'libudev'
--   No package 'libudev' found
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find UDEV (missing: UDEV_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindUDEV.cmake:82 (find_package_handle_standard_args)
  src/daemon/CMakeLists.txt:56 (find_package)

I understand exposure to udev for third party apps is something distro maintainers are sometimes iffy about so I have included the paths to the files below:

/usr/lib32/libudev.so
/usr/lib32/libudev.so.0
/usr/lib32/libudev.so.1
/usr/lib32/libudev.so.1.6.7
/usr/lib64/libudev.so
/usr/lib64/libudev.so.0
/usr/lib64/libudev.so.1
/usr/lib64/libudev.so.1.6.7

Any assistance would be greatly appreciated. 

[CMakeError.log](https://github.com/ckb-next/ckb-next/files/1862588/CMakeError.log)
[CMakeOutput.log](https://github.com/ckb-next/ckb-next/files/1862589/CMakeOutput.log)
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1862590/ckb-next-dev-detect-report.gz)

light2yellow, 2018-03-30:
Do you have `systemd-devel` installed?

crredsox13, 2018-03-30:
That was it. I was also missing libappindicator-devel. Seems to be working now. Thanks for your help.

closed 2018-03-31 by tatokis

egberts, 2023-09-09:
Update:

To fix it and make ckb-next usable on Debian 12:

```
sudo apt-get install qtbase5-dev
sudo apt-get install qtdeclarative5-dev
sudo apt install libquazip1-qt5-1
sudo apt install libquazip1-qt5-dev
sudo apt install pkg-config 
sudo apt install libxcb-screensaver0
sudo apt install libxcb-screensaver0-dev
sudo apt install libxcb-ewmh-dev
```