ckb-next/ckb-next #135: The ckb.app does not open in OS X (10.9.5).

jirikaspar, 2018-03-30:
# Bug report

## General information

### Source

pre-built pkg for macOS

## Logs & crash reports

n/a

## What is the program's current behavior?

I deleted it using the terminal command that you wrote.

## What is the expected behavior of the program?

I should be able to open ckb.app.

## Any other notes & comments?

I tried to open the program by right-clicking (control-clicking) on ckb.app and selecting Open, but it did not work. I also restarted my computer, but it did not help.

tatokis, 2018-03-30:
Unfortunately 10.9 support was dropped some time ago.

We should be able to make a build that works on 10.9 though.
I'll have a look later today or tomorrow and see what can be done.

jirikaspar, 2018-03-30:
"Unfortunately 10.9 support was dropped some time ago."

I have just tried to find this information in the [wiki](https://github.com/ckb-next/ckb-next/wiki), but I was not successful. Has this information been included anywhere? If not, then it might be worth adding it under *macOS Installation* so that users know which versions of OS X are currently supported.

"We should be able to make a build that works on 10.9 though.
I'll have a look later today or tomorrow and see what can be done."

Thank you for your willingness to check what can be done, I really appreciate it.

tatokis, 2018-03-30:
We weren't aware of it either.
I only now just found out about it by looking at https://doc.qt.io/qt-5/supported-platforms.html listing `macOS 10.11, 10.12, 10.13 (x86_64)` as the supported platforms.

If anything, you are the first person to point it out. At the very least, we were asking people to test something on 10.11 and it took months before someone actually did, so there was no way of telling that Qt dropped support for platforms we barely get any feedback on.

jirikaspar, 2018-03-30:
I shall attempt that once I find out how.

tatokis, 2018-03-31:
This is the best I could do for now. You will need to manually install it.
http://ckb-next.info/mac/ckbnext239-g42db047-bin.zip
```
3405bfc4e1658d9b950f6777cc3c8e1c74ea9feac0b47158ea80486410098274  ckbnext239-g42db047-bin.zip
```

tatokis, 2018-04-08:
I finally got a proper package for it. Please report back if it works
http://ckb-next.info/mac/ckb-next_v0.2.9-158-gaf64f6d_-legacy.dmg

closed 2018-04-10 by tatokis