ckb-next/ckb-next #137: STRAFE RGB not detected but a new firmware is available

dikey94, 2018-03-31:
Hello everyone,
I'm trying to configure my STRAFE RGB with ckb-next. It use to work before ckb-next was installed, but for now is dead. Opening ckb-next gui window gives me "A new firmware is available for your STRAFE RGB (v2.05). Would you like to install it now?" but clicking Yes doesn't me any result.
I'd appreciate any help.
Thanks.

ZirconiumX, 2018-03-31:
This sounds like a communication failure between the GUI and daemon.

However, you've deleted quite a lot of useful information in the template. What OS are you using? What version of ckb-next?

tatokis, 2018-03-31:
Re-read the template and fill it in properly

dikey94, 2018-03-31:
Sorry.
It was semi-automatic installation of ckb-next:
```
git clone https://github.com/ckb-next/ckb-next.git
cd ckb-next
./quickinstall
```

I'm using Debian GNU/Linux 9 (stretch) 64bit.

> sh ./ckb-next-dev-detect

 gives me 
> ./ckb-next-dev-detect: 8: ./ckb-next-dev-detect: Syntax error: newline unexpected

It's ckb-next 0.2.9-124-g42db047

ZirconiumX, 2018-03-31:
Just run `./ckb-next-dev-detect`, because it's tripping over what I think
is a bashism, and `/bin/sh` is not bash on Debian.

dikey94, 2018-03-31:
```
damian@Damian-PC:~$ ./ckb-next-dev-detect
bash: ./ckb-next-dev-detect: Permission denied
damian@Damian-PC:~$ sudo ./ckb-next-dev-detect
sudo: ./ckb-next-dev-detect: command not found
```

but ckb-next-dev-detect is there

ZirconiumX, 2018-03-31:
`chmod +x ./ckb-next-dev-detect`

dikey94, 2018-03-31:
```
damian@Damian-PC:~$ chmod +x ./ckb-next-dev-detect
damian@Damian-PC:~$ sudo ./ckb-next-dev-detect
./ckb-next-dev-detect: 8: ./ckb-next-dev-detect: Syntax error: newline unexpected

```

tatokis, 2018-03-31:
Please clone the repository again in another directory and try simply running ./ckb-next-dev-detect

dikey94, 2018-03-31:
```
damian@Damian-PC:~/Desktop/ckb-next$ ./ckb-next-dev-detect
This script will collect information about the Corsair devices in your system.
Please make sure they are plugged in and press Enter.

dmesg: read kernel buffer failed: Operation not permitted
Done! Please send the following report to the developers.
Location: /home/damian/Desktop/ckb-next/ckb-next-dev-detect-report.gz
```

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1865342/ckb-next-dev-detect-report.gz)

ZirconiumX, 2018-03-31:
`bash --version`

This looks very strange. It should not be choking on that line of code.

dikey94, 2018-03-31:
```
damian@Damian-PC:~/Desktop/ckb-next$ sudo ./ckb-next-dev-detect
This script will collect information about the Corsair devices in your system.
Please make sure they are plugged in and press Enter.

Done! Please send the following report to the developers.
Location: /home/damian/Desktop/ckb-next/ckb-next-dev-detect-report.gz
```

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1865345/ckb-next-dev-detect-report.gz)

dikey94, 2018-03-31:
```
damian@Damian-PC:~/Desktop/ckb-next$ bash --version
GNU bash, version 4.4.12(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

```

ZirconiumX, 2018-03-31:
Right, you should drop by IRC. It's the blue Freenode button at the top of
the readme.

tatokis, 2018-03-31:
Turns out it was a hardware + firmware issue.
Resolved by plugging the strafe to a usb hub and updating the firmware using that.

closed 2018-03-31 by tatokis