ckb-next/ckb-next #138: MacOS - Prevents "Shake mouse pointer to locate"

mdahlke, 2018-03-31:
After installing, the `Shake mouse pointer to locate` setting no longer functions.

I discovered this after I was trying to allow a System Extension and clicking on the `Allow` button did nothing. After a little bit of Googling, I stumbled upon this: https://github.com/tekezo/Karabiner-Elements/issues/1017#issuecomment-335165442

So I started remove everything that wasn't necessary on my system, and once I sent `ckb-next` to the trash, I was able to click the `Allow` button AND the `Shake mouse pointer to locate` was working again.

OS Version: `10.13.3 (17D102)`
ckb-next v0.2.8

ZirconiumX, 2018-03-31:
From what I read on the issue you linked, this is from Apple's side, not
ours. There's not much we can do, I'm afraid.

stumbled upon this: tekezo/Karabiner-Elements#1017 (comment)

mdahlke, 2018-03-31:
I guess I'm  confused why keyboard software is messing with the mouse pointer. I don't know anything about software development on a mac, but I don't see why this software needs to interfere with the mouse.

P.S. I am very glad this exists and is overall done well. This issue will not prevent me from uninstalling, its just an annoyance.

tatokis, 2018-03-31:
I assume it is because we provide virtual mouse and keyboard input, and since macOS detects that, it doesn't let the user click it.
I honestly can not tell at this point. Maybe it is because the code registers mouse input too, as the driver supports mice too.

ZirconiumX, 2018-03-31:
ckb-next is not "just keyboard software". We control mice and mousepads,
too.

However, we do this by asking the kernel to give us USB packets from the
keyboard, which we can read and interact with. Perhaps Apple thinks this is
snooping and won't let you change settings? I don't know.

Nobody currently on the dev team has a Mac to test with.

mdahlke, 2018-03-31:
I know you may not like this suggestion, but is there anyway of decoupling the mouse and keyboard software. Whether it be two separate pieces or a setting that can completely disable either the keyboard functionality or the mouse functionality depending on what you are using it for?

tatokis, 2018-03-31:
@mdahlke That's possible, but the reason it is done this way is so that you can bind mouse buttons to keyboards and keyboard keys to mice.
Such behaviour would break that functionality, especially for mice that have a many buttons on them such as the scimitar.

mdahlke, 2018-03-31:
@tatokis I understand. So I guess this is just a feature request at this point for being able to disable either mouse functionality, or keyboard functionality depending on your usage and needs. Maybe this could be part of the setup process, or auto-detected based on the peripherals connected (that's assuming that the keyboard/mice binding works only with Corsair products and not with something like a Corsair keyboard and a LG mouse)?

tatokis, 2018-03-31:
This only works for Corsair devices, yes, but you don't need to have a Corsair mouse to bind the scroll wheel to your Corsair keyboard, or a Corsair keyboard to bind keys to your Corsair mouse.

PiotrKrzyzek, 2018-04-17:
Can confirm that after installing ckb-next 'shake mouse to locate' doesn't work.

tatokis, 2018-04-30:
@mdahlke Can you check if this fixes both the shake mouse pointer to locate and also that the Allow button works?
(Reboot after updating to the new version.)
http://ckb-next.info/mac/ckb-next_v0.3.0-4-g14320ab.dmg

mdahlke, 2018-05-01:
@tatokis I have v0.3.0-4 installed.
* The mouse pointer shake to locate is working.
* I'm not sure what you mean with the "Allow" button. Where am I looking for this and what does it _allow_ me to do?

New bug introduced:
* None of my `CTRL`, `CMD`, or `ALT` keys work on the keyboard (Corsair STRAFE RGB)
   Those keys **do** still work on the built-in mac keyboard

tatokis, 2018-05-01:
@mdahlke I guess I'll revert the change.
By allow I meant the system extension prompt you were on about in the original post.

It doesn't matter though, since it breaks modifiers. @PiotrKrzyzek Can you try it and confirm that `CMD + click` is broken?
Basically what happens is the event tap we use to inject modifiers is what causes the problems this issue was opened for... *Sigh*

PiotrKrzyzek, 2018-05-01:
Hm ... Let's see:

Keyboard commands work just fine. But yes, after reading this yeah ...

Can confirm: "cmd + click" is borked.

Though mouse shaking is working at least. :)

mdahlke, 2018-05-01:
@tatokis Yes, I was able to click the `Allow` button and the `Locate to shake` worked as well when installing the daemon you sent over (http://ckb-next.info/mac/ckb-next_v0.3.0-4-g14320ab.dmg).

Also, my modifier keys started working again. I didn't reboot or anything (excluding after the install). My system started lagging a bit and and keys didn't really do anything for a a few seconds, and then all of sudden the modifiers are working as expected. Very strange and I don't know how it can be replicated....

@PiotrKrzyzek are your modifiers still not working?

PiotrKrzyzek, 2018-05-01:
Nope, not working and that version also messed something up that the average ckb-next CPU usage is now ~25% :|

mdahlke, 2018-05-01:
I'm not seeing the same CPU usage issue...
[image]

PiotrKrzyzek, 2018-05-01:
I think this conversation belongs in a seperate thread, but here's the Activity Monitor shows for me:

[image]

I jumps between ~15% and 37%.

Where previously it was hovering between 5% and 9%. So, something changed. I did manage to get it back down to between 8.4% and 13.7% when I disabled 'show animation' though. Not sure why things changed suddenly though.

tatokis, 2018-07-19:
http://ckb-next.info/mac/ckb-next_v0.3.0-62-g68fedda_10.11-SDK.dmg
Karabiner

closed 2018-07-19 by tatokis