ckb-next/ckb-next #139: Segfault when running ckb-next

azacock, 2018-04-01:
Installed on Debian Sid, using quickinstall with the latest from master. Seems to install correctly, and ckb-next-daemon is running without issues.

But, when I run ckb-next, the window pops up and immediately dies with a segfault. Here's everything that gets spit out to the console. Am I missing a runtime dep or something?

```
QObject::startTimer: Timers cannot have negative intervals

(process:23352): GLib-GObject-WARNING **: 17:41:32.272: invalid (NULL) pointer instance

(process:23352): GLib-GObject-CRITICAL **: 17:41:32.272: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:23352): Gtk-CRITICAL **: 17:41:32.273: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:23352): GLib-GObject-WARNING **: 17:41:32.273: invalid (NULL) pointer instance

(process:23352): GLib-GObject-CRITICAL **: 17:41:32.273: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:23352): Gtk-CRITICAL **: 17:41:32.273: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:23352): GLib-GObject-CRITICAL **: 17:41:32.273: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(process:23352): GLib-GObject-WARNING **: 17:41:32.273: invalid (NULL) pointer instance

(process:23352): GLib-GObject-CRITICAL **: 17:41:32.273: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:23352): Gtk-CRITICAL **: 17:41:32.273: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:23352): Gtk-CRITICAL **: 17:41:32.273: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:23352): GLib-GObject-CRITICAL **: 17:41:32.274: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(process:23352): GLib-GObject-WARNING **: 17:41:32.274: invalid (NULL) pointer instance

(process:23352): GLib-GObject-CRITICAL **: 17:41:32.274: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:23352): Gtk-CRITICAL **: 17:41:32.274: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Path to settings: "/home/azathoth/.config/ckb-next/ckb-next.conf"
Looking for old settings in: "/home/azathoth/.config/ckb/ckb.conf"
Scanning  "/usr/local/libexec/ckb-next-animations/gradient"
Scanning  "/usr/local/libexec/ckb-next-animations/heat"
Scanning  "/usr/local/libexec/ckb-next-animations/mviz"
Scanning  "/usr/local/libexec/ckb-next-animations/pinwheel"
Scanning  "/usr/local/libexec/ckb-next-animations/rain"
Scanning  "/usr/local/libexec/ckb-next-animations/random"
Scanning  "/usr/local/libexec/ckb-next-animations/ripple"
Scanning  "/usr/local/libexec/ckb-next-animations/wave"
Starting  "/usr/local/libexec/ckb-next-animations/wave"
Starting  "/usr/local/libexec/ckb-next-animations/wave"
Starting  "/usr/local/libexec/ckb-next-animations/gradient"

(process:23352): Gtk-WARNING **: 17:41:32.461: Screen for GtkWindow not set; you must always set
a screen for a GtkWindow before using the window

(process:23352): Gdk-CRITICAL **: 17:41:32.461: IA__gdk_screen_get_display: assertion 'GDK_IS_SCREEN (screen)' failed

(process:23352): Gdk-CRITICAL **: 17:41:32.461: IA__gdk_keymap_get_for_display: assertion 'GDK_IS_DISPLAY (display)' failed

(process:23352): GLib-GObject-WARNING **: 17:41:32.461: invalid (NULL) pointer instance

(process:23352): GLib-GObject-CRITICAL **: 17:41:32.461: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:23352): GLib-GObject-WARNING **: 17:41:32.556: invalid (NULL) pointer instance

(process:23352): GLib-GObject-CRITICAL **: 17:41:32.556: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:23352): Gdk-CRITICAL **: 17:41:32.556: IA__gdk_screen_get_root_window: assertion 'GDK_IS_SCREEN (screen)' failed

(process:23352): Gdk-CRITICAL **: 17:41:32.556: IA__gdk_screen_get_display: assertion 'GDK_IS_SCREEN (screen)' failed

(process:23352): Gdk-CRITICAL **: 17:41:32.556: IA__gdk_x11_display_get_xdisplay: assertion 'GDK_IS_DISPLAY (display)' failed

(process:23352): Gdk-CRITICAL **: 17:41:32.556: IA__gdk_screen_get_number: assertion 'GDK_IS_SCREEN (screen)' failed
[1]    23352 segmentation fault  ckb-next
```

tatokis, 2018-04-01:
Run ckb-next-dev-detect

azacock, 2018-04-01:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1866735/ckb-next-dev-detect-report.gz)

tatokis, 2018-04-01:
Try installing qt5ct and then launching the GUI with `QT_QPA_PLATFORMTHEME=qt5ct ckb-next`

azacock, 2018-04-01:
Same segfault.

tatokis, 2018-04-01:
Don't think there is much that can be done on the project side. This is a Qt issue, as we do not use gtk, gdk, nor glib.

It's either a problem with your system, or a bug. Last time I tried ckb-next on debian (and someone else recently also set it up on a debian system), there were no such issues.

azacock, 2018-04-01:
Looking around, it seems to be a problem with appindicator.

I removed `libappindicator1` and `libappindicator-dev` and instead installed `libappindicator3-1` and `libappindicator3-dev`, and everything seems to be working perfectly now. Maybe something to add to the troubleshooting page.

tatokis, 2018-04-01:
Any idea what theming engine debian uses for Qt?

tatokis, 2018-04-01:
To expand on that, we do not link against libappindicator3 because Qt typically uses gtk2, and gtk3 isn't allowed, so as far as cmake is concerned, there was a lack of appindicator1, so it was disabled.

tatokis, 2018-04-03:
I can't reproduce the bug.
[image]

craw169, 2018-04-18:
I am seeing the same on Fedora 27.  I originally installed ckb from fedora package and it was version 0.2.9.  i uninstalled and then cloned the repo and did quickinstall.  The daemon is running but i get seg fault when running the GUI.  I tried adding QT_QPA_PLATFORMTHEME=qt5ct as suggested above but still get seg fault.  Should i open new issue or should this one be re-opened?  Here is my terminal output when running ckb-next:

crawford@TheMachine ~] $ QT_QPA_PLATFORMTHEME=qt5ct ckb-next
QObject::startTimer: Timers cannot have negative intervals

(process:4313): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(process:4313): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:4313): Gtk-CRITICAL **: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:4313): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(process:4313): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:4313): Gtk-CRITICAL **: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:4313): GLib-GObject-CRITICAL **: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(process:4313): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(process:4313): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:4313): Gtk-CRITICAL **: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:4313): Gtk-CRITICAL **: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:4313): GLib-GObject-CRITICAL **: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(process:4313): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(process:4313): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:4313): Gtk-CRITICAL **: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Path to settings: "/home/crawford/.config/ckb-next/ckb-next.conf"
inotify_add_watch("/home/crawford/.config/qt5ct") failed: "No such file or directory"

(process:4313): Gtk-WARNING **: Screen for GtkWindow not set; you must always set
a screen for a GtkWindow before using the window

(process:4313): Gdk-CRITICAL **: IA__gdk_screen_get_display: assertion 'GDK_IS_SCREEN (screen)' failed

(process:4313): Gdk-CRITICAL **: IA__gdk_keymap_get_for_display: assertion 'GDK_IS_DISPLAY (display)' failed

(process:4313): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(process:4313): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:4313): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(process:4313): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:4313): Gdk-CRITICAL **: IA__gdk_screen_get_root_window: assertion 'GDK_IS_SCREEN (screen)' failed

(process:4313): Gdk-CRITICAL **: IA__gdk_screen_get_display: assertion 'GDK_IS_SCREEN (screen)' failed

(process:4313): Gdk-CRITICAL **: IA__gdk_x11_display_get_xdisplay: assertion 'GDK_IS_DISPLAY (display)' failed

(process:4313): Gdk-CRITICAL **: IA__gdk_screen_get_number: assertion 'GDK_IS_SCREEN (screen)' failed
Segmentation fault (core dumped)
[crawford@TheMachine ~] $

tatokis, 2018-04-21:
@craw169 Can you please run it with gdb and post a backtrace?
```
sudo gdb ./bin/ckb-next-daemon
(gdb) r
[...]
(gdb) bt full
```

tatokis, 2018-04-22:
A temporary workaround for now is to install qt5ct, the gtk2 theme, and use that instead.
https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#glib-critical-errors-and-lock-up-at-colour-chooser

Absolutely no idea what causes this, as I got a report of it happening on Ubuntu 16.04 with Qt 5.5...

craw169, 2018-04-22:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1935754/ckb-next-dev-detect-report.gz)

tatokis, 2018-04-22:
2ee2bbd7df3ea7e4d7d24ebbfc2c95ea9e3332dc Should fix this.
@craw169 since you have the Lighting Node, can you also run `sudo usbhid-dump -d 1b1c:0c0b`

craw169, 2018-04-22:
here you go....output from usbhid-dump you requested:

[crawford@TheMachine ckb-next] $ sudo usbhid-dump -d 1b1c:0c0b
[sudo] password for crawford: 
001:008:000:DESCRIPTOR         1524414233.865020
 06 C0 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 10
 09 01 81 02 95 40 09 01 91 02 95 00 09 01 B1 02
 C0

[crawford@TheMachine ckb-next] $ 

i will try 2ee2bbd now.

Thank you

ZirconiumX, 2018-04-22:
Unrelated, but do you by any chance watch Person of Interest?

craw169, 2018-04-22:
tried 2ee2bbd and i am no longer getting a seg fault.  thank you!  hmmn....no don't watch it but maybe i should start.  why do you ask?

closed 2018-04-22 by tatokis

ZirconiumX, 2018-04-22:
The Machine is a plot point in it. Was wondering if your box was named after it.