ckb-next/ckb-next #140: Can't install ckb-next with quickinstall error 2

maucm, 2018-04-02:
Hello, I have been trying to install ckb but for some reason when I run quick install I get error 2 as an output, this is what I get more specifically, I'm recently changed to ubuntu and I just want to be able to use my corsair keyboard please! 

Error message:

/home/mauca/anaconda3/lib/libharfbuzz.so.0: undefined reference to `FT_Get_Var_Blend_Coordinates'
collect2: error: ld returned 1 exit status
src/gui/CMakeFiles/ckb-next.dir/build.make:1223: recipe for target 'bin/ckb-next' failed
make[2]: *** [bin/ckb-next] Error 1
CMakeFiles/Makefile2:674: recipe for target 'src/gui/CMakeFiles/ckb-next.dir/all' failed
make[1]: *** [src/gui/CMakeFiles/ckb-next.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
An error occurred,
press enter to exit.

What can I do? Thank you in advance!

maucm, 2018-04-02:
Ok so when I ran sudo apt-get install build-essential cmake libudev-dev qt5-default zlib1g-dev libappindicator-dev libpulse-dev libquazip5-dev somehow, all packages were installed to my Anaconda3 directory, where should they be installed and how do I specify the directory?

tatokis, 2018-04-02:
Try running `printenv | grep -i anaconda`.
The old one uses qmake, the new one uses cmake, which is an entirely different build system.
My guess is anaconda overrides some environment variables that cmake uses and installs in that directory.

maucm, 2018-04-03:
Hello, 

Is there a way I can use the old one to make my keyboard and mouse function, if not, what can I do to install the packages where they are supposed to, in order to install ckb-next. 

This what I get:
`PATH=/home/mauca/anaconda3/bin:/home/mauca/bin:/home/mauca/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin`

tatokis, 2018-04-04:
When you say the packages aren't installed where they are supposed to, what do you mean?
Which packages? ckb-next? The dependencies?
Maybe try running `which cmake` and `which apt-get`

light2yellow, 2018-04-04:
Presumably you need a freetype2 version >= 2.7.1 in your anaconda environment.

tatokis, 2018-04-04:
@light2yellow Pretty sure installing inside anaconda is a bad idea

light2yellow, 2018-04-04:
It is, just like using anaconda in a linux distribution. But I respect the OP's decision, and would not suggest the removal of it. So, since it's in the PATH, and the error comes from anaconda's package root, all other dependencies are also pulled from there, so manipulating on the native sysroot is not going to help, therefore it makes sense to figure out the version of freetype2 on the side of anaconda.

light2yellow, 2018-04-04:
Another reason manipulating on the native sysroot probably won't help is that you can't do a partial upgrade (although as I type this I decided to make sure Canonical disallows this ([but they don't, *sigh*](https://wiki.ubuntu.com/U%2B1/partial_upgrade)), so you cannot be stuck with an incompatible combo.

And regarding Infinality, which is in 99% the reason of this problem, if one googles for it, - these patches are long broke and if anything, OP should have already encountered issues with other packages. @maucm but if you're _actually_ still using these patches, install vanilla fontconfig-freetype2-cairo asap. But I doubt anyone even cared to provide a package for Ubuntu, so there's that, anyway.

closed 2018-04-25 by tatokis