ckb-next/ckb-next #141: Build from source fails Mac OS 10.13.3

ekrof, 2018-04-08:
Hello I'm having trouble building from the latest source with the following error message: 

```
CMake Error at cmake/modules/DeployQt5.cmake:216:
  Parse error.  Expected a command name, got unquoted argument with text
  "//".
Call Stack (most recent call first):
  src/gui/CMakeLists.txt:491 (include)
```

I tried to go back to the stable 0.29 but there was a driver mismatch because of my previously installed newer builds. Attached are my CMakeOutput.log and CmakeError.log files

[CMakeError.log](https://github.com/ckb-next/ckb-next/files/1887767/CMakeError.log)
[CMakeOutput.log](https://github.com/ckb-next/ckb-next/files/1887768/CMakeOutput.log)

tatokis, 2018-04-08:
My apologies. It appears that in the process of trying to fix something else I broke this.
Try reverting https://github.com/ckb-next/ckb-next/commit/9ac819ef86290572f139039c414214cb68f071bc

tatokis, 2018-04-08:
Should be fixed now

closed 2018-04-08 by tatokis

ekrof, 2018-04-08:
Thanks dude you're awesome

commit f5802bf8 (2018-04-08): Fix cmake comment regression

Closes #141