ckb-next/ckb-next #142: Lots of trouble compiling latest source or installing for the first time

intellectualburger, 2018-04-09:
stumbled upon CKB when starting to use my K70 LUX rgb with my macbook. Downloaded and installed the PKG which is version  0.2.9. It works well except when the laptop goes to sleep the keyboard is no longer detected until i restart my computer. Then i read the PKGs are outdated and to get the latest version i need to download the source. 
ive followed all the commands in the tutorials on the wiki, installed all the pre requisites, i am just hitting a wall when it comes to understanding how to compile.

i got to the section "Generate the desired configuration with CMake, compile and install:"
i copied and pasted the command below this step and terminal said directory does not exist. i am really stuck here

tatokis, 2018-04-09:
All you need to do is run `brew install cmake qt5 quazip` and then run the `quickinstall` script, either by double clicking it or by doing `./quickinstall` in a terminal.
It will take care of the rest.

I should rewrite the installation page to make it more obvious.

That said, I don't think the sleep issue is fixed in the latest code.
I can't test it either, because I don't have a mac that can go to sleep.

Maybe once you get it installed you can help get some logs if you wish, to try and troubleshoot this.

tatokis, 2018-04-09:
There's also this more recent pre-release build http://ckb-next.info/mac/ckb-next_v0.2.9-158-gaf64f6d_-legacy.dmg

intellectualburger, 2018-04-09:
Thank you for writing back so fast! i double clicked the script in the directory of the initial source and it says completed but i dont know where it installed.

ill try this pre-release build too thanks!

tatokis, 2018-04-09:
It installs it to your system. After it's installed you might need to reboot and start the program from launchpad.

tatokis, 2018-04-09:
There was bug that was preventing the code from compiling on mac, which I have now fixed.
That is probably why it didn't get installed.

intellectualburger, 2018-04-09:
btw when i open ckb-next it says "driver mismatch (2.9) please update ckb-daemon"

is that because i didnt correctly uninstall the 2.9 pkg release from the releases page first?>

tatokis, 2018-04-09:
It usually means the old one is still running. Rebooting fixes it.

intellectualburger, 2018-04-09:
kind of random thing i figured out, i noticed the bios switch on the back of my k70 was set to polling rate 4ms mode which isnt default, idk why. i switched to 1 and now it works more often with ckb even after unplugging, will continue testing

ZirconiumX, 2018-05-03:
@intellectualburger Can we close this?

tatokis, 2018-07-19:
Closing due to inactivity

closed 2018-07-19 by tatokis