ckb-next/ckb-next #144: The app is not opening

MrCube88, 2018-04-10:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl -O https://github.com/ckb-next/ckb-next/blob/master/ckb-next-dev-detect
./ckb-next-dev-detect
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

# **I can't run that command... It gives me an error message: -bash: ./ckb-next-dev-detect: Permission denied**
---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

The app is not opening...

### Source

  * pre-built pkg for macOS
-->

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

The program is not opening.

## What is the expected behavior of the program?

The program should open normally.

## Any other notes & comments?

---

# Something else

I made a video that shows the application that does not open and the error message.

https://youtu.be/EszF4a6Jt1o

tatokis, 2018-04-10:
Can you please try running `open /Applications/ckb.app` and send its output?

MrCube88, 2018-04-10:
LSOpenURLsWithRole() failed with error -600 for the file /Applications/ckb.app.

MrCube88, 2018-04-10:
I have installed it but the app doesn't appear in the launchpad....

ZirconiumX, 2018-04-10:
Run it from Finder Applications.

MrCube88, 2018-04-10:
When I open the app it closes almost instantly...

Here is the error message:

Process:               ckb-next [8176]
Path:                  /Applications/ckb-next.app/Contents/MacOS/ckb-next
Identifier:            ckb-next
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           ckb-next [8176]
User ID:               501

Date/Time:             2018-04-10 15:55:31.971 +0200
OS Version:            Mac OS X 10.13.4 (17E199)
Report Version:        12
Anonymous UUID:        F46CB17F-9F96-BCAC-F026-C7AEA9509CF6

Sleep/Wake UUID:       3F408E61-2BAA-447B-9342-1F9E5868300C

Time Awake Since Boot: 170000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000068
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x68:
--> 
    __TEXT                 0000000104fdf000-0000000105160000 [ 1540K] r-x/rwx SM=COW  /Applications/ckb-next.app/Contents/MacOS/ckb-next

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_c.dylib             	0x00007fff6b274ced flockfile + 18
1   libsystem_c.dylib             	0x00007fff6b27d728 vfprintf_l + 28
2   libsystem_c.dylib             	0x00007fff6b2763c9 fprintf + 176
3   ckb-next                      	0x000000010509e287 main + 455
4   libdyld.dylib                 	0x00007fff6b1ea015 start + 1

Thread 1:: Qt bearer thread
0   libsystem_kernel.dylib        	0x00007fff6b33acfa __select + 10
1   org.qt-project.QtCore         	0x0000000106184f13 qt_safe_select(int, fd_set*, fd_set*, fd_set*, timespec const*) + 547
2   org.qt-project.QtCore         	0x0000000106185e25 QEventDispatcherUNIXPrivate::doSelect(QFlags<QEventLoop::ProcessEventsFlag>, timespec*) + 693
3   org.qt-project.QtCore         	0x0000000106186fc7 QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 231
4   org.qt-project.QtCore         	0x000000010612d051 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 401
5   org.qt-project.QtCore         	0x0000000105f720ee QThread::exec() + 110
6   org.qt-project.QtCore         	0x0000000105f75b5a 0x105f4d000 + 166746
7   libsystem_pthread.dylib       	0x00007fff6b502661 _pthread_body + 340
8   libsystem_pthread.dylib       	0x00007fff6b50250d _pthread_start + 377
9   libsystem_pthread.dylib       	0x00007fff6b501bf9 thread_start + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff6b33b292 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff6b502009 _pthread_wqthread + 1035
2   libsystem_pthread.dylib       	0x00007fff6b501be9 start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff6b33b292 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff6b50220e _pthread_wqthread + 1552
2   libsystem_pthread.dylib       	0x00007fff6b501be9 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff6b33b292 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff6b50220e _pthread_wqthread + 1552
[... 281 lines cut ...]
Stack                             10.5M        7 
VM_ALLOCATE                         64K        7 
__DATA                            22.3M      226 
__FONT_DATA                          4K        2 
__LINKEDIT                       195.6M       10 
__TEXT                           153.3M      232 
__UNICODE                          560K        2 
mapped file                       49.3M       15 
shared memory                      612K       10 
===========                     =======  ======= 
TOTAL                            550.0M      549 

Model: iMac16,2, BootROM IM162.0218.B00, 4 processors, Intel Core i5, 3,1 GHz, 8 GB, SMC 2.32f20
Graphics: Intel Iris Pro Graphics 6200, Intel Iris Pro Graphics 6200, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E424C54424C41522D4E5544
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E424C54424C41522D4E5544
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x156), Broadcom BCM43xx 1.0 (7.77.37.29.1a7)
Bluetooth: Version 6.0.5f3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD HTS541010A9E662, 1 TB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: Corsair Gaming GLAIVE RGB Mouse
Thunderbolt Bus: iMac, Apple Inc., 27.4

tatokis, 2018-04-10:
Try this one https://github.com/ckb-next/ckb-next/releases/download/v0.3.0-pre/ckb-next_v0.3.0-pre.dmg
I just built and tested it.

MrCube88, 2018-04-10:
It's not working D: I tried to open it from the launchpad, the finder and I tried to run `open /Applications/ckb-next.app` but nothing happens (no error messages)

MrCube88, 2018-04-10:
Ok... I have opened the app and it works normally... But when I try to drag a window or move the mouse while pressing left-click the pointer moves to the up-left corner of the screen....

tatokis, 2018-04-10:
Give me a moment. This issue was fixed, but I accidentally broke it again. I will update the builds and give you a new one.

tatokis, 2018-04-10:
https://github.com/ckb-next/ckb-next/releases/download/v0.3.0-pre-2/ckb-next_v0.3.0-pre-2.dmg
Might need to reboot afterwards.

MrCube88, 2018-04-10:
Ok all fixed! Thank you so much!

closed 2018-04-10 by MrCube88