ckb-next/ckb-next #145: [HELP]: Creating Macro to print "Non-English@ Charter

Niko9911, 2018-04-10:
I have been trying to create macro, where keyboard just inputs one charter. That carter is `¨`. Is it possible to set non-english charters into the macros...?

ZirconiumX, 2018-04-10:
This character is [U+00A8 DIAERESIS](https://apps.timwhitlock.info/unicode/inspect?s=%C2%A8), which can be entered through custom unicode codes.

Mac OS would require following [this guide](https://apple.stackexchange.com/a/183056) and holding Option while typing `0`-`0`-`a`-`8`, if I'm understanding it correctly.

Linux would require Ctrl-Shift-`u`-`0`-`0`-`a`-`8`.

You can then do these commands while recording a macro, and bind it to a key.

tatokis, 2018-04-21:
Closing due to no reply

closed 2018-04-21 by tatokis