ckb-next/ckb-next #146: Side Buttons of Corsair M65 n=Not Working

FatBBPanduh, 2018-04-11:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1896987/ckb-next-dev-detect-report.gz)

## What is the program's current behavior? Side mouse buttons dont work

## What is the expected behavior of the program? The side buttons of my mouse should work

## Any other notes & comments? My side mouse buttons wont work for discord when i run ckb on a mac with high sierra. IDK if its my betterdiscord (https://github.com/denBot/BetterDiscordOSXInstaller) or not but my mouse button 4 and 5 wont work at all

---

ZirconiumX, 2018-04-11:
This sounds like a duplicate of #38.

tatokis, 2018-04-11:
Have a look at #19. It might help

FatBBPanduh, 2018-04-13:
Issue #19 and #38 dont help

ZirconiumX, 2018-04-13:
You've been very unhelpful with this issue, not giving any information other than that the side buttons don't work (and we've shown in the issues that the side buttons don't work even without ckb-next because macOS doesn't automatically bind those buttons)

FatBBPanduh, 2018-04-13:
@ZirconiumX the side buttons worked before i started using ckb-next and they worked on the first day i was using ckb-next

mikeck1, 2018-06-23:
@ZirconiumX As far as the side buttons not working. How do you bind "back" to a button? The macro tool in ckb-next records: 
`+mouse4=0,-mouse4=0,+mouse1=0,-mouse1=0 `

I was able to get mouse1 or "click" to bind to mouse 4 or the "typical back button", but can't get the back button to go back like I want.

ZirconiumX, 2018-06-23:
That's because unlike in Windows, macOS does not assign the
forward/backward mouse buttons to navigate a browser.

It requires manual fiddling to bind mouse4/mouse5 to forward/backwards.

@tatokis, because this is a pretty common request, should we have an option
to automatically bind those buttons?

tatokis, 2018-06-23:
@ZirconiumX I have no idea how or what to bind them to. There is literally no forwards/backwards as far as I am aware.

mikeck1, 2018-06-24:
@tatokis I did get back to work but cannot get forward to work. 

http://sensible-side-buttons.archagon.net/

-Also if I click the aim button (below forward/backward) it makes my MacBook Pro (15-inch, 2017)  go to sleep lol.

kchen13, 2018-07-08:
The solution posted by @tatokis linked here [#19](https://github.com/ckb-next/ckb-next/issues/19) is the solution I've been using for ages with ckb and Corsair's own CUE beta software (now lost since I did a fresh install). ckb-next has been working fine though after using a newer build. 

My guess is if you set up the keys to bind with the scripts shown in above link and it doesn't work in that app it's possible that discord app may not have macOS default back and forth shortcuts. The native keyboard shortcut for back and forth, safari,finder, iTunes, etc is `cmd + [` and `cmd + ]` which essentially triggers these two options.
[image]

I think if the developers of ckb were to automatically bind the forward and backward buttons, those are the keyboard shortcuts they would would bind it to. They essentially work in every single program where back and forth are used. Even without the keyboard shortcuts, setting the buttons to run scripts that triggers `cmd + [` and `cmd + ]`  works great. Is it ideal? Well that's up for debate but does it work? Absolutely.

I even set the sniper/aim button to invoke mission control and the DPI Up button to invoke notification center. Note that I changed the keyboard shortcut of Notification Center in my settings to `option + spacebar` but from these examples you should be able expand and customize to your own needs.

Mission Control
`osascript -e 'do shell script "/Applications/Mission\\ Control.app/Contents/MacOS/Mission\\ Control"'`

Notification Center
`osascript -e 'tell application "System Events" to keystroke " " using option down'`

Here's a screenshot of how it's setup. Just paste the code into the the box for Launch program on key press.
[image]

And here's the back and forth from the link above.
Back:
`osascript -e 'tell application "System Events" to keystroke "[" using command down'`
Forward:
`osascript -e 'tell application "System Events" to keystroke "]" using command down'`

I hope this helps anyone using the M65 Pro RGB or any mouse.

chrisroselli, 2018-07-13:
Thank you @kchen13 This worked great!

tatokis, 2018-08-17:
Any issues regarding keystrokes not being picked up have been fixed with Karabiner.
On the next release, mouse4 and mouse5 will be bound to ⌘[ and ⌘] by default.

@mikeck1 sadly we can't integrate that code as ever since the karabiner branch got merged, we no longer send input events to the OS via standard userspace APIs.

This is the best solution that we can offer out of the box at the moment. Apologies for the late response to the issue.

If anyone wishes to revert this setting, they can rebind mouse4 and mouse5 manually to their respective options and then proceed to handle the mouse events with third party software.

closed 2018-08-17 by tatokis

tatokis, 2019-01-05:
@BedHeadNinja  Make sure you are on version 0.3.2 and try creating a new profile

macakanonimni, 2019-10-04:
Hey I am trouble setting this up/neither I understand how this works
My goal is to bind left ctrl to the forward button and left shift on the back button which I was able to do on my previous computer, but since I switched it doesn't work anymore on the new Mac.
I don't want to sound lazy like I don't check posts or anything, but literally have no clue what to do.
Any help would be appreciated :)

commit 34cd718a (2018-08-17): Bind forwards and backwards to macros on mac

Closes #146