ckb-next/ckb-next #149: Caps Lock issues on macOS

mariobrostech, 2018-04-14:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

Hi, I'm using an iMac Retina 5K 27" Late-2014 Model running macOS Sierra with the Corsair Strafe RGB keyboard.
Everything works great until I hit the Caps Lock button. Then, everything becomes a mess.
My computer starts freezing every 5 seconds (for a duration of about 5 seconds) indefinitely until the entire computer is restarted. It's really weird, but it's really annoying! I have since remapped the Caps Lock key to Shift, but it's not ideal. Please fix this issue when you can. It's not an urgent problem, but it's a nice-to-have. I have attached my log files below:

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1911428/ckb-next-dev-detect-report.gz)

tatokis, 2018-04-17:
Closing as duplicate of #1.
When I last looked into it, I couldn't find a way to work around it, and from what I saw it seems to be an OS issue that was fixed. Either update, or set the caps lock indicator to RGB in the settings and reboot.

closed 2018-04-17 by tatokis

tatokis, 2018-04-17:
I have absolutely no idea. I just know it happens on an old sierra installation I have, and it stopped happening to someone else when he updated. I think he was still using Sierra.

If you just want a quick fix, just set Indicator to RGB and reboot in the settings. https://github.com/ckb-next/ckb-next/issues/1#issuecomment-359292303

Rebooting after setting it seems necessary, as it appears that once the issue starts happening, it won't stop.

mariobrostech, 2018-04-17:
> settings. #1 (comment)

Thanks for the tips. I will keep them in mind 👌