ckb-next/ckb-next #153: CBK can find my mouse but not keyboard

LadyIce, 2018-04-16:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Try the latest code by building the software manually.
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.
- There are 3 types of issues: new device, feature, bug. Use only one. Delete redundant sections before sending.

---

# Devices information

Open a terminal and execute:
```
curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
```

Upload `ckb-next-dev-detect-report.gz` with this issue.

---

# New device support request

Upload:
  * Linux:
    * `ckb-next-dev-detect-report.gz` from the section above
  * macOS:
    * `ckb-next-dev-detect-report.gz` from the section above
    * full USB bus dump by following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

OvermindDL1, 2018-04-16:
You seem to be missing information in your issue, like the results of the hardware scan, what hardware you have, etc... etc... etc...?  o.O?

LadyIce, 2018-04-16:
I apologize. I downloaded the software because I own a Mac and bought a
Corsair K55 keyboard and a Corsair Harpoon Mouse. As you probably know that
Corsair does not offer compatible software.

tatokis, 2018-04-16:
@LadyIce Assuming you are running 0.3.0-pre, can you run these commands in a terminal and post the output?
If you're running 0.2.9, it doesn't have support for the K55.
```
sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist
sudo /Library/Application\ Support/ckb-next-daemon
```

LadyIce, 2018-04-17:
I'll give it a try. Thank you.

LadyIce, 2018-04-20:
/Library/LaunchDaemons/org.ckb-next.daemon.plist: No such file or directory

LadyIce, 2018-04-20:
Yeah I downloaded 2.9. I can't find 0.3.0-pre

tatokis, 2018-04-20:
https://github.com/ckb-next/ckb-next/releases/download/v0.3.0-pre-2/ckb-next_v0.3.0-pre-2.dmg 

It's in the releases page

tatokis, 2018-06-27:
Closing due to inactivity

closed 2018-06-27 by tatokis