ckb-next/ckb-next #154: CKB Unable to find K55, is found in detect.

AngelusDulcis, 2018-04-16:
This will be a thing since these went on sale, I have a K55 plugged into a Fedora system, CKB can find the mouse

I did use the fedora repo to install it via DNF. 

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1917006/ckb-next-dev-detect-report.gz)

tatokis, 2018-04-16:
@AngelusDulcis 0.2.9 doesn't have support for the K55.
Update to the latest code from git.

AngelusDulcis, 2018-04-16:
@tatokis Will do, turning out to be a pain to get the pre-requisites installed, will report back once this 3gb download completes.

tatokis, 2018-04-16:
There's a 3GB dependency? What package is it?

AngelusDulcis, 2018-04-16:
--   Package 'libudev', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find UDEV (missing: UDEV_LIBRARY UDEV_INCLUDE_DIR)

I am not finding a lot of help getting this installed on Fedora. This is using quickinstall script.

AngelusDulcis, 2018-04-16:
Fixed that, even though I checked provides in dnf, it was in systemd-devel.

tatokis, 2018-04-16:
udev is part of systemd. `systemd-devel` is the package.
https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#build-from-source

AngelusDulcis, 2018-04-16:
@tatokis  Thank you for the help so far :) Sadly I tried that script, and even more adventure showed up! 

dnf install gcc gcc-c++ make cmake glibc zlib-devel qt5-qtbase-devel quazip-qt5-devel libappindicator-devel systemd-devel pulseaudio-libs-devel

package qt5-qtbase-devel-5.9.1-9.fc27.i686 requires pkgconfig(gl), but none of the providers can be installed

package mesa-libGL-devel-17.2.2-4.fc27.x86_64 requires pkgconfig(xxf86vm), but none of the providers can be installed

nothing provides pkgconfig(xf86vidmodeproto) needed by libXxf86vm-devel-1.1.4-6.fc27.i686

package libappindicator-devel-12.10.0-16.fc27.x86_64 requires pkgconfig(gtk+-2.0), but none of the providers can be installed

package gtk2-devel-2.24.31-6.fc27.x86_64 requires libXrandr-devel, but none of the providers can be installed

nothing provides pkgconfig(xproto) needed by libXrandr-devel-1.5.1-4.fc27.x86_64

tatokis, 2018-04-16:
Quick search revealed https://forums.fedoraforum.org/showthread.php?308822-dnf-Error-nothing-provides&s=849167a128ff70c5eea729075eb33f47&p=1754493#post1754493

AngelusDulcis, 2018-04-16:
*head desks*

AngelusDulcis, 2018-04-16:
And after that, life is great, thank you @tatokis!!  

I'm certain this will pop up again, so Fedora friends, go look at you /etc/dnf/dnf.conf and make certain 

`exclude=xorg-x11*`

Is commented out.

closed 2018-04-16 by AngelusDulcis