ckb-next/ckb-next #155: K70 RGB under macOS HighSerria 10.13, anytime w

baozhuoran, 2018-04-17:
# Before you proceed
# Devices information
K70-RGB
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1921212/ckb-next-dev-detect-report.gz)

# Bug report

## General information

### Source
download  released .mpkg from github

## Logs & crash reports

## What is the program's current behavior?
The first time it works. After I pull out the usb cable of my K70 keyboard, and re-plug in, the software can not recognize the device, and will show the "no device connected" window. I have to sudo-kill the ckb-next-daemon, and restart, now it can recognize the device again.
## What is the expected behavior of the program?
As long as the daemon is on, it should always recognize the supported devices once they are connencted.

ZirconiumX, 2018-04-17:
That sounds like the Mac driver doesn't have hotplug support. @tatokis, since you're the only person with anything resembling a Mac system, can you investigate?

tatokis, 2018-04-30:
Sorry for not replying earlier.
Going through the logs, I don't see why this would happen. Only guess is that they require a delay before being initialised.

Does this issue happen when using another USB Controller and/or a USB hub?

danielylee, 2018-05-04:
Having the same issue when I connect my K70 to my desktop. When I restart mac, ckb-next recognizes keyboard but no keystrokes are sensed.

tatokis, 2018-05-04:
Can the lights be controlled while in that state?

baozhuoran, 2018-05-04:
No, since it shows disconnected in the software.

tatokis, 2018-05-04:
Right, so @danielylee has another issue.
@baozhuoran looking at the logs again, I see that the daemon gets stuck disconnecting the device.
Apparently I can now reproduce this issue, so I'll have a go at fixing it.

tatokis, 2018-05-04:
@baozhuoran Can you try with this daemon and see if the issue is fixed?
```
sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist
sudo ./ckb-next-daemon-dcblock
```
http://ckb-next.info/mac/
```
075e7c1cf0ddd848f9fa8f17954888e3d56609ced48da731890d7c7e6b28d00f  ckb-next-daemon-dcblock
```

tatokis, 2018-05-05:
@baozhuoran Sorry, I have a new build that should contain a proper fix.
Please test this one instead.
```
sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist
chmod +x ckb-next-daemon-dcblock-2
sudo ./ckb-next-daemon-dcblock-2
```
http://ckb-next.info/mac/ckb-next-daemon-dcblock-2
```
c800c254296127be2c894cf55fc137cd15712ccc17821f9ffc2d015555ccfa43  ckb-next-daemon-dcblock-2
```

baozhuoran, 2018-05-05:
thank you so much, I left my kb at company, I will test on Monday.

danielylee, 2018-05-05:
@tatokis running this worked for me, but the last command got stuck on
`[I] Registering EventTap for modifier keys.`
I had to Ctrl+C to get out of the process and then start
`sudo launchctl load /Library/LaunchDaemons/org.ckb-next.daemon.plist`

tatokis, 2018-05-07:
@baozhuoran Sorry, but ping

tatokis, 2018-05-08:
@AaronMartone The issue is that the daemon would hang when a device would be unplugged while it was running. Your log doesn't show a device being unplugged.

AaronMartone, 2018-05-08:
OK. With it running and the keyboard plugged in, I disconnected it and reconnected it and got the following:

```
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Connecting Corsair STRAFE RGB Gaming Keyboard at /var/run/ckb1
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[I] Starting input thread for /var/run/ckb1
[I] Setup finished for /var/run/ckb1
```

The keyboard was detected successfully after I reconnected it.

baozhuoran, 2018-05-09:
Sorry I was on vacation yesterday.

Here is what I got with the new build. 
zhuoran$ chmod +x ckb-next-daemon-dcblock-2 
zhuoran$ sudo ./ckb-next-daemon-dcblock-2
    ckb-next: Corsair RGB driver 0.3.0-9-g33c256d
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair Gaming K70 LUX RGB Keyboard  at /var/run/ckb1
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[I] Starting input thread for /var/run/ckb1
[E] os_sendindicators (usb_mac.c:222): Got return value 0xe0004051
[I] Registering EventTap for modifier keys.
[E] os_usbsend (via firmware.c:24): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:24): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:24): Got return value 0xe0004051
[E] os_usbsend (via firmware.c:24): Got return value 0xe00002ed
[W] _start_dev (device.c:43): Unable to load firmware version/poll rate
[I] Setup finished for /var/run/ckb1
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[W] iterate_devices_usb (usb_mac.c:761): Unable to seize USB handle, continuing anyway...
[I] Connecting Corsair Gaming K70 LUX RGB Keyboard  at /var/run/ckb1
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[I] Starting input thread for /var/run/ckb1
[I] Setup finished for /var/run/ckb1
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb1
[W] iterate_devices_usb (usb_mac.c:761): Unable to seize USB handle, continuing anyway...
[I] Connecting Corsair Gaming K70 LUX RGB Keyboard  at /var/run/ckb1
[E] iterate_devices_hid (usb_mac.c:879): Failed to create device plugin: e00002be
[I] Starting input thread for /var/run/ckb1
[I] Setup finished for /var/run/ckb1

Where is the UI for this build? Where should I adjust the led?
Thanks for the achievement and effort!

danielylee, 2018-05-09:
Having the same success with @AaronMartone, however the key bindings don't save every time I replug in the keyboard. Anyone else having this issue?

tatokis, 2018-05-09:
Try this http://ckb-next.info/mac/ckb-next_v0.3.0-9-g33c256d_-SDK.dmg

```
9d0c7003c8e03372b4f262d8c683e8af19bce1b15e92c1fd96be8676c8f82b98  ckb-next_v0.3.0-9-g33c256d_-SDK.dmg
```

baozhuoran, 2018-05-09:
Works like a charm. Thanks @tatokis !!

danielylee, 2018-05-09:
@tatokis You are a god, no need for the daemon to run in the background.

dibstern, 2018-05-10:
Amazing @tatokis, and thanks so much to @ZirconiumX for referring me here. Tested and works. Not sure if this will stick - if it doesn't, you'll hear from me, but if it does, you won't :)

asmyers, 2018-05-10:
Was also having these issues and fixed with @tatokis DMG supplied above.  Many thanks!!!!

tatokis, 2018-05-11:
Thank you everyone for your help. The fix will be included in the next release.
To anyone else stumbling across this, just install the pkg in the dmg linked above and you should be set.

closed 2018-05-11 by tatokis

dibstern, 2018-05-14:
Ever since downloading this fix from @tatokis, my tilde key no longer works. It shows as being assigned to the tilde key in the viewer, but now it outputs this key: §

Was there something in that DMG that caused this? Is there anyone else experiencing this?

ZirconiumX, 2018-05-14:
That sounds to me that ckb is detecting the wrong keyboard layout. There
should be a drop-down for choosing the virtual keyboard layout, but I'll
have to defer to @tatokis on that; the GUI is not my area of expertise.

dibstern, 2018-05-14:
My apologies, I realised this quickly and immediately deleted the comment,
but emails still go through. That was indeed the case.

ZirconiumX, 2018-05-14:
Yeah, Corsair only have a field for the physical keyboard layout, which
means we have to guess what the virtual layout is; sometimes we get it
wrong.