ckb-next/ckb-next #156: LL120 fans

tgeorgiop, 2018-04-18:
Hello. Does this work with ll120 corsair fans? Thanks.

ZirconiumX, 2018-04-18:
You're the first person to ever request adding fan support, so not yet.

However, we're always up for the challenge. You'll need to join us in IRC, so we can tell you how to collect the information we need.

OvermindDL1, 2018-04-18:
/me has been tempted to see if a Phillips ambx fan/light support pr would be accepted...

ZirconiumX, 2018-04-18:
After a brief look, the thing we need to support is not the fans themselves, but the Lighting Node Pro.

craw169, 2018-04-18:
i have a lighting node pro and was hoping i could help to get it to work.  i am a java programmer not c though so i am not sure how helpful i could be.

ZirconiumX, 2018-04-19:
Don't worry! First you need to run `ckb-next-dev-detect` and upload the output, craw169.

tatokis, 2018-04-22:
@craw169 If you have some time, would you mind dropping by the IRC room so that we can work on the lighting node pro?

craw169, 2018-04-22:
i would be glad to help in any way i can.  i have never used IRC though
so i will have to set that up.  i am kind of out of computer time today
(i have to finish my taxes today but i have more time tomorrow).  

thank you!

ZirconiumX, 2018-04-22:
There's a web client button at the top of the readme for that.

vlinkz, 2018-06-15:
You can use this, I haven't tested it but others have https://github.com/Chlorek/Colorsair

simoncoulton, 2018-08-08:
Also keen for this, here's an output from `ckb-next-dev-detect`:

```
ckb-next-dev-detect - Wed Aug  8 15:41:19 AEST 2018

Kernel: Darwin Hackintosh.local 17.6.0 Darwin Kernel Version 17.6.0: Tue May  8 15:22:16 PDT 2018; root:xnu-4570.61.1~1/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.13.5
BuildVersion:	17F77

USB Devices:
        H100i v2:

          Product ID: 0x0c09
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 1.00
          Serial Number: 7289_2.0
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair Components, Inc.
          Location ID: 0x14b00000 / 1
          Current Available (mA): 500
          Current Required (mA): 50
          Extra Operating Current (mA): 0
          Built-In: Yes
--
--
        Lighting Node PRO:

          Product ID: 0x0c0b
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 0.05
          Serial Number: 40B0011306191EAFDFD3245AC01B00F5
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14400000 / 8
          Current Available (mA): 500
          Current Required (mA): 50
          Extra Operating Current (mA): 0
          Built-In: Yes

IOReg:
  |   "USB Vendor Name" = "Corsair Components, Inc."
  |   "idVendor" = 0x1b1c
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "USB Serial Number" = "7289_2.0"
  |   "IOClassNameOverride" = "IOUSBDevice"
  | }
  | 
  +-o AppleUSBInterface@0  <class AppleUSBInterface, id 0x100000358, registered, matched, active, busy 0 (0 ms), retain 5>
  |   {
  |     "IOCFPlugInTypes" = {"2d9786c6-9ef3-11d4-ad51-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |     "bcdDevice" = 0x100
  |     "IOUserClientClass" = "IOUSBInterfaceUserClientV3"
  |     "idProduct" = 0xc09
  |     "bInterfaceSubClass" = 0x0
  |     "bConfigurationValue" = 0x1
  |     "locationID" = 0x14b00000
  |     "IOClassNameOverride" = "IOUSBInterface"
  |     "bInterfaceProtocol" = 0x0
  |     "idVendor" = 0x1b1c
  |     "bAlternateSetting" = 0x0
  |     "iInterface" = 0x0
  |     "bInterfaceNumber" = 0x0
  |     "bInterfaceClass" = 0xff
  |     "bNumEndpoints" = 0x2
  |   }
  |   
  +-o Google Chrome  <class IOUSBDeviceUserClientV2, id 0x10000087e, !registered, !matched, active, busy 0, retain 5>
      {
        "IOUserClientCreator" = "pid 2198, Google Chrome"
      }
      

  |   "USB Vendor Name" = "Corsair"
  |   "idVendor" = 0x1b1c
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "USB Serial Number" = "40B0011306191EAFDFD3245AC01B00F5"
  |   "IOClassNameOverride" = "IOUSBDevice"
  | }
  | 
  +-o AppleUSBInterface@0  <class AppleUSBInterface, id 0x1000003cf, registered, matched, active, busy 0 (0 ms), retain 5>
  |   {
  |     "IOCFPlugInTypes" = {"2d9786c6-9ef3-11d4-ad51-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |     "bcdDevice" = 0x5
  |     "IOUserClientClass" = "IOUSBInterfaceUserClientV3"
  |     "idProduct" = 0xc0b
  |     "bInterfaceSubClass" = 0x0
  |     "bConfigurationValue" = 0x1
  |     "locationID" = 0x14400000
  |     "IOClassNameOverride" = "IOUSBInterface"
  |     "bInterfaceProtocol" = 0x0
  |     "idVendor" = 0x1b1c
  |     "bAlternateSetting" = 0x0
  |     "iInterface" = 0x0
  |     "bInterfaceNumber" = 0x0
  |     "bInterfaceClass" = 0x3
  |     "bNumEndpoints" = 0x2
  |   }
  |   
  +-o Google Chrome  <class IOUSBDeviceUserClientV2, id 0x100000885, !registered, !matched, active, busy 0, retain 5>
      {
        "IOUserClientCreator" = "pid 2198, Google Chrome"
      }
      

Daemon log:

GUI log:
```

Running a AIO cooler and 4 LL120's with a Lighting Node Pro.

remotefixonline, 2018-08-13:
If you are still looking for people to test, I grabbed a lighting node pro today, I have it in a box that quad boots win10LTSB,server2016,fedora28, and kali  its on a evga x299 micro board.

ZirconiumX, 2018-08-15:
I've been lacking motivation to work on this, and I think @Chlorek said he wanted to help, but your offer is greatly appreciated all the same.

mbeach42, 2018-09-13:
Any progress on this?

ZirconiumX, 2018-09-13:
Nope.

Semperverus, 2018-09-17:
I personally second the Node Pro support request, and also possibly the headphone stand? I have a lot of corsair products and just made the full switch to Linux... -__-

```
ckb-next-dev-detect - Sun Sep 16 20:42:15 PDT 2018

Kernel: Linux Devin-Desktop 4.18.7-arch1-1-ARCH #1 SMP PREEMPT Sun Sep 9 11:27:58 UTC 2018 x86_64 GNU/Linux

OS Version:

USB Devices:
Couldn't open device, some information will be missing

Bus 003 Device 002: ID 1b1c:1b3e Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b3e 
  bcdDevice            1.10
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      58
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
    Interface Descriptor:
[... 707 lines cut ...]
8 ms
/dev/input/ckb2/fwversion
0110
/dev/input/ckb2/features
corsair scimitar rgb pollrate adjrate bind notify fwversion fwupdate hwload
/dev/input/ckb2/layout

/dev/input/ckb2/productid
1b3e
/dev/input/ckb2/serial
0503D01EAF0D2028592BE8C7F5001944
/dev/input/ckb2/model
Corsair Gaming SCIMITAR PRO RGB Mouse
/dev/input/ckb2/notify0
/dev/input/ckb2/cmd

Environment:
QT_SCREEN_SCALE_FACTORS=DP-1=1;DP-2=1;DP-3=1;HDMI-1=1;DVI-D-1=1;
XDG_SESSION_DESKTOP=KDE
XDG_CURRENT_DESKTOP=KDE
QT_AUTO_SCREEN_SCALE_FACTOR=0

Kernel command line:
BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=2da25ce8-169d-4981-be42-4b6c43929d65 rw quiet splash amdgpu.dpm=0
```

ZirconiumX, 2018-09-17:
The ST100 is already supported, because it's a reboxed Polaris.

etrigan63, 2018-10-23:
I will be installing a set of LL120's + Lighting Node Pro tonight. Running KDE Neon Bionic but switching to Kubuntu Cosmic as well. Will post my findings when I finish.

becausewat, 2019-01-31:
In addition to 
* https://github.com/Chlorek/Colorsair
and
* https://github.com/benburkhart1/lighting-node-pro
here is the output of `ckb-next-dev-detect` from my machine after a fresh reboot with a set of 5 LL120s and a Lighting Node Pro (there are also 4 sticks of Vengeance RGB Pro, so that may muddy the output):
```
ckb-next-dev-detect - Wed Jan 30 11:32:41 PST 2019

Kernel: Linux nanachi 4.20.5-arch1-1-ARCH #1 SMP PREEMPT Sat Jan 26 12:59:18 UTC 2019 x86_64 GNU/Linux

OS Version:
LSB Version:    1.4
Distributor ID: Arch
Description:    Arch Linux
Release:        rolling
Codename:       n/a

USB Devices:
Couldn't open device, some information will be missing

Bus 001 Device 003: ID 1b1c:0c0b Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c0b 
  bcdDevice            0.07
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

Dmesg on current boot:
Jan 30 11:29:48.797903 nanachi kernel: usb 1-14: New USB device found, idVendor=1b1c, idProduct=0c0b, bcdDevice= 0.07
Jan 30 11:29:48.798147 nanachi kernel: usb 1-14: Manufacturer: Corsair
Jan 30 11:29:48.798326 nanachi kernel: hid-generic 0003:1B1C:0C0B.0005: hiddev1,hidraw4: USB HID v1.11 Device [Corsair Lighting Node PRO] on usb-0000:02:00.0-14/input0

Dmesg on last boot:
Jan 29 14:01:12.795557 nanachi kernel: usb 1-14: New USB device found, idVendor=1b1c, idProduct=0c0b, bcdDevice= 0.07
Jan 29 14:01:12.795783 nanachi kernel: usb 1-14: Manufacturer: Corsair
Jan 29 14:01:12.795952 nanachi kernel: hid-generic 0003:1B1C:0C0B.0005: hiddev1,hidraw4: USB HID v1.11 Device [Corsair Lighting Node PRO] on usb-0000:02:00.0-14/input0

Daemon on current boot:
-- Logs begin at Tue 2018-07-24 14:35:31 PDT, end at Wed 2019-01-30 11:31:59 PST. --
-- No entries --

Daemon on last boot:
-- Logs begin at Tue 2018-07-24 14:35:31 PDT, end at Wed 2019-01-30 11:31:59 PST. --
-- No entries --

Dev nodes:
find: '/dev/input/ckb*': No such file or directory

Environment:
XDG_SESSION_DESKTOP=i3
XDG_CURRENT_DESKTOP=i3

Kernel command line:
BOOT_IMAGE=/vmlinuz-linux root=UUID=9db9156b-5d28-4ee3-acca-12e0d905a529 rw quiet
```

ProxyNeko, 2019-04-25:
Hello o/ New face around here but how are things with this issue? Do you still need testers because I have a 360 rad with three of the ll120 fans and the controller all wired up and I realized after installing them I need a way to turn them off when I'm not at my desk because my desktop stays on nearly all the time plus sometimes it's nice to have them be a static color rather than cycle through them

DirkTas67, 2019-06-19:
ckb-next-dev-detect - Wed Jun 19 21:02:16 CEST 2019

Kernel: Linux Cadmium 5.1.10-050110-generic #201906151034 SMP Sat Jun 15 10:36:59 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

USB Devices:

Bus 005 Device 003: ID 1b1c:1b4f Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b4f 
  bcdDevice            3.24
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K68 RGB Mechanical Gaming Keyboard
  iSerial                 3 0300B030AF3580C95B20DEB8F5001BC2
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           66
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     192
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
[... 283 lines cut ...]
/dev/input/ckb0/connected
/dev/input/ckb1 0300B030AF3580C95B20DEB8F5001BC2 CORSAIR K68 RGB Mechanical Gaming Keyboard
/dev/input/ckb1
/dev/input/ckb1/notify1
/dev/input/ckb1/pollrate
1 ms
/dev/input/ckb1/fwversion
0324
/dev/input/ckb1/features
corsair k68 rgb pollrate bind notify fwversion fwupdate
/dev/input/ckb1/layout
iso
/dev/input/ckb1/productid
1b4f
/dev/input/ckb1/serial
0300B030AF3580C95B20DEB8F5001BC2
/dev/input/ckb1/model
CORSAIR K68 RGB Mechanical Gaming Keyboard
/dev/input/ckb1/notify0
/dev/input/ckb1/cmd

Environment:

Kernel command line:
BOOT_IMAGE=/@/boot/vmlinuz-5.1.10-050110-generic root=UUID=d80bc7f9-2988-422f-88ab-45e041ee1aae ro rootflags=subvol=@ amd_iommu=on iommu=pt kvm_amd.npt=1

CenTrrikK, 2019-06-23:
Here another report. I could help with testing, too, if needed.

ckb-next-dev-detect - Sun Jun 23 10:12:44 CEST 2019

Kernel: Linux lnxgolf 5.1.11-arch1-1-ARCH #1 SMP PREEMPT Mon Jun 17 18:56:30 UTC 2019 x86_64 GNU/Linux

OS Version:

USB Devices:
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing

Bus 001 Device 006: ID 1b1c:0c09 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c09 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0

Bus 001 Device 005: ID 1b1c:0c0b Corsair 
[... 81 lines cut ...]
Daemon on current boot:
-- Logs begin at Thu 2019-03-14 23:34:15 CET, end at Sun 2019-06-23 10:12:44 CEST. --
Jun 23 10:08:15.203103 lnxgolf systemd[1]: Started Corsair Keyboards and Mice Daemon.
Jun 23 10:08:15.203950 lnxgolf ckb-next-daemon[29612]: ckb-next: Corsair RGB driver 0.4.0-27-g10e6b54
Jun 23 10:08:15.204090 lnxgolf ckb-next-daemon[29612]: [I] Root controller ready at /dev/input/ckb0

Daemon on last boot:
-- Logs begin at Thu 2019-03-14 23:34:15 CET, end at Sun 2019-06-23 10:12:44 CEST. --
-- No entries --

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/pid
29612
/dev/input/ckb0/version
0.4.0-27-g10e6b54
/dev/input/ckb0/connected

Environment:
XDG_SESSION_DESKTOP=gnome
XDG_CURRENT_DESKTOP=GNOME
QT_IM_MODULE=ibus

Kernel command line:
\\vmlinuz-linux root=UUID=f1c7d053-75a2-4e86-a96f-a1704f2cf669 rw add_efi_memmap initrd=intel-ucode.img intel_iommu=on iommu=pt initrd=\initramfs-linux.img

mrdev023, 2019-06-25:
ckb-next-dev-detect - Tue Jun 25 17:23:00 CEST 2019

Kernel: Linux florian-pc 5.1.8-1-MANJARO #1 SMP PREEMPT Sun Jun 9 20:44:14 UTC 2019 x86_64 GNU/Linux

OS Version:
LSB Version:	n/a
Distributor ID:	ManjaroLinux
Description:	Manjaro Linux
Release:	18.0.4
Codename:	Illyria

USB Devices:
Couldn't open device, some information will be missing

Bus 003 Device 002: ID 1b1c:0c0b Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c0b 
  bcdDevice            0.05
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

Dmesg on current boot:
Jun 25 17:20:52.058904 florian-pc kernel: usb 3-2: New USB device found, idVendor=1b1c, idProduct=0c0b, bcdDevice= 0.05
Jun 25 17:20:52.059157 florian-pc kernel: usb 3-2: Manufacturer: Corsair
Jun 25 17:20:52.059344 florian-pc kernel: hid-generic 0003:1B1C:0C0B.0004: hiddev0,hidraw1: USB HID v1.11 Device [Corsair Lighting Node PRO] on usb-0000:03:00.0-2/input0

Dmesg on last boot:
Jun 25 17:08:43.050892 florian-pc kernel: usb 3-2: New USB device found, idVendor=1b1c, idProduct=0c0b, bcdDevice= 0.05
Jun 25 17:08:43.051128 florian-pc kernel: usb 3-2: Manufacturer: Corsair
Jun 25 17:08:43.051302 florian-pc kernel: hid-generic 0003:1B1C:0C0B.0004: hiddev0,hidraw1: USB HID v1.11 Device [Corsair Lighting Node PRO] on usb-0000:03:00.0-2/input0
Jun 25 17:19:04.884004 florian-pc kernel: audit: type=1130 audit(1561475944.880:53): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=ckb-next-daemon comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'

Daemon on current boot:
-- Logs begin at Fri 2019-06-21 17:30:19 CEST, end at Tue 2019-06-25 17:21:53 CEST. --
Jun 25 17:20:54.684131 florian-pc systemd[1]: Started Corsair Keyboards and Mice Daemon.
Jun 25 17:20:54.687974 florian-pc ckb-next-daemon[1286]: ckb-next: Corsair RGB driver 0.4.0-27-g10e6b54
Jun 25 17:20:54.688250 florian-pc ckb-next-daemon[1286]: [I] Root controller ready at /dev/input/ckb0

Daemon on last boot:
-- Logs begin at Fri 2019-06-21 17:30:19 CEST, end at Tue 2019-06-25 17:21:53 CEST. --
Jun 25 17:19:04.883024 florian-pc systemd[1]: Started Corsair Keyboards and Mice Daemon.
Jun 25 17:19:04.885438 florian-pc ckb-next-daemon[7020]: ckb-next: Corsair RGB driver 0.4.0-27-g10e6b54
Jun 25 17:19:04.885748 florian-pc ckb-next-daemon[7020]: [I] Root controller ready at /dev/input/ckb0
Jun 25 17:20:01.453346 florian-pc ckb-next-daemon[7020]: [I] Caught signal 15
Jun 25 17:20:01.453346 florian-pc ckb-next-daemon[7020]: [I] Closing root controller
Jun 25 17:20:01.453346 florian-pc ckb-next-daemon[7020]: [I] Removed device path /dev/input/ckb0
Jun 25 17:20:01.453192 florian-pc systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
Jun 25 17:20:01.455585 florian-pc systemd[1]: ckb-next-daemon.service: Succeeded.
Jun 25 17:20:01.456303 florian-pc systemd[1]: Stopped Corsair Keyboards and Mice Daemon.

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/pid
1286
/dev/input/ckb0/version
0.4.0-27-g10e6b54
/dev/input/ckb0/connected

Environment:
XDG_SESSION_DESKTOP=deepin
QT_BEARER_POLL_TIMEOUT=-1
XDG_CURRENT_DESKTOP=Deepin
QT_DBL_CLICK_DIST=15

Kernel command line:
BOOT_IMAGE=/boot/vmlinuz-5.1-x86_64 root=UUID=21760dfd-4f39-474d-80da-f141deeabdfb rw quiet resume=UUID=a983cd60-9445-4516-8c8e-30963f906f37

Skylead, 2019-07-19:
I've really enjoyed ckb-next for my k70 and would love to help with Lighting Node Pro support

@ZirconiumX Through a happy shipping incident I actually ended up with 1 busted box of LL120 and a good box that newegg sent out to me. I'll be using one of the Lighting Node Pro's and the 5 functioning fans, however I have a spare Lighting Node Pro and a busted fan that I think will work after I gave it some love, but is likely to be noisy and not good for actual long term use. I am willing to ship the spare LNP and semi-working LL120 fan out to aid development.

mrdev023, 2019-08-04:
https://github.com/benburkhart1/lighting-node-pro project work with ll120 !

mrdev023, 2019-08-06:
I begin work to implement it but i never develop this project type. So, i may not be able to do it.
https://github.com/mrdev023/ckb-next
Do not hesitate to help me on the project.

mrdev023, 2019-08-08:
I finish to implement ll120 driver. now i add ll120 on GUI

olielvewen, 2019-11-03:
@mrdev023 

If you have finish to develop the lighting node pro support make a PR and open another issue about the GUI.
Thanks a lot.

gravityfargo, 2023-04-21:
Stale, and closing in favor of #639

closed 2023-04-21 by gravityfargo