ckb-next/ckb-next #157: CKB doesn't recognize ALT key on K70 RGB

niklasbeab, 2018-04-19:
Im on high sierra, and my ALT key is not recognized by the software. I tried to map different functions to the key but not a single one worked in some way.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1927806/ckb-next-dev-detect-report.gz)

tatokis, 2018-04-19:
Can you control the light on the Alt key? Is it the left or the right alt? What's the keyboard layout? Have you picked the right layout from the device settings?

niklasbeab, 2018-04-20:
Lightning is controllable on the key, yes. Actually both alt keys doesn't seem to work, left nor right.

Im using the german keyboard layout which is also selected in the device settings.

niklasbeab, 2018-04-20:
Just noticed and i feel kinda dumb right now HAHA: I meant the windows keys, not the alt keys.

ZirconiumX, 2018-04-20:
Can you [join us in IRC?](http://webchat.freenode.net/?channels=%23ckb-next&uio=d4)

niklasbeab, 2018-04-20:
Sure

niklasbeab, 2018-04-20:
[PLISTS.zip](https://github.com/ckb-next/ckb-next/files/1932308/PLISTS.zip)

All .plist files from the preference folder :)

niklasbeab, 2018-04-20:
[BROKEN.zip](https://github.com/ckb-next/ckb-next/files/1932319/BROKEN.zip)

The broken profile

Profil called "Profi 1"

tatokis, 2018-04-20:
@niklasbeab I had a look, and I think I see the issue here.
My guess is you accidentally pressed the winlock button, which disables the windows/super key.
After importing your profile and turning off winlock, the keys started working again.

If you still have the broken profile, switch to it and press the winlock button. It's the one on the top, to the right of the brightness button that has the windows logo inside a lock.

closed 2018-04-22 by tatokis