ckb-next/ckb-next #159: New device?         CORSAIR K63 Wireless Mechanical Gaming Keyboard:            Product ID: 0x1b45

dartoledo, 2018-04-21:
I have what it looks like a new device:
USB Devices:
        CORSAIR K63 Wireless Mechanical Gaming Keyboard:

          Product ID: 0x1b45
I fixed locally by updating https://github.com/ckb-next/ckb-next/blob/d7181772039d6fd193c4d498b538c2ad75e24ea5/src/daemon/usb.h
0x1b45 -> 0x1b40

ZirconiumX, 2018-04-21:
We already know about this; we've been working on it in the `wireless`
branch, which can light up a Dark Core. At the moment it doesn't yet light
up a K63WL, but we think we can do it soon.

dartoledo, 2018-04-21:
Wireless branch? cool ! thank you use that then.

closed 2018-04-21 by dartoledo

ZirconiumX, 2018-04-21:
If you could join us in IRC (the blue Freenode button) in a few hours,
that'd help us light up the keyboard, since we don't have any hardware of
our own.

dartoledo, 2018-04-21:
ok joined, fyi it lights up when using wireless branch

ZirconiumX, 2018-04-21:
Yeah, but as far as I know you can't set custom animations for the keyboard
in the GUI. There's also no battery status indicator because we haven't
figured that out yet.

dartoledo, 2018-04-21:
let me add connect logs here: 
OS Version:
ProductName:	Mac OS X
ProductVersion:	10.12.6
BuildVersion:	16G29

USB Devices:
        CORSAIR K63 Wireless Mechanical Gaming Keyboard:

          Product ID: 0x1b45
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 1.02
          Serial Number: xxxx
          Speed: Up to 12 Mb/sec
          Manufacturer: CORSAIR
          Location ID: 0x00300000 / 1
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

--
--
        CORSAIR K63 Wireless USB Receiver:

          Product ID: 0x1b50
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 1.02
          Serial Number: xxxx
          Speed: Up to 12 Mb/sec
          Manufacturer: CORSAIR
          Location ID: 0x14d00000 / 1
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
          Built-In: Yes

Daemon log:
    ckb-next: Corsair RGB driver 0.2.9-185-g8531096
[I] Root controller ready at /var/run/ckb0
[I] Registering EventTap for modifier keys.

[I] Caught signal 15
[I] Closing root controller
[I] Removed device path /var/run/ckb0
    ckb-next: Corsair RGB driver 0.2.9-185-g8531096
[I] Root controller ready at /var/run/ckb0
[I] Registering EventTap for modifier keys.

[I] Caught signal 15
[I] Closing root controller
[I] Removed device path /var/run/ckb0
    ckb-next: Corsair RGB driver 0.3.0-pre
[I] Root controller ready at /var/run/ckb0
[I] Registering EventTap for modifier keys.
[I] Registering EventTap for modifier keys.

[I] Caught signal 15
[I] Closing root controller
[I] Removed device path /var/run/ckb0
    ckb-next: Corsair RGB driver 0.3.0-pre
[I] Root controller ready at /var/run/ckb0
[I] Registering EventTap for modifier keys.
[I] Registering EventTap for modifier keys.

[I] Caught signal 15
[I] Closing root controller
[I] Removed device path /var/run/ckb0
    ckb-next: Corsair RGB driver 0.2.9-173-gb0a01d6
[I] Root controller ready at /var/run/ckb0
[I] Registering EventTap for modifier keys.
[... 93 lines cut ...]
Scanning  "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/ripple"
Scanning  "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/wave"
Starting  "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/random"
Starting  "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/ripple"
No GPG detected, signature verification disabled
Downloaded new firmware list. 31 entries found.
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Starting  "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Starting  "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/wave"
Starting  "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"
Can't find valid firmware version for device 6981 "6981"

ZirconiumX, 2018-04-21:
@tatokis would have to judge if the mutex errors are okay or not, but the
GUI firmware errors are to be expected (we don't currently support wireless
firmware update, because the dongle and the keyboard need to be updated
simultaneously, and have multiple sections).

With no disrespect, Mac is a little awkward for us to develop on; do you
have access to a Linux installation?

dartoledo, 2018-04-21:
Yeah no worries. Sorry no, I can do virtualbox with linux  at most.