ckb-next/ckb-next #161: feature request: Hyper key ( "cmd", "alt", "shift", "ctrl") modifiers as one

gleba, 2018-04-27:
Hello!
The **cbk-next**  is not compatible with the [Karabiner](https://github.com/tekezo/Karabiner-Elements) where there is this functional.
```
# echo 'macro caps+c:+lshift,+lalt,+lwin,+lctrl,+c,-c,-lctrl,-lwin-,-lalt,-lshift' > /var/run/ckb1/cmd
```
this does not work on *Corsair STRAFE* + 0.3.0 Pre-release
any ideas how to do it?

tatokis, 2018-04-27:
Try recording the macro with the GUI with no delay set. There seems to be a typo in your command above.

tatokis, 2018-04-27:
Also, you should probably only bind the caps key to lshift,lalt,lwin,lctrl, and then press C separately.
At the moment it's not possible to use key combinations for macros.

ZirconiumX, 2018-04-27:
>At the moment it's not possible to use key combinations for macros.

Then https://github.com/ckb-next/ckb-next/wiki/CKB-Daemon-Manual#key-macros should be updated, I think.

tatokis, 2018-04-27:
>Then https://github.com/ckb-next/ckb-next/wiki/CKB-Daemon-Manual#key-macros should be updated, I think.
Or it should be fixed in the code

gleba, 2018-04-27:
i think this is killer feature
when the **caps+v** is
[image]
now i use *hack*
```
echo 'macro caps+c:+f13,-f13' > /var/run/ckb1/cmd
```
[image]
that's laggy, but works in programs that have the ability to bind F13+ functional keys

In other configuration, I want to use a combination f9+f8 and f9+f7 for volume control and f1+g for launch my other secrets, such as rgb-highlight other secret combinations 
```
echo 'macro f1+f2:+f14,-f14' > /var/run/ckb1/cmd
echo 'macro f1+q:+f15,-f15' > /var/run/ckb1/cmd
```
This is what makes a cbk-next absolutely crazy **awesome!**

ZirconiumX, 2018-04-27:
I don't understand the first killer feature bit, as such.

But for combinations, I believe you should use bindings, not macros for that.

gleba, 2018-04-27:
> I don't understand the first killer feature bit, as such.

it's possible that your system does not have many hotkeys and you have enough modifiers Ctrl and iwin.
The simplest example is to call the clipboard history -
**`Caps + V`** better than **`Ctrl + Shift + Alt + V`**

... But perhaps this can only be understood with the help of hundreds of functions, such as clipboard history, color picker, alfred workflows, mission control and many other similar system tools.

gleba, 2018-04-27:
> But for combinations, I believe you should use bindings, not macros for that.
How?
```bind f1+1:F19```
??

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo