ckb-next/ckb-next #162: AWS Workspaces Issue

tchachra, 2018-04-28:
I installed ckb-next today on OS X - 10.3.5 High Sierra Beta.
Works fine...with the exception of Amazon AWS Workspaces...
I was able to login to my workspace as intended and my keyboard was fully functional in OS X...however, I was unable to use the keyboard in AWS Workspaces.  AWS workspaces are virtual desktops provided by AWS running Windows 10 vm's on Windows 2016 datacenter servers.

awkspace, 2018-04-30:
Having the same issue here.

Corsair Strafe RGB
ckb-next v0.3.0-rc2
Workspaces v2.4.4.638
MacBook Pro (13-inch, 2017)
3.5GHz Intel Core i7
16 GB 2133 MHz LPDDR3

ZirconiumX, 2018-04-30:
Can one of you try to reproduce the issue running Linux, so we know whether it's OS specific or not?

tatokis, 2018-04-30:
Duplicate of #38 but since it seems to have been fixed, we need to try and figure out how it was fixed in Karabiner and replicate the fix.
https://github.com/tekezo/Karabiner-Elements/issues/447#issuecomment-380413128

tatokis, 2018-04-30:
From a quick search I had, the only other viable way to post events that doesn't have these issues is through a kext.
Make what you will out of that...

awkspace, 2018-04-30:
@ZirconiumX Unfortunately, there is no Workspaces client in Linux, so I can't reliably replicate the problem.

tatokis, 2018-04-30:
I doubt it, but can you check if this helps the situation? (Reboot after updating to the new version.)
http://ckb-next.info/mac/ckb-next_v0.3.0-4-g14320ab.dmg

tchachra, 2018-04-30:
Just tried it. No luck. Same issue.

tatokis, 2018-05-08:
Just a quick update. I have completely rewritten the mac input code using Karabiner's VirtualHIDDevice. I'll hopefully have a dmg up soon for testing, that should fix all of these issues.

tatokis, 2018-05-08:
Keep in mind this is a test, but it should work.
After installing, go to Preferences -> Security & Privacy and make sure to allow the software.

You should reboot after doing that. Once you reboot, make sure key bindings work.

http://ckb-next.info/mac/ckb-next_v0.3.0-12-g0139c5d-vhid.dmg

```
749de7ebb40092df977ee1cc3bb387bbbd26660e9a0ad5f95248202d7da0c89c  ckb-next_v0.3.0-12-g0139c5d-vhid.dmg
```

awkspace, 2018-05-08:
@tatokis I did the following:

- Performed a full uninstall of ckb-next-v0.3.0-rc2.
- Rebooted.
- Installed the build you gave by going to Security & Privacy and allowing the installer to launch.
- Rebooted again.
- Now the keyboard is unresponsive at boot until I unplug & plug it back in. (It also doesn't appear in the RGB mode I had configured anymore - just the firmware default.)
- After unplugging & plugging it back in, I can type on the keyboard again, but there's still no RGB.
- ckb-next doesn't seem to recognize the keyboard at all, not displaying it in its window.

tatokis, 2018-05-08:
Can you please run ckb next dev detect and upload the report?

tatokis, 2018-05-08:
Actually, no need to do that. Just make sure to run `sudo launchctl unload /Library/LaunchDaemons/org.ckb-next.daemon.plist` and possibly sudo rm /var/log/ckb-next-daemon.log because the log kept getting spammed with new messages repeatedly.

awkspace, 2018-05-09:
I ran those two commands, but it still can't detect my keyboard.

Pardon my ignorance, but how do I run `ckb next dev detect` on MacOS?

tatokis, 2018-05-09:
@awkspace Those commands were to prevent the daemon from filling up your disk with a big log. I will send a new dmg to test in about 4~5 hours.

tatokis, 2018-05-09:
Try this one http://ckb-next.info/mac/ckb-next_v0.3.0-13-g0de6291-karabiner2.dmg

```
3395113aeb7db73d77fbf66e9330a1065019e27f9e23b71f95d8ceaadd34f88c  ckb-next_v0.3.0-13-g0de6291-karabiner2.dmg
```

tchachra, 2018-05-10:
Same issue....as well as a new one, once installed the lighting colors default to red and cannot be changed.  I installed ckb-next, rebooted, and the led's were all red...I did force update the firmware on the keyboard to reset it to factory and still I was only getting red colored led's.  Nonetheless, I still could not type in Amazon Workspaces. 
Thanks.

tatokis, 2018-05-10:
Updating the firmware doesn't reset its settings.
Additionally, have you allowed the kext to be loaded in Security and Privacy? It needs to be allowed separately from the installer.

Send back the output of `ls /Library/Extensions` and `sudo kextutil /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext`

tatokis, 2018-05-10:
Also either run ckb-next-dev-detect from the root of this repo or `curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash` and upload the report.

awkspace, 2018-05-11:
Tried the ckb-next_v0.3.0-13-g0de6291-karabiner2.dmg build you provided.

The good news is, it can find my keyboard again!

The bad news is, it still doesn't work in Workspaces.

1. Output of `ls /Library/Extensions`:

```shell
$ ls /Library/Extensions
ACS6x.kext
ATTOCelerityFC8.kext
ATTOExpressSASHBA2.kext
ATTOExpressSASRAID2.kext
ArcMSR.kext
CalDigitHDProDrv.kext
HighPointIOP.kext
HighPointRR.kext
PromiseSTEX.kext
SoftRAID.kext
org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext
```

Output of `sudo kextutil /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext`:

```shell
$ sudo kextutil /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext
$ echo $?
0
```

Output of `ckb-next-dev-detect`:

```shell
 curl https://raw.githubusercontent.com/ckb-next/ckb-next/master/ckb-next-dev-detect 2>/dev/null | bash
This script will collect information about the Corsair devices in your system.
2018-05-11 10:29:11.628 system_profiler[625:22825] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Done! Please send the following report to the developers.
```

And the associated report file: https://linx.li/o5bsgg94.gz

awkspace, 2018-05-30:
I know this isn't a very helpful update, but a few days and reboots later, "suddenly" the driver has started working. Sometimes I get duplicate keystrokes in Workspaces but a quick unload/load fixes that problem.

tatokis, 2018-06-05:
Sorry for the lack of responses, I'll look more into it in the upcoming weeks. Busy with school

tatokis, 2018-08-16:
If it doesn't work with the latest release, I am not sure what can be done.

closed 2018-08-16 by tatokis