ckb-next/ckb-next #165: Windows / super key doesn't work K95 on Ubuntu 18.04

petecog, 2018-05-01:
# Bug report

## General information
Hi, love ckb-next. Brilliant work. However spotted an issue after recent upgrade from Ubuntu 16.04 to 18.04 and upgrade of ckb-next to v0.3.0

### Source
  * ckb-next-git package in releases and local build as per instructions in wiki.

## Logs & crash reports

Upload:
 * `ckb-next-dev-detect-report.gz` from the section above
 * any additional crash reports you've managed to gather

## What is the program's current behaviour?
With the daemon running the super key does nothing e.g. window switching, dash start etc. With daemon stopped, normal function returns.

I checked the key binding [image] seems fine, and re-assigning dosn't fix. 

If I assign a different key to 'super' (e.g. alt) then it behaves as the super key should. 

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/1963564/ckb-next-dev-detect-report.gz)

## Any other notes & comments?
Bit of a cli noob, happy to supply other info if you tell me what you need.

tatokis, 2018-05-01:
Make sure winlock isn't enabled

petecog, 2018-05-01:
sorry, drawn a blank. How?

tatokis, 2018-05-01:
There's a button with a windows logo inside a lock on the top left of your keyboard. Pressing that disables and enables the super key.

raziel5746, 2019-12-25:
Awesome. I had my Windows key locked, and I didn't even know that was an option.
Thank you!

zee-bit, 2020-11-20:
Thanks so much. I had the windows key lock logo on the windows key itself, so just pressing `Fn + Win` fixed it for me. :)