ckb-next/ckb-next #166: Macro is ignored when pressing additional key.

Sur3, 2018-05-03:
Bug-report:
Hi when putting some single key events on the side keys of my corsair scimitar I can use them while keeping right mouse button pressed, but when having some macros on the sidekeys they are ignored while right mouse button is pressed. It would be great if macros could always execute when the button is pressed regardless of other buttons.

My ckb-next version 0.2.9.

Sur3, 2019-08-12:
Problem still exists in 0.4.0

tatokis, 2019-08-16:
Unfortunately no one has really worked on it

tosirius, 2020-05-04:
Having the same issue here too. Just installed today too. ckb-next 0.4.2 using a scimitar and a K70 rapidfire.

tatokis, 2020-05-04:
I fixed this in master, will be in the next release

closed 2022-10-03 by tatokis